z88dk/z88dk #1329: zcc +zx not creating binary / tap file

gazsp, 2019-11-15:
The following command does not produce a binary:

```
zcc +zx -v -O2 -lndos -no-cleanup -lim2 -create-app \
    src/screen.asm \
    src/mouse.asm \
    src/initmouse.asm \
    src/screen.c \
    src/keyboard.c \
    src/isr.c \
    src/main.c
```

The files are all compiled / assembled to /tmp/ but no .bin / .tap file is created.

I'm using the latest release on macOS.

The output of the command is:

```
PROCESSING src/screen.asm
cat "src/screen.asm" >> "/tmp/tmpBY6NUKL5.asm"
cat "/tmp/tmpBY6NUKL5.asm" >> "/tmp/tmpBY6NUKL5.tmp"
cat "/tmp/tmpBY6NUKL5.tmp" >> "/tmp/tmpBY6NUKL5.asm"
z80asm -I"/Users/garypearman/z88dk/lib/config//../..//lib/target/zx/def" -D__SCCZ80  -s  -mz80 -I"/Users/garypearman/z88dk/lib/config//../..//lib" -I"/Users/garypearman/Dropbox/Work/spectrum/mouse2/src"  "/tmp/tmpBY6NUKL5.asm"

PROCESSING src/mouse.asm
cat "src/mouse.asm" >> "/tmp/tmp57TGcAIB.asm"
cat "/tmp/tmp57TGcAIB.asm" >> "/tmp/tmp57TGcAIB.tmp"
cat "/tmp/tmp57TGcAIB.tmp" >> "/tmp/tmp57TGcAIB.asm"
z80asm -I"/Users/garypearman/z88dk/lib/config//../..//lib/target/zx/def" -D__SCCZ80  -s  -mz80 -I"/Users/garypearman/z88dk/lib/config//../..//lib" -I"/Users/garypearman/Dropbox/Work/spectrum/mouse2/src"  "/tmp/tmp57TGcAIB.asm"

PROCESSING src/initmouse.asm
cat "src/initmouse.asm" >> "/tmp/tmpZZ3fNrmK.asm"
cat "/tmp/tmpZZ3fNrmK.asm" >> "/tmp/tmpZZ3fNrmK.tmp"
cat "/tmp/tmpZZ3fNrmK.tmp" >> "/tmp/tmpZZ3fNrmK.asm"
z80asm -I"/Users/garypearman/z88dk/lib/config//../..//lib/target/zx/def" -D__SCCZ80  -s  -mz80 -I"/Users/garypearman/z88dk/lib/config//../..//lib" -I"/Users/garypearman/Dropbox/Work/spectrum/mouse2/src"  "/tmp/tmpZZ3fNrmK.asm"

PROCESSING src/screen.c
z88dk-ucpp -iquote"." -D__Z88DK -D__SPECTRUM -DSPECTRUM -D__SPECTRUM__ -D__Z80 -DZ80  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/Users/garypearman/z88dk/lib/config//../..//include"    "src/screen.c" "/tmp/tmpApFaibVX.i2"
z88dk-zpragma -sccz80 < "/tmp/tmpApFaibVX.i2" > "/tmp/tmpApFaibVX.i"
sccz80  -ext=opt -mz80 -standard-escape-chars   "/tmp/tmpApFaibVX.i" -o "/tmp/tmpApFaibVX.opt"
z88dk-copt  "/Users/garypearman/z88dk/lib/config//../..//lib/z80rules.9" < "/tmp/tmpApFaibVX.opt" > "/tmp/tmpApFaibVX.op1"
z88dk-copt  "/Users/garypearman/z88dk/lib/config//../..//lib/z80rules.2" < "/tmp/tmpApFaibVX.op1" > "/tmp/tmpApFaibVX.opt"
z88dk-copt  "/Users/garypearman/z88dk/lib/config//../..//lib/z80rules.1" < "/tmp/tmpApFaibVX.opt" > "/tmp/tmpApFaibVX.asm"
cat "/tmp/tmpApFaibVX.asm" >> "/tmp/tmpApFaibVX.tmp"
cat "/tmp/tmpApFaibVX.tmp" >> "/tmp/tmpApFaibVX.asm"
z80asm -I"/Users/garypearman/z88dk/lib/config//../..//lib/target/zx/def" -D__SCCZ80  -s  -mz80 -I"/Users/garypearman/z88dk/lib/config//../..//lib" "/tmp/tmpApFaibVX.asm"

PROCESSING src/keyboard.c
z88dk-ucpp -iquote"." -D__Z88DK -D__SPECTRUM -DSPECTRUM -D__SPECTRUM__ -D__Z80 -DZ80  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/Users/garypearman/z88dk/lib/config//../..//include"    "src/keyboard.c" "/tmp/tmp0uGrVQi5.i2"
z88dk-zpragma -sccz80 < "/tmp/tmp0uGrVQi5.i2" > "/tmp/tmp0uGrVQi5.i"
sccz80  -ext=opt -mz80 -standard-escape-chars   "/tmp/tmp0uGrVQi5.i" -o "/tmp/tmp0uGrVQi5.opt"
z88dk-copt  "/Users/garypearman/z88dk/lib/config//../..//lib/z80rules.9" < "/tmp/tmp0uGrVQi5.opt" > "/tmp/tmp0uGrVQi5.op1"
z88dk-copt  "/Users/garypearman/z88dk/lib/config//../..//lib/z80rules.2" < "/tmp/tmp0uGrVQi5.op1" > "/tmp/tmp0uGrVQi5.opt"
z88dk-copt  "/Users/garypearman/z88dk/lib/config//../..//lib/z80rules.1" < "/tmp/tmp0uGrVQi5.opt" > "/tmp/tmp0uGrVQi5.asm"
cat "/tmp/tmp0uGrVQi5.asm" >> "/tmp/tmp0uGrVQi5.tmp"
cat "/tmp/tmp0uGrVQi5.tmp" >> "/tmp/tmp0uGrVQi5.asm"
z80asm -I"/Users/garypearman/z88dk/lib/config//../..//lib/target/zx/def" -D__SCCZ80  -s  -mz80 -I"/Users/garypearman/z88dk/lib/config//../..//lib" "/tmp/tmp0uGrVQi5.asm"

PROCESSING src/isr.c
z88dk-ucpp -iquote"." -D__Z88DK -D__SPECTRUM -DSPECTRUM -D__SPECTRUM__ -D__Z80 -DZ80  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/Users/garypearman/z88dk/lib/config//../..//include"    "src/isr.c" "/tmp/tmpkyIs8NOU.i2"
z88dk-zpragma -sccz80 < "/tmp/tmpkyIs8NOU.i2" > "/tmp/tmpkyIs8NOU.i"
sccz80  -ext=opt -mz80 -standard-escape-chars   "/tmp/tmpkyIs8NOU.i" -o "/tmp/tmpkyIs8NOU.opt"
sccz80:"src/isr.c" L:43 Warning:Implicit definition of function 'mousePointer_DrawLoop' it will return an int. Prototype it explicitly if this is not what you want. [-Wimplicit-function-definition]
z88dk-copt  "/Users/garypearman/z88dk/lib/config//../..//lib/z80rules.9" < "/tmp/tmpkyIs8NOU.opt" > "/tmp/tmpkyIs8NOU.op1"
z88dk-copt  "/Users/garypearman/z88dk/lib/config//../..//lib/z80rules.2" < "/tmp/tmpkyIs8NOU.op1" > "/tmp/tmpkyIs8NOU.opt"
z88dk-copt  "/Users/garypearman/z88dk/lib/config//../..//lib/z80rules.1" < "/tmp/tmpkyIs8NOU.opt" > "/tmp/tmpkyIs8NOU.asm"
cat "/tmp/tmpkyIs8NOU.asm" >> "/tmp/tmpkyIs8NOU.tmp"
cat "/tmp/tmpkyIs8NOU.tmp" >> "/tmp/tmpkyIs8NOU.asm"
z80asm -I"/Users/garypearman/z88dk/lib/config//../..//lib/target/zx/def" -D__SCCZ80  -s  -mz80 -I"/Users/garypearman/z88dk/lib/config//../..//lib" "/tmp/tmpkyIs8NOU.asm"

PROCESSING src/main.c
z88dk-ucpp -iquote"." -D__Z88DK -D__SPECTRUM -DSPECTRUM -D__SPECTRUM__ -D__Z80 -DZ80  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/Users/garypearman/z88dk/lib/config//../..//include"    "src/main.c" "/tmp/tmph2qR8jxm.i2"
```

suborb, 2019-11-15:
That looks like the preprocessor is tripping over something in main.c - can you share that file (and any headers it depends on).

You might be able to get going again by editing lib/config/zx.cfg and inserting the following line:

    CPP z88dk-zcpp

This will switch back to the old preprocessor which isn't quite as capable so could fail to parse any preprocessor directives.

gazsp, 2019-11-15:
Trying this on a single source file gives the following:

```
> zcc +zx -O3 -no-cleanup -vn -lndos -lim2 -create-app test.c
Error: cannot read file 'im2.lib'
```

Maybe this is the issue but isn't being reported for some reason on the previous project build?

zx70, 2019-11-15:
please try first with the simpler examples available in z88dk/examples.
e.g.
  zcc +zx -O3 -lndos  -create-app -Dspritesize=16 dstar.c 

I'm not 100% sure but I think the im2 funtions in z88dk are not in a separate lib, where did you get the code you are trying to build?

gazsp, 2019-11-15:
The code built fine with a previous version of Z88DK - everything broke after I upgraded to the latest version. 

Removing -lim2 fixed the issue - weird there was no error message initially :-/

closed 2019-11-15 by gazsp

zx70, 2019-11-15:
So it could be that an improvement is now able to spot the error.
Did I understand correctly ?

gazsp, 2019-11-15:
It is a bit odd that the error message isn't shown in all instances. The docs also still reference the need for -lim2 (which apparently now isn't needed)

https://github.com/z88dk/z88dk/wiki/interrupts

zx70, 2019-11-15:
it was probably kept aside in the past.   I'll update the docs and remove it.

zx70, 2019-11-15:
ok I think the mistery is solved, there is still a makefile able to create im2.lib.   If you had the file in the lib folder (got from past nightly builds or kept since forever) the linker had no problems dealing with redundant funtion names by keeping the last one.