z88dk/z88dk #1332: appmake - scz180 - generating app binaries

feilipu, 2019-11-17:
I'm working on the scz180 platform in the new library, and specifically on connecting the romwbw hbios library to connect to diskio for file system access.

hbios requires that its buffers be above `0x8000` because it uses a banked memory model, and the common RAM area is above `0x8000`.

Unimportant Q 1. - I can set the data origin to `0x8000` to make sure all data is in common RAM, but is there a simple more granular way to get just specific identified buffers to be in the common space?

Rhetorical Q 2. Perhaps the BSS buffers can start at `0x8000` and these can be the disk buffers?

To keep things simple I have set the data section origin as `0x8000` in the [crt defaults](https://www.z88dk.org/wiki/doku.php?id=libnew:target_embedded&s[]=crt), together with selecting rom model 2. This is all good, but the issue I have is that path to generate an executable binary becomes complex, multi-step.

`NOTE 2: ”-create-app” causes the toolchain to generate “test.bin” automatically. The manual steps are shown for pedagogical reasons.`

Well, following the manual pedagogical steps works fine.

Important Q 3. - I'm finding that adding `-create-app` to the command line doesn't create a `test.bin` file from `test_CODE.bin` and friends with compressed rom model. Only `test.ihx` is produced. Is that expected?

suborb, 2019-11-17:
Isn’t that expected with the appmake options you’ve got: +hex and +rom —ihex

I think +glue is what you want.

feilipu, 2019-11-17:
It is just that the wiki suggests that `test.bin` will be generated automatically, alongside `test.ihx`, when `-create-app` is added to the command line, generically.

The `+glue` option is not the default. Perhaps the question is  “how to set the correct default behaviour, to get a `test.bin`?”

suborb, 2019-11-19:
It's weird, but for newlib I don't think there is an obvious way if the DATA section doesn't follow on from the _CODE section - where's the wiki page that suggests that it will? 

All of the sub-commands use the internal function `fopen_bin()`, this creates a temporary file with the concatenated _CODE, _DATA files and then passes a handle back to the utility to process as it wants.

I think you may have to write a simple `+noop` sub-command that opens and writes to the .bin file.

feilipu, 2019-11-19:
The `test.bin` result is mentioned here in the [Embedded section of the wiki](https://www.z88dk.org/wiki/doku.php?id=libnew:target_embedded#understanding_the_output_files_generated_by_the_compiler), where the “form final binary” is discussed. The wiki is clear, in Note 2, that the steps to generate the `test.bin` final form are only provided for completeness, and the `-create-app` option actually shortcuts the process to provide the final result, along with intermediate results too.

It is not a big issue for me now that I’ve decided that the `0x8000` or higher HBIOS buffers can be in the BSS section and the DATA section follows on directly from CODE. But, never the less, I’m not seeing a `test.bin` file being generated across any model, so I’m probably doing it wrong. It would be good to know what is correct.

feilipu, 2019-11-24:
__TO DO__ `+noop` or adding `test.bin` output to another relevant option.

closed 2019-11-24 by feilipu

suborb, 2019-11-24:
Sorry. I lost track of this one. I’m guessing t there’s been a few changes to the z80asm output since the wiki page was written. 

I think the rename output is fairly close to what you want though it does capitalise the file name as well.