z88dk/z88dk #1333: Can't compile z88dk package from github

dikdom, 2019-11-18:
Hello,
cloned the source (git clone https://github.com/z88dk/z88dk.git),
installed the required packages (sudo apt install dos2unix libboost-all-dev texinfo texi2html libxml2-dev subversion bison flex zlib1g-dev)
set the environment and executed the build.sh:
[...]
make[1]: Entering directory '/home/sanyi/devel/z88dk/src/sccz80'
cc -g -O2 -MMD -Wall -I../../ext/uthash/src/ -g -pedantic -std=gnu99   -c -o callfunc.o callfunc.c
In file included from ccdefs.h:20,
                 from callfunc.c:9:
define.h:10:11: fatal error: uthash.h: No such file or directory
  #include "uthash.h"
           ^~~~~~~~~~
compilation terminated.
make[1]: *** [<builtin>: callfunc.o] Error 1
make[1]: Leaving directory '/home/sanyi/devel/z88dk/src/sccz80'
make: *** [Makefile:99: bin/sccz80] Error 2

real    0m11.052s
user    0m9.901s
sys     0m0.964s

What do I do wrong?
(The package from the sourceforge and the latest nightly build is building... I would like to contribute with a new platform, but I can't without building project..)

suborb, 2019-11-18:
I think you've forgotten to clone with submodules!

You can add them with: `git submodule update --init --recursive`

dikdom, 2019-11-18:
Erm... 
I didn't forget. I wasn't aware at all I needed to do that.(okay, okay...) 
Now it compiles properly. Thanks for the advice!! 

But I have one more. Is there any doc about what is the process if I implement a new platform (Videoton TVC) and I would like it to be in the main branch? (gradually ...)

suborb, 2019-11-18:
Ah. I see the problem, the installation page assumes you start of with a tarball. The submodule comment is in the README in the repo. I shall update.

Videoton TVC is similar to the Enterprise isn't it? There's already an existing +enterprise port though it does need a bit of attention.

In terms of adding a new platform, there are some docs about the place for newlib on the old wiki: https://www.z88dk.org/wiki/doku.php?id=temp:front Sometimes examples are better though, and @feilipu just added a new target in this PR: https://github.com/z88dk/z88dk/pull/1315

However, adding a new target (especially one that may be related to an existing target), is generally easier in the classic library - it looks like the last general machine added was the Bandai Supervision 8000 in this PR: https://github.com/z88dk/z88dk/pull/1114

The general steps I follow are:

1. Figure out how to get software into an emulator and update appmake to make it work.
2. Get text mode printing working - these days I tend to skip a ROM call and hit the hardware directly and get the VT52 console working
3. Get ROM based console input running
4. Get graphical printing working - allowing use of custom fonts/UDGs
5. Get the graphics routines working
6. Get the PSG up and running
7. Get the joystick ports up and running
8. Get the INKEY$ style keyboard reading running

dikdom, 2019-11-18:
Thanks for the reply!
Actually I already managed to compile dead simple test my file by 'zcc +tvc test.c' (included stdio.h, integer defined, incremented and printf called).
I mainly followed the enterprise as an example though I haven't tested the compiled raw binary yet, but I guess it should be ok (not finished, only fputc implemented).
I planned to implement fputc, fgetc, fputs, joy, some graphical routines at the first iteration, along with the appmake. Later on CART image creation, custom HW support, etc... But that is the far future.
Do I have to merge my commits directly to the master branch (when everything is OK. Is there a test fw for the libraries?) or some review process is defined first (i don't know, maybe on a branch first)? This approach is not clrear to me, this is the first time I contribute to a github project.. 

I'm sorry for asking so much.

suborb, 2019-11-18:
So you need to fork the project (i.e. have your own repository), make your changes and then make a PR back into this repository.

Then either @zx70 or I will review, check that it works and merge. After that you can make more changes in your repository, create another PR and the cycle will repeat.

I'm looking forward to a new target - it's been a few months since one.

closed 2019-11-18 by dikdom