z88dk/z88dk #1342: Local variable usage in inline assembly

dikdom, 2019-11-22:
Sorry my noob question..
how can I access in an inlined asm file a externally declared variable?
Do I have to play around with the stack pointer also? 
Is there any doc about accessing C variables (local, global) in inline asm?

feilipu, 2019-11-23:
To answer the question, the naming is simply mangled by adding an underscore to the assembly version. Conversely, if you want to reach a label or function defined in assembly from C, make sure it has an underscore in assembly, and then it will be identifiable as the same thing without the underscore from C. The only exception to this is using sccz80 where library functions do not need a pre-pended underscore.

More generally, since it is quite easy to move fluidly between C and assembly, it is more robust NOT to inline assembly into C files. Typically, you'd write an assembly function, and then call it with C calling conventions from another C function.

These different functions are kept in different files, because SDCC can't separate functions in one file into separate sections, which minimises the linked code pulled into the resulting binary. That is why the z88dk library is written with so many tiny files. Keeping one function per file improves linking granularity.

Phillip

dikdom, 2019-11-23:
Thanks!
Ok, I understand it is unwise to use it in production, I just wanted to test something fast. 
Finally I managed to compile it, but the variable (`char str[40]`) had to be defined global, before the method and then the inline assembly saw it properly (as `_str`).
But what if I would like to use in the inline assembly the method's locally defined (in the method, on the stack) variable? I tried the _ and the __ prefix also..

feilipu, 2019-11-24:
In your command line, use the `--list` to generate a `mycfile.c.lis` file. Then you can see what assembly sccz80 or sdcc generated from your code. You'll see that both of them use the stack for local variables, but in different ways. For example sdcc uses either `ix` or `hl` depending on how many local variables it needs to use. It doesn't always create a stack frame with `ix`.

There are some [examples here](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/math/float/math32/c/asm), if you like to look quickly. These are compiled [from this C](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/math/float/math32/c) by sdcc.

If you're careful, you can calculate which variable is in which stack location, by knowing whether it was a passed parameter, or was declared within the function. But, the result it will not be portable or even robust to changes in version of the compilers.

This also gets complicated quickly as both sccz80 and sdcc have different methods of passing parameters on  the stack depending on how many parameters are being passed, and what types they are. (sccz80 passes `char` in two bytes. sdcc passes `char` in one byte.) as one example.

Using inlined assembly can also break the peephole optimisation of the generated code. For sccz80 this is even the case for the `--c-code-in-asm` option, so it is best not to use this option when compiling finished work.

In short. Look at the result of your code, see where the variable is located, and go from there.

dikdom, 2019-11-26:
Ok, thanks! I thought I could refer the local variables by their (somehow mangled) names also in the inlined assembly. I get it, I will avoid using for longer implementations.

closed 2019-11-26 by dikdom