z88dk/z88dk #1343: About fgetc_cons

dikdom, 2019-11-23:
Hello,
I would like to ask, in case of a new platform implementation how should the fgetc_cons function behave?
* should it return immediately if a key is pressed
* or should it allow the platform editor to work and when the ENTER (RETURN, etc..) is pressed it should return the characters one-by-one?

In case of the first one: where/how is the editor implemented? (Delete, crsr movement, insert, etc...)?
In case the latter: is there a way to call an 'init' routine before calling the getc_cons for the first time? Is there a way to define the end char?

suborb, 2019-11-23:
`fgetc_cons()` is supposed to return the keypress without echoing the character to the screen.

For line editing you can override `fgets_cons()`, to override place it before the `@stdio/stdio.lst` in your `.lst` file.

To get something run before first call, put the code into the section `code_crt_init`, there's an example here (where we setup some screen mode variables): https://github.com/z88dk/z88dk/blob/master/libsrc/target/multi8/gencon/generic_console.asm#L302 - note that the code placed in this section isn't a subroutine so there shouldn't be `ret`

Regarding the namings, some targets have switchable implementations, for output they main ones are:

* fputc_cons_native() -> Uses the BIOS to output to the screen
* fputc_cons_generic() -> Uses the VT52 to output to the screen (usually addresses hardware directly)

Likewise with input we've got:

* fgetc_cons_native() -> Uses the BIOS to read a character from the keyboard
* fgetc_cons_inkey() -> Uses the `in_*` to read a character from the keyboard.

The `in_*` functions directly read the hardware ports and through the `in*` interface support multiple key presses.

dikdom, 2019-11-24:
Ok, thanks. But still, I have some more questions I couldn't find in the wiki, sorry...
1) Does it mean that if I implement `fgets_cons()`, then `scanf()` will also use that method instead of calling `fgetc_cons()`.
2) In case I implement `fgetc_cons_native()` will it be used automagically (just as `fgetc_cons()` ) or do I need to define a pragma/switch in the target's config? Is it simly just a matter of naming? 
3) When I implemented `fgetc_cons()` for the keypresses, there was no cursor (not even an underscore) but the main problem was that I was not able to delete, because the string was returned to `scanf()` on pressing 'DEL' (0x08). Do I miss anything? Is scanf() is used to input some formatted data, fgets() - unsafely - is used to input a line, right? Do both of these methods end up at using getc_cons(), or getc_cons_native()?

suborb, 2019-11-24:
No worries. I think the first thing to know is that stdio in classic is character orientated, there’s no way to automatically switch the terminal into line mode. Hence, scanf will consume characters. 

This was a bit of a conscious choice to reduce memory usage, in newlib the terminal driver does support line mode, but stdio doesn’t support files  

So, to get line editing capabilities a user needs to do an fgets and then a sscanf

Fgetc_cons_native gets used automatically. The switch happens in lib/crt/classic/crt_runtime_selection.asm

Without you implementing fgets_cons the default implementation will end up calling your implementation of fgetc_cons. The delete key is handled - at the top of your crt0 file you define what the delete key is.

dikdom, 2019-11-26:
Let's assume I like the z88dk implementation of gets() and I would like to call it to input a string from console. How can I call it? How can I create a FILE of STDIN? (I guess gets(str, 20 STDIN) wouldn't work - or what should I include??). Or shall I call directly the fgets_cons()? (It works, but seems a bit odd calling that method directly.)

Secondly let's assume I would like to override the behaviour of gets_cons as you suggested earlier. I implemented it in .c, managed to include in the Makefile tree and it is compiled to .o but it seems that not my function is used in favor of the built-in. How can I replace the original implementation? Simply edit the original file and check the #defines ?

zx70, 2019-11-26:
I suggest to ignore the fgets_cons hint, it is misleading in your case.
Your first choice to support a new target is between the classic and the new lib.
File support can be added in a second moment,  don't worry for the file streams they'll be partially available automatically and can be extended in the future. 
The make process can be used to build a whole library block,  otherwise you could prepare a light partial lib which the linker will incorporate replacing the functions where overlapped.

dikdom, 2019-12-01:
Ok, thanks for the clarifications!

closed 2019-12-01 by dikdom