z88dk/z88dk #1344: sccz80 - const char* const

feilipu, 2019-11-24:
sccz80 throws multiple errors  for this code sample:
```c
#define FF_STR_VOLUME_ID    1
#define FF_VOLUME_STRS     "MD1","MD0","IDE0","SD0","SD1"
static const char* const VolumeStr[FF_VOLUMES] = {FF_VOLUME_STRS};    /* Pre-defined volume ID */
```

dikdom, 2019-11-24:
char* is one string, char** is the array of strings, or do I miss something?

suborb, 2019-11-24:
I thought I’d fixed handling the second const at one point in time. I’ll take a look.

feilipu, 2019-11-24:
@dikdom 
> char* is one string, char** is the array of strings, or do I miss something?

`static const char* const VolumeStr[FF_VOLUMES];`

Means `VolumeStr` is a constant pointer to arrays of characters (strings) that are constant and `static` in this situation means visible only within the file.

I read somewhere the way to understand these declarations. You just read from right to left and it becomes much easier to understand.

`static const char* VolumeStr[FF_VOLUMES];`

Means `VolumeStr` is a pointer to arrays of characters (strings) that are constant and `static` in this situation means visible only within the file.

`static char* const VolumeStr[FF_VOLUMES];`

Means `VolumeStr` is a constant pointer to arrays of characters (strings) that aren't constant and `static` in this situation means visible only within the file.

I do hope this is correct... :man_shrugging:

closed 2019-11-24 by suborb