z88dk/z88dk #1348: .def file

dikdom, 2019-12-09:
How to make the constants defined in the file `lib/target/XXXX/def/XXXX.def` visible in the C library source files? What would be a good approach?

feilipu, 2019-12-10:
The typical process is to put the configurations into a .m4 file in the config directory, for newlib.

For the [`rc2014` target for example](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/target/rc2014/config), configurations are defined for the various pieces of hardware, and using .m4 language they are made available in _private.inc and/or _public.inc and/or [`config_rc2014.h`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/config_rc2014.h).

The configuration definitions are selected from the [master config file](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/config.m4), for each target.

zx70, 2019-12-12:
In the classic library there's no immediate way to bind a .def file to, say an header file. 
I'm sometimes exaggerating in preparing those files,  but the initial idea is always to keep them under the hood,  to be included by machine code portions only. 
Then,  usually someone begins asking how to do low level, target specific  things in C ..

feilipu, 2020-01-22:
Ok?

closed 2020-01-22 by feilipu