z88dk/z88dk #1349: [newlib] - hbios as a target?

feilipu, 2019-12-10:
The [ROMWBW](https://github.com/wwarthen/RomWBW) [HBIOS](https://github.com/wwarthen/RomWBW/tree/master/Source/HBIOS/Config) supports a number of different machines, including SBC 1/2, Zeta 1/2, N8, Mark IV, RC2014, SCZ180, and Easy Z80.

I've copied the HBIOS support into both RC2014 and SCZ180 platforms, but I'm thinking it might be sensible to make a separate target for HBIOS, similar to the CP/M target directory, but limited to the "hardware abstraction layer" drivers only. This would allow support for more machines, without a proliferation of new targets.

At this stage, I don't think it would be more than a collection of HAL drivers, because there's no need to have a CCP or BDOS equivalent in the target. This could include unified access to the disk, serial, video interfaces via the [described API](https://github.com/wwarthen/RomWBW/blob/master/Doc/RomWBW%20Architecture.pdf).

Is this OK to do?

suborb, 2019-12-12:
I think that's a sensible decision. Any (optional) platform specific code can be contained within a -subtype= as we do for classic CP/M.

I really should try and get hold of a modern z80 system - they're looking like a lot of fun.

feilipu, 2019-12-13:
I'm recommending the [SC130 ](https://www.tindie.com/products/tindiescx/sc130-romwbw-cpm-computer-kit/)as a "good, fast, cheap" thing. I still can't understand how Stephen can sell it so inexpensively. The USB and SD Card adapters come included too!

suborb, 2019-12-13:
Thanks for the recommendation. That’s sorted my post Christmas self-present!

feilipu, 2019-12-23:
Started with #1363.

feilipu, 2019-12-24:
With #1363, there are now three ways to get the a similar result for ROMWBW HBIOS targets.
```
zcc +hbios
zcc +rc2014 -subtype=hbios
zcc +scz180 -subtype=hbios
```
Generally things are working as they should, but there are still some oddities to fix.
 [ ] - returning from `fprintf()` zsdcc pops 4 bytes sometimes, and 2 bytes other times.
 [ ] - `fscanf()` works for some, but not for others. Also true for `-subtype=cpm` on these targets, btw.

Something for the quiet time before New Year.

feilipu, 2020-01-01:
Done a bit of benchmarking of disk driver hardware.
[Pretty happy with these results.](https://groups.google.com/d/msg/retro-comp/cfZZrpIfkCY/CF6Ni1-4AwAJ).

Still need to integrate video and audio, and sort out `fscanf()`, but otherwise this is done.

suborb, 2020-01-02:
Well, I think that's the first thing I've screwed up this year, it probably won't be the last.

I really should just stick to software, I've never been good friends with soldering irons!

feilipu, 2020-01-02:
I'd offer to build it but shipping from Aus is too expensive. I think SC (who's UK based) would be happy to help. Shall I enquire?

feilipu, 2020-01-06:
I'm going to leave this now, with serial and disk layers working properly.

User code -> [FatFS](https://github.com/feilipu/z88dk-libraries/tree/master/ff) -> [diskio_hbios](https://github.com/feilipu/z88dk-libraries/tree/master/diskio_hbios) -> [HBIOS](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/arch/hbios.h)

I don't have access to video hardware for testing (motivation), and it will be a while before I can get back to the audio hardware.

feilipu, 2020-01-13:
Will open a new Issue for video / audio etc.

Otherwise, I've been using this target now with the RC2014 and SCZ180 quite a bit, and it is a nice way to work. Using the gbgmon and uploading Hex is very fast. Provides probably the largest clear memory space (`0x0100` - `0xF000` or stretch to `0xFFE0`) of any target.

closed 2020-01-13 by feilipu