z88dk/z88dk #1351: zxn support in zsdcc

spth, 2019-12-14:
Upsteam sdcc has finally started work on ZX Next support in the compiler.

However, with a shortage of developers there, it would be good if the stuff outside the compiler itself could be done by others:

https://sourceforge.net/p/sdcc/feature-requests/656/
https://sourceforge.net/p/sdcc/feature-requests/655/

spth, 2019-12-14:
For reference, the main SDCC feature request for ZX Next support is:
https://sourceforge.net/p/sdcc/feature-requests/573/

feilipu, 2020-06-29:
The z80n has now been included in the basis for the zsdcc code, using sdcc v4.0.2 r11690.

spth, 2020-06-29:
The main problem with current z80n support in sdcc is that push #nn gets misassembled by sdas (sdas doesn't really support mixed-endian data).

This is not a big problem for C users (as generation of push #nn is disabled until the assembler bug is fixed, so there now are no regression test failures), but those writing inline asm or otherwise using sdas should be aware of it.

closed 2022-12-16 by feilipu