z88dk/z88dk #1353: Compile error .asm file (z88dk v1.99C, windows 10)

akalulyadurov, 2019-12-19:
hi,
please fix that bug.

### test2.asm
org 8000h
start:	ld a, 0b00000011
		and 0b00000010
		out ($80), a
		ret

c:\WORK\test1>**zcc +rc2014 -subtype=acia -clib=sdcc_iy -SO3 -v -m --list --max-allocs-per-node200000 test1.asm -o test1 -create-app**

PROCESSING test1.asm
type "test1.asm" >> "C:\Users\webde\AppData\Local\Temp\zccF4022.asm"
type "C:\Users\webde\AppData\Local\Temp\zccF4022.asm" >> "C:\Users\webde\AppData\Local\Temp\zccF4022.tmp"
type "C:\Users\webde\AppData\Local\Temp\zccF4022.tmp" >> "C:\Users\webde\AppData\Local\Temp\zccF4022.asm"
z80asm -D__SDCC_IY -I"c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014" -D__SDCC --list   -s  -mz80 -I"c:\z88dk\lib\config\..\..\/lib" -I"c:\WORK\test1"  "C:\Users\webde\AppData\Local\Temp\zccF4022.asm"

PROCESSING c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014\rc2014_crt.asm.m4
m4 -I "c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014" -I "c:\z88dk\lib\config\..\..\/src/m4"  --define=__STARTUP=0  < "c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014\rc2014_crt.asm.m4" > "C:\Users\webde\AppData\Local\Temp\zccF4023"
type "C:\Users\webde\AppData\Local\Temp\zccF4023" >> "c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014\rc2014_crt.asm"
type "c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014\rc2014_crt.asm" >> "C:\Users\webde\AppData\Local\Temp\zccF4023.asm"
type "C:\Users\webde\AppData\Local\Temp\zccF4023.asm" >> "C:\Users\webde\AppData\Local\Temp\zccF4023.tmp"
type "C:\Users\webde\AppData\Local\Temp\zccF4023.tmp" >> "C:\Users\webde\AppData\Local\Temp\zccF4023.asm"
WILL ACT AS CRT

GENERATING OUTPUT
type "C:\Users\webde\AppData\Local\Temp\zccF4022.lis" >> "test1.asm.lis"

PROCESSING CRT
z80asm -D__SDCC_IY -I"c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014" -D__SDCC --list   -s  -mz80 -I"c:\z88dk\lib\config\..\..\/lib" -I"c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014"  "C:\Users\webde\AppData\Local\Temp\zccF4023.asm"

z80asm -mz80 -b -d  -o"test1" -m -s -L.  -L"c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\lib\sdcc_iy"   -D__SDCC_IY -I"c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014" -D__SDCC --list  -irc2014    "C:\Users\webde\AppData\Local\Temp\zccF4023.o" "C:\Users\webde\AppData\Local\Temp\zccF4022.o"
Error at file 'c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014\rc2014_crt.asm' line 1492: symbol '_main' not defined
type "C:\Users\webde\AppData\Local\Temp\zccF4023.lis" >> "test1.lis"
Errors in source file c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014\rc2014_crt.asm:
Error at file 'c:\z88dk\lib\config\..\..\\libsrc\_DEVELOPMENT\target\rc2014\rc2014_crt.asm' line 1492: symbol '_main' not defined
                   ^ ----       call _main                ; hl = return status

feilipu, 2019-12-20:
The z88dk assumes that you're working mainly in C, so as standard there needs to be [a `_main` entry point somewhere in your program](https://en.cppreference.com/w/cpp/language/main_function).

Either as `int main (void)` if you're using C.
Or in assembly as `_main:`

Try that.

aralbrec, 2019-12-20:
Yes the entry point has to be a public label "_main"

PUBLIC _main
_main:

The "public" exports the symbol to the global scope so that the crt calling _main can see it.

Besides that, you should remove the org as well.  The memory map is supplied by z88dk automatically unless you override that.  Instead, you place your code into sections which are part of the memory map.  The toolset in z88dk is more like commercial tooling in that it has a linker that places things into memory according to a memory map.

So:

```
SECTION code_user

PUBLIC _main

_main:

  ld a, 0b00000011
  and 0b00000010
  out ($80), a
  ret
```

Section "code_user" is a good generic place to put code.

akalulyadurov, 2019-12-20:
Thanks a lot,

after zcc compiling I have got a very big BIN file 6.5kb size. 

z80asm gives a small size BIN file. But how to convert BIN to Intel HEX?

**test1.asm**
section data_user
section code_user
PUBLIC _main
_main:	ld a, 0b00000011
		and 0b00000010
		out ($80), a
		ret

C:\TASKS\z80>**z80asm --cpu=z80  -v -m -l -g -b -r8000H test1.asm**

feilipu, 2019-12-20:
`zcc` is putting your short code fragment into a proper C environment with the code located in the right place. That is what you should do normally.
```sh
zcc +rc2014 -subtype=acia -clib=sdcc_iy -SO3 -v -m --list --max-allocs-per-node200000 test1.asm -o test1 -create-app
```
that will generate a `test1.ihx` file which is the HEX code you need to upload to the RC2014 using [hexload](https://github.com/RC2014Z80/RC2014/tree/master/BASIC-Programs/hexload).

`-subtype=acia` assumes that you are running a version of Grant Searle MSBASIC, with the acia serial device. The origin is assumed to be `0x9000`, as it needs to be for a 32kB RC2014 Classic or Mini with 32kB of RAM.

If you're using CP/M then you want the `-subtype=cpm` option, and you can use PIP to load the HEX, and LOAD to convert it to a `test1.com` file.

If you're using ROMWBW then you can use the `-subtype=hbios` option, and use the dbgmon `M` and `L` instructions to load the HEX file and then `R 100` to run it with origin `0x0100`.

akalulyadurov, 2019-12-20:
And how to get Intel HEX file from BIN file obtained after z80asm compilation?

feilipu, 2019-12-20:
`appmake +hex --help` will get you going.

akalulyadurov, 2019-12-21:
**appmake +hex --org $8000** doesn't change Origin of the binary and gives the same HEX as **appmake +hex**.

**test1.asm:**
org $8000
start:	ld a, $02
		out ($00), a
		ret

**test1.hex:**
:050000003E02D300C91F
:00000001FF

c:\WORK\test1>**z80asm -v --cpu=z80 -m -l -g -b -s -o=test1.bin test1.asm**
Library 'z80asm-z80-.lib' not found
Library '/usr/local/share/z88dk/lib/z80asm-z80-.lib' not found
Reading library 'c:/z88dk/lib/z80asm-z80-.lib'
Predefined constant: __CPU_Z80__ = $0001
Assembling 'test1.asm' to 'test1.o'
Reading 'test1.asm' = 'test1.asm'
Writing object file 'test1.o'
Module 'test1' size: 5 bytes

Code size: 5 bytes ($8000 to $8004)
Creating binary 'test1.bin'

c:\WORK\test1>**appmake +hex -b test1.bin -o test1.hex --org $8000**

c:\WORK\test1>**appmake +hex -b test1.bin -o test1.hex**
Warning: could not get the code ORG, ORG defaults to 0

feilipu, 2019-12-21:
Perhaps it would be easiest if you described your platform, and what outcome you are looking for?

[I don't even know what half the options you put in the `z80asm` string do. I've never had a need to use them, and I never use `z80asm` directly].

akalulyadurov, 2019-12-21:
I have RC2014 mini, Small Computer Monitor v1.0, Windows 10, Tera Term.
I am trying to load HEX to RAM at $8000 Origin.

**test1.asm**
org $8000
start: ld a, $02
out ($00), a
ret

**test1.hex** generated by z80asm
:05**0000**003E02D300C91F
:00000001FF

I successfully loaded HEX I have got from asm80.com to my RC2014 mini at $8000 Origin. Here is what I desire:
**test1.z80.hex** generated by asm80.com
:05**8000**003E02D300C99F
:00000001FF

feilipu, 2019-12-21:
Ok. Thanks. That makes it easier.

The RC2014 SCM and the RC2014 MSBASIC both have the same `RST08` `RST10` based serial interfaces. This means that you have two easy choices. There are many other ways too. But these fall to hand.

1. Use the `zcc +rc2014 -subtype=basic` command. This by default will create HEX that loads from `0x9000`. (This is because MSBASIC uses from `0x8000` to`0x8FFF` for its own needs. SCM uses `0xFC00` to `0xFFFF`). Load your HEX and run it with `G 9000`. [Check here](https://groups.google.com/d/msg/rc2014-z80/cGiPYtc4O8o/wlhbBJkvBAAJ) for more info.

Just noting there will be a stack issue with this option, because the program stack will overwrite the SCM monitor RAM, but all that will happen is you will reboot after a program rather than returning to monitor.

2. Roll your own.
```
zcc +z80 --no-crt -v -m --list -Ca--filler=0xFF @test1.lst -o test1
appmake +glue -b test1 --ihex --pad --filler 0xFF --recsize 24 --clean
```
You can find an example `test1.lst` file [here](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/rc2014_NascomBasic32k_HexLoadr/nascom32k.lst),
that loads an example `test1_map.asm` file [here](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/rc2014_map.asm),
followed by example assembly `test1.asm` [here](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/rc2014_NascomBasic32k_HexLoadr/int32k.asm).

Obviously, the files are not called `test1.xxx`, but I'm sure you can extrapolate for what you need.

aralbrec, 2019-12-21:
That large 6.5K bin for your small asm program is very suspicious.  The subtype you are using will put in drivers for the serial interface but those are not particularly big either unless felipu (who wrote them) knows better.

The org is under control of the memory map, not by an org directive given in asm files.  Providing a separate org will lead to the linker thinking the asm file is not part of the output code and it will generate a small separate binary just for that file.  The org can be changed within the memory map being supplied using pragmas, which are special instructions to the crt, and can control a lot of things.  These pragmas are usually stored in a separate file that is read by zcc when given "-pragma-include:XXXX".  The pragma to control the org address is "CRT_ORG_CODE"; the default is sensible already for the rc2014 target unless you need more control over placement.

There is some detailed information for the generic z80 target here (and this link goes to the list of pragmas):

https://www.z88dk.org/wiki/doku.php?id=libnew:target_embedded#crt_configuration

This applies to all targets in the newlib but each target, eg the rc2014, is tailored to produce files that are sensible to the platform.

feilipu, 2020-01-22:
Wrote a [z88dk wiki entry at RC2014 repository](https://github.com/RC2014Z80/RC2014/wiki/Using-Z88DK). Perhaps that helps too.

closed 2020-01-22 by feilipu