z88dk/z88dk #1355: [newlib] optional / alternate library inclusions

feilipu, 2019-12-21:
Across the subtypes associated with some of my targets, I'm bumping into the need to have various parts of the newlib library omitted at different times.

For example, when a `-subtype=rom` compile is done the serial drivers are required, but when a `-subtype=app` compile is done the serial drivers are already available and can be provided by a `.def` file.

This question arises now because (if I'm not mistaken) there's been a behaviour change recently, where a `.def` file `PUBLIC`/`defc` CAN'T override a newlib function linkage. I'm seeing this (below) were before where there wasn't a problem (the linker managed to prefer the `.def` file), and duplicate library code is being attached where it shouldn't be necessary.
```
Error at module 'cpm22_asm': symbol 'asci0RxLock' already defined in module 'yabios_def_asm'
```
But this is also an issue for further targets like RC2014 and SCZ180, where there are different and multiple IDE / SD drivers available to chose from, for example. It would be good to choose the correct disk library functions (providing the same API) to higher layer FAT file systems code.

I'm not sure how to achieve this result. Any thoughts or pointing at other examples would be helplful.
Thanks :+1:

aralbrec, 2019-12-21:
Actually this is critical behaviour.  Symbols defined in user files should be found before libraries are searched, including defc symbols.

feilipu, 2019-12-21:
@aralbrec Ok, if that described behaviour is critical, I'm calling bug. But, there's been quite a bit of change so bisecting the issue is hard (I've spent nearly a week trying to work out where I'd screwed up).

Previously, I could build "[yabios](https://github.com/feilipu/yaz180/tree/master/yabios)" and then just flip directory to "[cpm22](https://github.com/feilipu/yaz180/tree/master/cpm22)", which is an app, and just build. But now I see errors (as above) from things already defined in the `yabios_def.asm` file.

Building yabios is works fine. Just prepared a new one.
```
zcc +yaz180 -subtype=rom -SO3 -v -m -clib=sdcc_iy -llib/yaz180/ff -llib/yaz180/time --math32_z180 --max-allocs-per-node400000 @yabios.lst -gpf yabios.rex -o yabios -create-app
```
But the CP/M build... doesn't build.
```
zcc +yaz180 -subtype=app --no-crt -m --list @cpm22.lst -o cpm22; appmake +glue -b cpm22 --ihex --clean
```
The definitions in `yabios_def.asm` aren't overriding, and additionally library labels (functions and variables 'defb`)  are being added to the build although they already exist.

feilipu, 2019-12-21:
At the risk of conflating two issues now, the (new ???) need to have multiple library options, depending on subtype selected, is also important.

Specifically, for the FatFS file system, there may be multiple methods of reading a physical disk sector, but the FAT layer doesn't care about this. Ideally there is only 1 FAT code, but depending on whether the SD CISO SPI, CF 8-bit, or IDE 16-bit hardware interfaces are required, they can be selected.

I guess the easiest switch would be via subtype, but then different versions of target library per subtype are needed.

suborb, 2019-12-21:
As a datapoint, classic overriding of functions using libraries still seems to be working - eg adding -lndos to a +test compile. 

@feilipu cant you just do the same as for classic +cpm - have the ide access functions in a separate library but everything else is shared and in a single library?

feilipu, 2019-12-21:
@suborb classic +cpm has its file system in assembly in z88dk. Unfortunately newlib doesn't, so I'm using a FAT file system in C, which is in a [separate library](https://github.com/feilipu/z88dk-libraries/tree/master/ff) and is installed using `z88dk-lib`.

The way I'm working around the issue now is to rebuild for the scz180 platform referencing a [HBIOS shim](https://github.com/feilipu/z88dk-libraries/blob/master/ff/source/diskio_hbios.c) or to include the whole [SD card](https://github.com/feilipu/z88dk-libraries/blob/master/ff/source/diskio_sd.c) driver depending on what I'm working on. But, this is clunky (and pretty slow).

Your suggestion to split the lower level hardware abstraction off into a separate library is worth pursuing. :+1: But, then I'd still have to watch which hardware library was installed, as the names of the provided disk API will clash.

feilipu, 2019-12-21:
The symptom was caused by the regular expression parser not adding some symbols to its output.

Based on [`+ asm_asci[01]_`](https://github.com/feilipu/yaz180/blob/master/yabios/yabios.rex#L28) it should have added `asm_asci1_flush_Rx` but it only added [`asm_asci1_flush_Rx_di`](https://github.com/feilipu/yaz180/blob/master/yabios/yabios.def#L143) which calls the `asm_asci1_flush_Rx` function.

Because `asm_asci1_flush_Rx` was called directly in the CP/M build, but was not added in the `.def` file, additional code to provide the function was added, which included the `asci0RxLock` symbol.

The work around was simply not to call `asm_asci1_flush_Rx`, and the issues go away.

However still not explained are
1. why a matching `PUBLIC` symbol present in the map file was not exported by the regular expression processor, and
2. why the private `.def` file couldn't override the library symbol.

feilipu, 2019-12-22:
The only mystery now is why this (my) issue only revealed itself now, after working for 18 months.

closed 2019-12-22 by feilipu