z88dk/z88dk #1357: Double cursor when entering text

DarwinNE, 2019-12-21:
Hi to all,
  I noticed that on several platforms, a "double cursor" is shown when text is entered following a call to fgets_cons. Here is a minimal code that should shows this behaviour:

~~~~
#include<stdio.h>
#define SIZE 80

int  main(void)
{
    char buffer[SIZE];
    fgets_cons((buffer),(SIZE));

    printf("You typed: %s\n",buffer);
    return 0;
}
~~~~

If I compile for the MSX target with

~~~~
zcc +msx -create-app bug.c -lndos -zorg=32768 -o bug
~~~~

Instead of getting a single cursor, I get both an underline and a block. Here is an example of what I get running the previous code and typing "testing".

[image]

The input works fine, though, but the double cursor is somewhat inelegant :-)
I have been reported that the same behaviour occours on the CP/M target using fgets with stdin, I think it comes from the same origin. Here is the double cursor after ">" on one of my programs running on a Kaypro CP/M computer:

[image]

Cheers,
D.

suborb, 2019-12-21:
Yes, I know the cause of that one.

The `fgets()` implements a simple software cursor (the `_`) so if the firmware implements one as well you end up with both of them.

If you switch to the cross platform VT52 engine (`--generic-console`) then the firmware cursor is  disabled since we bypass firmware printing and hit the hardware directly. As a result, you only end up with one cursor.

However, the problem persists if you use firmware routines so I'll add a compile time flag to disable printing of the software cursor to work around that one. I'm in the middle of adding another DDR machine but I'll pick it up next, so probably a couple of days.

DarwinNE, 2019-12-21:
Thank you very much!

suborb, 2019-12-21:
I've added a compile time pragma to disable the soft cursor in `fgets_cons()` - just add the following option:

DarwinNE, 2019-12-23:
Thank you, I'll check as soon as possible!

DarwinNE, 2019-12-26:
Hello,
  I tested my program, made a git pull and compiled from sources z88dk (v15793-0eaa84cff-20191225), but I am still getting the double cursor.

I compiled the program with the following command:
~~~~
zcc  +msx -create-app -subtype=msxdos -zorg=32768 -pragma-define:CLIB_DISABLE_FGETS_CURSOR=1  bug.c -o bug
~~~~

Cheers,
D.

suborb, 2019-12-26:
Hi Davide, I've just checked the behaviour on the MSX and I can't see the double cursor - did you rebuild msx_clib.lib? To do so go into libsrc and then enter the following commands:

    rm msx_clib.lib
    make msx_clib.lib
    cp msx_clib.lib ../lib/clibs/

DarwinNE, 2019-12-27:
Hi,
  I checked again and indeed it was my fault! I was using an outdated file.
Everything is working perfectly now, thank you very much!
D.

suborb, 2019-12-27:
No worries. I’m glad to hear that it’s all good now.

closed 2019-12-27 by suborb