z88dk/z88dk #1361: [all targets] Compilation of all Z88DK targets in Cross Chase fail

Fabrizio-Caruso, 2019-12-22:
@suborb 
All targets now fail with this error
```
sccz80:"chase/main.c" L:322 Warning:Implicit definition of function 'constructItems' it will return an int. Prototype it explicitly if this is not what you want. [-Wimplicit-function-definition]
Error at file 'chase/main.c' line 223: symbol '_constructItems' not defined
Errors in source file c:/z88dk/\lib\target\abc80\classic\abc80_crt0.asm:
Error at file 'chase/main.c' line 223: symbol '_constructItems' not defined
                   ^ ----               #endif
make: *** [Makefile:2225: abc80_light] Error 1
```

suborb, 2019-12-23:
That should be good now, apologies for that - the first "const" bit of constructItems vanished!

Fabrizio-Caruso, 2019-12-23:
I think my large project can provide a nice non-reg test. It gets broken very often and it is constantly cross checked with other compilers including GCC.

Fabrizio-Caruso, 2019-12-23:
It is still broken as of now. I suppose I need to check once the next snapshot is released (on the 24th of Dec)

feilipu, 2020-01-22:
@Fabrizio-Caruso still broken?

Fabrizio-Caruso, 2020-02-23:
@feilipu , it may be broken for other reasons.
For example attache is now broken and fails with:
Error: cannot read file 'gfxattache.lib'
make: *** [Makefile:3242: attache] Error 1

Fabrizio-Caruso, 2020-02-23:
@feilipu @suborb I am closing this but I have opened two target specific issues:
#1411,  #1412 which are most probably new problems.

Cross Chase once again provides a good non-regression test.
This is why I think it that the compilation of a slimmed down version of Cross Chase should be run as a non-reg test.

closed 2020-02-23 by Fabrizio-Caruso