z88dk/z88dk #1370: Text scrolling for console programs on Sam Coupe

DarwinNE, 2019-12-29:
Hello,
  I am currently working on a text adventure game that I compile to many different platforms.
I tried to use z88dk to obtain a version for Sam Coupe, but I currently do not know much about this target. One of my beta testers reported that the text is printed correctly, but once it's done a screen full of text, the computer starts not showing any new text properly.

Since I have successfully compiled the game for many other platforms using z88dk, I suspect there may be an issue with the Sam Coupe library.

I am currently not equipped to replicate the bug, but here is a code that may help showing the problem:
~~~~
#include<stdio.h>
#define SIZE 80

int  main(void)
{
    char buffer[SIZE];
    int i;

    while(1) {
        fgets(buffer,SIZE,stdin);    
    
        for(i=0;i<5;++i)
            fputs(buffer, stdout);
    }

    return 0;
}
~~~~
Can someone please try to test it on the Sam Coupe (I used `zcc +sam -clib=ansi`)?

Kind regards,
D.

suborb, 2019-12-29:
For some reason when scrolling the Sam port swaps colour to black paper, white ink and then restores them to whatever they were beforehand.

Unfortunately, if you don't set any colours then the "beforehand" colours are presumed to be black ink, black paper!

I've not spent any time on the Sam at all, as a target it's fairly minimal so it'll possibly get a visit form me at some point next year to make it a bit more feature rich.

closed 2019-12-29 by suborb

suborb, 2020-01-02:
Sorry @DarwinNE - I didn’t see your edit. 

Colour can be changed using codes 16,n and 17,n as per the Spectrum ROM

DarwinNE, 2020-01-02:
Thank you very much!

commit a1b1e0ab (2019-12-29): Fix #1370: (sam/ansi) Set the default colours correctly