z88dk/z88dk #1371: [newlib] - allow bss origin to float

feilipu, 2020-01-01:
Working with the `hbios` targets (including RC2014 and SCZ180 subtargets) it is necessary to keep buffers used by the `hbios` api above the `0x8000` page fold. Memory above `0x8000` is not swapped when the api is called, and can be used by the api to read and write data.

An easy option to maintain this restriction to set the bss section origin to `0x8000`, and ensure that relevant `hbios` buffers are in bss or on the heap (above bss), using the C pragma.
`#pragma output CRT_ORG_BSS = 0x8000`

Where `__DATA_END_tail` is lower than `0x8000` there is no issue. But, when `__DATA_END_tail` grows above `__BSS_head`, currently `__BSS_head` remains set to the defined value, causing data and bss sections to collide.

The enhancement would be to allow the `__BSS_head` to attach itself (or float) to `__DATA_END_tail` yet not sink below a defined value. Perhaps a negative value could be used?

`#pragma output CRT_ORG_BSS = -0x8000` =  ORG of the bss section will append to the data section in memory, or originate at nomiated location, whichever is lower in memory.

This logic and request could equally be extended to the data section origin, to ensure that all data and bss remained above a waterline, set by `CRT_ORG_DATA = -0x8000` for example.

suborb, 2020-01-01:
I think this needs some z80asm support to allow this to happen.

There's probably a code work around that can be done without z80asm changes - eg picking up the address of the buffer bases from a variable that's assigned on startup using a comparison of __BSS_end and 0x8000

feilipu, 2020-01-01:
Thinking about this further, _imho_ this floating origin behaviour should actually be the default behaviour. There is no advantage to create a new configuration option.

If the `__DATA_head` is configured to fall between `__CODE_head` and `__CODE_END_tail`, or
if the `__BSS_head` is configured to fall between `__DATA_head` and `__DATA_END_tail`,
then I should probably be warned that I've created an impending "car crash".

Ideally, the linker could fix the collision by floating the origins upwards as needed (recognising page alignment).

But in in any case I'd like a _"WARNING - You are being very stupid"_ to appear, please.

feilipu, 2020-01-19:
Resolved this by adding [warning and fence options](https://github.com/z88dk/z88dk/commit/7b5b854c437a2bd4089829e433e94cf88b11c2e9) to `appmake +rom` and `appmake +hex`.

And put them into the [configs](https://github.com/z88dk/z88dk/commit/b2ee0909561209a6a90e8ab516d053cb0ff6259d).
Close enough.

closed 2020-01-19 by feilipu