z88dk/z88dk #1382: newlib / sccz80 - cpm input

feilipu, 2020-01-15:
Been going through the [EXAMPLES](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES), checking they're good on "my" targets. Found & fixed some [issues](https://github.com/z88dk/z88dk/commit/59925304ed96133e071f376486110802fc63d111) with empire.c and sdcc/math32 around use of atof. Using the short test program below.

Moved on to check with sccz80 and found it works not, neither with math32 nor math48.

```sh
 zcc +cpm -v -SO3 -clib=new --list -m test.c -o test --math32 -create-app
 zcc +cpm -v -SO3 -clib=new --list -m test.c -o test -lm -create-app
```
There seems to be an issue around the INPUT routine.
The format string is printed, rather than being used as a specifier.

```c
#include <stdlib.h>
#include <stdio.h>
#include <string.h>
#include <stdarg.h>
#include <math.h>
#include <input.h>

#pragma printf = "%s %d %u %f"

void INPUT(char *p, char *fmt, ...);

char buf[64];  // input buffer

float H2;

void main(void)
{
    puts("\nHello World\n");

    do
    {
        INPUT("How many bushels do you wish to sell? ", "%30[0-9.-]s", buf);
        H2 = atof(buf);

        buf[sizeof(buf)-1] = 0;

        fprintf(stdout, "String %s, Float %f\n", buf, H2);
    }
    while ((H2 != 0.0));
}

void INPUT(char *p, char *fmt, ...)
{
   va_list arg;
   va_start(arg, fmt);
   
   printf("%s", p);
   in_wait_nokey();
   
   fflush(stdin);
   vscanf(fmt, arg);
}
```
Produces when `1234` entered.
```sh
A>test

Hello World

%30[0-9.-]s1234
String , Float 0.000000

A>
```
Should this work?

suborb, 2020-01-15:
I think you've hit the pain of varargs with __smallc semantics.See #330 - Alvin got something to work.

feilipu, 2020-01-15:
Ah. Ok. Something new to digest over a few nights.

I guess then the [empire.c](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/empire.c) program (as it stands today) won't work, and so I can close this issue

closed 2020-01-15 by feilipu

aralbrec, 2020-04-14:
The workaround for sccz80 works like this:

```
int error(char *fmt, ...)
{
   unsigned char *p;
   
   va_list v;
   va_start(v, fmt);

#ifdef __SCCZ80
   vsnprintf(ebuf, sizeof(ebuf), va_ptr(v,char *), v);
#else
   vsnprintf(ebuf, sizeof(ebuf), fmt, v);
#endif

   for (p = ebuf; p = strchr(p, '\n'); )
      *p = '\r';
   
   ebuf[strlen(ebuf) - 1] += 0x80;  
   return (int)ebuf;
}
```

imneme, 2021-01-03:
A better work around is to compile with `-set-r2l-by-default` and `-D__Z88DK_R2L_CALLING_CONVENTION` or just the latter and mark varargs functions with `__stdc`.  See #1664.