z88dk/z88dk #1387: Debugging C source code

obiwanjacobi, 2020-01-30:
I am researching if (and how) it is possible to generate debugging symbol information that would allow me to relate addresses to C source code. The map file that seems to be the current best source of debugging symbol information is mainly focused at .asm files - by the looks of it.

Is my perception correct? 
Is there a way to compile my own debugger symbol file by using several existing files?
Do you have any (other) suggestions?

Context: I have a Z80 in-circuit-emulator that does debugging and I am trying to determine how feasible it is to write a debugger extension for VS-code for it based on z88dk sources.

suborb, 2020-02-02:
Mapping assembler lines to code lines can be done with -gcline with sccz80. 

However it doesn't help with local variable locations (i.e. the stack relative position).

At this point in time, the z80asm object file format isn't rich enough to contain the debugging we need. I'm not sure if @pauloscustodio has thoughts in this area.

pauloscustodio, 2020-02-02:
No thoughts and open to suggestions. As far as I know there is no standard object file with debugging info for z80 systems, and each system/emulator implements its own. Is there a particular setup we should target for? We might need some new pragmas to pass information from C to the assembler and linker, but this is the easy part.

suborb, 2020-02-02:
I've not used it, but I feel like it's easiest to support the `cdb` format from sdcc. These are text files which could be processed to match whatever format is needed.

There's a definition here: https://github.com/roybaer/sdcc-wiki/wiki/CDB-File-Format

SDCC generates a `.adb` file containing type information and augments the output assembler with C records for line numbers and X records for end of function.

I think asxxx then appends L records for the addresses of labels and C/X records

pjshumphreys, 2020-06-01:
If you use the cli options -m, -S and --c-code-in-asm you can get an assembly file with the original c source code embedded in comments and a map file to relate symbols to addresses in memory.

If you then use an assembly level debugger open next to the asm file in your editor you can follow along to know which c source code lines are getting executed.

--c-code-in-asm does break optimisations, but the results are probably good enough for debugging purposes.

obiwanjacobi, 2020-06-01:
(not looked at the details of those switches you mention) 
Is there any address data in there?

pjshumphreys, 2020-06-01:
In the .map file there is, together with the label in the .asm file each address maps to

obiwanjacobi, 2020-06-02:
Only having addresses for the labels makes a lousy debugger...

aralbrec, 2020-06-04:
An attempt to integrate the list files into an emulator for debugging purposes can be seen here:
https://www.youtube.com/watch?v=TmpHy_x2Fko

suborb, 2020-06-28:
#1390 was raised to address this. I've started work on .adb support but I'm getting sidetracked by other things...

closed 2020-06-28 by suborb