z88dk/z88dk #1389: Freeze a nightly and release?

suborb, 2020-02-02:
It's been a year and there's been some great new features that have been added:

- The mostly ieee maths library
- 8080 support
- Gameboy support
- sdcc 4.0.0

It's relatively quiet at the moment so now seems like a good time to release - any objections?

The other question is version number. I'm minded just to go to 2.0 and then step up to 2.1 - we could be on 1.99X for a while otherwise.

Xalior, 2020-02-02:
You make a release announcement that goes “2.0, C compiler, gameboy” you’ll get a LOT more of a pick up by retro coverage sites than a point release, too ;-)

feilipu, 2020-02-02:
I’ve nothing in the backlog, (except minor possible fixes for a LUT Module for RC2014) and everything seems to be working as expected. So a good time afaik.

Quite like SDCC team’s stable big number plan. So 2.0.0 would be right.

SpectrumNext is shipping soon, so there needs to be a release for that, too. People will want an anchor release to install. 

I would like to see the big library reorganisation later, and that might break some things for a while. And then also better handling of 3rd party libraries.

feilipu, 2020-02-02:
Yes, added GameBoy, SpectrumNext, RomWBW, RC2014 support. Covering all the big devices in the news these days.

pauloscustodio, 2020-02-02:
Ok for a 2.0 release.

suborb, 2020-02-02:
Cool, I think the changelog is up-to-date so I'll take a cut and promote tomorrow evening.

I'm not the biggest (if at all) external forum user so if someone else wants to take on the announcements outside of the z88dk.org forum then that would be appreciated.

Xalior, 2020-02-02:
> 
> I'm not the biggest (if at all) external forum user so if someone else wants to take on the announcements outside of the z88dk.org forum then that would be appreciated.
> 
> 

I’m one of the admins of the ZX Z88DK facebook user group, so happy to reproduce your announce there :-D

-Dx

feilipu, 2020-02-02:
I can cover RC2014 and RetroBrew. 

Doesn't @aralbrec have SpectrumNext contacts? They're shipping in just a few days, so getting in final pre-shipping announcement would be good. Otherwise I can put a note on their forum.

suborb, 2020-02-03:
It is done, hopefully it works :)

closed 2020-02-03 by suborb

suborb, 2020-02-12:
And it didn't quite work....bugs:

- #1399 - zsdcpp was incorrectly named (on osx)
- #1394 - (Classic) +zx didn't handle the enter key
- #1397 - the compiler ldexp() implementation was buggy
- (classic) +svi generic console was broken

feilipu, 2020-02-13:
... and I [mistyped a "i"](https://github.com/z88dk/z88dk/commit/785b948c1dd5d9f99a0892c7f9d6ac851ec77073). 😢

feilipu, 2020-02-13:
Related, I pushed a bunch of (hopefully related) issues into the v2.1 project.
The two things are:

- library reorganisation
- function trimming

Not sure if these are the best topics to have, but is a new "question" issue appropriate