z88dk/z88dk #1390: Debugging symbols support

suborb, 2020-02-03:
Breaking out from #1387 here are the tasks that will be needed:

- [ ] Get zsdcc to generate .adb when in c1mode
- [ ] zcc to treat .adb as an output file and concatenate them just before linking
- [ ] Add .adb file generation to sccz80
- [ ] Transform sdcc inline annotations to C_LINE pragmas
- [ ] Make copt skip over C_LINE pragmas
- [ ] z80asm to append link information to .adb file
- [ ] Update ticks to parse .adb file and allow source level debugging

spth, 2020-06-08:
I guess in the long term, ELF/DWARF is the future. The hc08, s08 and stm8 backends of SDCC already support ELF output with DWARF debug information.

suborb, 2021-01-17:
Prompted by @desertkun I moved this along a bit in the `feature/sccz80_cdb` branch, I've run out of the steam at the moment.

I've been focussing on sccz80 since previously that wasn't generating anything.

- `--debug` turns on adb information generation for sccz80 and sdcc
- zcc copies the .cdb file generated by sdcc
- For sccz80 adb information is written into symbols so ends up in the .map file with a `__CDBINFO__`. There's URL encoding of the CDB line happening here, with `_` as the replacement for `%`
- Ticks has been updated to read C_LINE information and allows a breakpoint to be set on a C line, there's also some status stuff to print where we are in a file in the prompt.
- Ticks can read source code and show it in the debugger
- I also added watchpoints to ticks because they're useful

The CDB information is extracted from the .map, decoded and then I've just run out of steam for the moment. What needs to be done (I think):

- [ ] C_LINE needs to be extended with scope/level information
- [ ] ticks needs to parse the cdb information and do something useful with it (tempted to lift the sdcc/debugger code TBH)
- [ ] Some scripts to convert between various formats (for emulicious taking the .map and generating an sdcc style cdb file is appropriate), other emulators read .map files, so adding the sdcc cdb into the .map makes sense there

Although adding support to ticks is fairly pointless in itself, it's hoped that it will provide some copy/paste/modify code for other emulators to pick up.

However, as I said I've run out of steam at the moment so I'm going to park this for a bit.

desertkun, 2021-01-17:
@suborb thank you for your efforts.

desertkun, 2021-02-17:
Some progress.

After playing around with @ladmanj 's [binutils-gdb](https://github.com/b-s-a/binutils-gdb) I was able to produce BFD library along with headers, which I used a simple project like this:
```c
    asection *bss = bfd_make_section_old_way (b, ".bss");
    asection *text = bfd_make_section_old_way (b, ".text");

    asymbol *new = bfd_make_empty_symbol (b);
    new->name = "_test";
    new->section = bss;
    new->flags = BSF_GLOBAL;
    new->value = 0x8E5E;

    asymbol *new2 = bfd_make_empty_symbol (b);
    new2->name = "_main";
    new2->section = text;
    new2->flags = BSF_FUNCTION;
    new2->value = 0x8109;

    asymbol *ptrs[3] = {new, new2, 0};

    bfd_set_symtab (b, ptrs, 2);
```

which produced a simple `hello.coff` file, which objdump understands: `z80-coff-objdump -t -x hello.coff`:
```
hello.coff:     file format coff-z80
hello.coff
architecture: z80, flags 0x00000030:
HAS_SYMS, HAS_LOCALS
start address 0x00000000

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .bss          00000000  00000000  00000000  00000000  2**0
                  ALLOC, LOAD
  1 .text         00000000  00000000  00000000  00000000  2**0
                  ALLOC, LOA
SYMBOL TABLE:
[  0](sec  2)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00008109 _main
[  1](sec  1)(fl 0x00)(ty   0)(scl   2) (nx 0) 0x00008e5e _test

```

Then I loaded up `z80-coff-gdb` to see what it has to say about it:
```
(gdb) symbol-file hello.coff
Reading symbols from hello.coff...
(No debugging symbols found in hello.coff)
```

Which is progress, but not much. I'd appreciate any hints, particularly on what section(s) does gdb want? Can we make do with coff because apparently elf for z80 is not supported. Many questions, not much of answers.

spth, 2021-02-17:
ELF support for Z80 has been merged into GNU binutils a year ago: https://sourceware.org/bugzilla/show_bug.cgi?id=25224

So we now only need to add it in SDCC. Shouldn't be hard: Just do what was done for STM8 before; patches for SDCC are welcome, and could be accepted after the 4.1.0 release.

desertkun, 2021-02-17:
Too bad I am stuck with sccz80 :(

desertkun, 2021-02-20:
@kfazz made some effort here https://github.com/kfazz/sdcc/tree/z180_dwarf
How can we steal this?

spth, 2021-02-21:
Though I wonder why that uses a --dwarf option. Other SDCC ports just automatically emit DWARF when --out-fmt-elf is combined with --debug.

desertkun, 2021-08-17:
Hey @suborb so I finally gave this `--debug` this a go with sccz80 and I am getting these:
```
zcc +zx -m -debug -create-app -o example main.o other.o -subtype=bin
Error at module 'other_c': symbol '__CDBINFO__S_3aG_24VALUE_5f_30_24_30_5f_30_24_30_28_7b_30_7d_29_2cE_2c_30_2c_30' already defined in module 'main_c'
Error at module 'other_c': symbol '__CDBINFO__S_3aG_24VALUE_5f_31_24_30_5f_30_24_30_28_7b_30_7d_29_2cE_2c_30_2c_30' already defined in module 'main_c'
Errors in source file /usr/local/share/z88dk/lib/config/../..//lib/target/zx/classic/spec_crt0.asm:
Error at module 'other_c': symbol '__CDBINFO__S_3aG_24VALUE_5f_30_24_30_5f_30_24_30_28_7b_30_7d_29_2cE_2c_30_2c_30' already defined in module 'main_c'
Error at module 'other_c': symbol '__CDBINFO__S_3aG_24VALUE_5f_31_24_30_5f_30_24_30_28_7b_30_7d_29_2cE_2c_30_2c_30' already defined in module 'main_c'
make: *** [example__.bin] Error 1
```
This appear to be happening when linker tries to merge symbols from object files. See my [complete minimum example](https://github.com/desertkun/z88dk-debug-error-minimum-example). I have included `header.h` in two object file, so this
```
enum the_enum
{
	VALUE_0 = 0,
	VALUE_1 = 1,
};
```
ends up defined in both object files, and linker doesn't like that at all when it tries to merge them. Maybe these symbols should be marked as weak or something? Or maybe linker should learn to ignore redefinitions to same exact value (`1` in that case)?

desertkun, 2021-08-19:
I have a PR with the fix for the issue #1837 

Actually, I went even further and implementing a whole "toolchain": 

- a [gdbserver support](https://github.com/desertkun/fuse/commit/fbd804e6900f52ee388445ce9da01da75fcbf554) (that's up to the spec mostly) for Fuse (only mac for now)
- A gdb-like client with help of ticks. Ticks has good debugger, but very limited emulator. So I've refactored it to support multiple backends (so original ticks still works as expected) and implementing a dbg client backend, that would connect to Fuse and debug it there. So, having this, we could continue improving ticks' debugger while simultaneously also improving gdb debuger.

I hope to submit a new PR with these changes maybe next week, but I need this merged first.
This is a major effort and I have never done cross-repo merges before. So I kinda hope for your support while I am going to also try to convince Fuse's maintainers to accept the gdbserver portion.