z88dk/z88dk #1391: libsrc reorganisation

suborb, 2020-02-03:
It's time to pull newlib out of the attic and reorganise libsrc so that code can be shared in both directions.

Whilst doing this, I think it would be good to allow build parallelisation to work on libsrc/ builds - newlib in particular repeatedly clears out object files.

Beyond that I've not got any particularly strong thoughts on how to do this merge.

feilipu, 2020-02-03:
Related to #4.

suborb, 2020-04-20:
Sorry, a bit late coming back to this. 

Parallel builds and quicker incremental builds. This is one of the things I've been slowly sneaking into classic - mainly because my main machine is mid 2012 MacBook which isn't exactly a speed demon (this year would be the year to upgrade since decent keyboards are apparently inbound, but Catalina breaks far too many things I use)

Most of classic now builds object files into obj/target or obj/cpu as appropriate which massively speeds up rebuilding libraries. 

z80asm has a way of specifying the output directory rather than placing the object file along side, so as soon as config_private is resolved (Eg by adding a -I path) then parallel builds for newlib should be trivially possible since the output files won't be entangled.

feilipu, 2023-07-29:
Some thoughts on moving the newlib targets over to classic (at least for things I care about). To try to capture what can be preserved and what is unnecessary.

The maths libraries are not true targets, so they should be OK to move over provided that libraries are built with the same compiler options, and for same CPU, as available today. AFAIK, most of this is already replicated in classic anyway.

- am9511
- m (math48)
- math16
- math32

The two generic targets I don't know much about, except they make good templates to build further targets.

- z80
- z180

These three targets are quite extensive and complicated, because of their hardware capabilities and options (modes). IDK how hard it would be to replicate them in classic.

- sms
- zx
- zxn

These ones I can comment on...

- cpm - a vanilla CP/M serial only target that exclusively calls BDOS functions. There's nothing worth keeping here.

- hbios - offers low level access to [RomWBW](https://github.com/wwarthen/RomWBW) HBIOS for terminal and time (clock) functions. Use of the [HBIOS API](https://github.com/wwarthen/RomWBW/blob/master/Source/HBIOS/API.txt) means that programs can be run directly in the RomWBW monitor. I find this useful for FreeRTOS testing, but it could help to speed up a game that otherwise might have to go via CP/M BDOS to HBIOS. It could be extended to use the VDA API available in HBIOS, as further work. RomWBW has a large installed base.

- rc2014 - for the [RC2014](https://rc2014.co.uk/) target. The RC2014 is a modular solution, and there are a number of conflicting hardware options and differing access to hardare, either directly or via CP/M or HBIOS API, which [require rebuilding following configuration changes](https://github.com/z88dk/z88dk/issues/1568). This relates to serial (ACIA or SIO) ports, hard drive (16-bit IDE or 8-bit CF), memory (with/out 64kb shadow RAM). Development has stalled but it would also be possible to support a range of optional video, clock, and other Modules. It already has both z80 and 8085 CPU support, with the 8085 support built in classic. Currently in production, with a large installed base.

- scz180 - relates to the [Small Computer Central](https://smallcomputercentral.com/kits/) SC126, SC130, and SC131 products, with a z180 CPU and using the CSIO for access to SD Cards. The CSIO drivers are the only differential to the generic z180 target. However, there is also access to the time (clock) HBIOS API, since these products are all running RomWBW. Currently in production, with a large installed base.

- yaz180 - for the [YAZ180](https://github.com/feilipu/yaz180) with z180 CPU with fixed peripherals. Of interest to preserve is support for the PCA9665 device and I2C drivers using burst mode and interrupts, and 16-bit IDE and diskio drivers. It also has access to the am9511 using an interrupt queue mechanism. Otherwise it is similar to the z180 target with no direct video support. There were only a few handfuls of these devices built, so further support is pretty much on me as no one else would be overly interested.