z88dk/z88dk #1394: __STDIO_CRLF related ?

zx70, 2020-02-06:
I tried to build and run OTHELLO.C for the Spectrum, but it does not seem to recognize the ENTER code, so the player can't input its move.   I'm sure I bumped in this problem in the past but I couldn't find the original issue.
What am I missing ?

suborb, 2020-02-06:
Oh no. 

Yes. It’s a new thing. The move into target/ removed the swapping define - I had to rebuild target/zx/stdio for ts2068 since it’s shared and wanted to avoid an rm. 

I’ll need to check the other targets as well.

zx70, 2020-02-06:
No panic, at least it looks I finally found an effective way to check your changes  :)
While I'm typing:  I like very much the way you sorted out the graphics stuff !  Perhaps we'll discover something wrong but it is much neater now.

commit 5d0b95de (2020-02-06): Fix #1394: Ensure \n and \r are swapped for ZX + CPM
closed 2020-02-06 by suborb