z88dk/z88dk #1396: doc issues

dikdom, 2020-02-07:
Hello,
probably it's just me, but I'm missing from the supported machine list the 'Videoton TV Computer' from here: https://sourceforge.net/projects/z88dk/files/z88dk/2.0
On the other hand in the gihub the main page contains a non-working reference to the latest release (under Installation) and it still refers to 1.99C, while 2.0 is already released (see sourceforge link).

feilipu, 2020-02-07:
The Videotron TVC is mentioned in the announcement, if I'm reading correctly.

But there is a lot of documentation dating back pre-github still too. Not ideal.

dikdom, 2020-02-07:
Sorry, you're absolutely right about TVC on sourceforge, for some reason I missed that. But the broken link and invalid latest version in Github is ... fixed!

Thanks and have a great day! :)

closed 2020-02-07 by dikdom