z88dk/z88dk #1397: BUG crept in GFX lib?

zx70, 2020-02-09:
As forecasted by Dom I think I found something.
I initially thought to do some extra testing, but I'm busy this weekend, and I prefer to begin with this quick alert. 
At the moment all I know is that the GLOBE example fails drawing correctly few lines, I suppose it has to do with coordinates' ranges or sign/direction in relative position drawings.

zx70, 2020-02-09:
tested on zx spectrum, sccz80/classic only

suborb, 2020-02-09:
I've had a look and this isn't a graphics bug, but is a floating point bug.

It seems that this change/idea: 6701a84cd65c714bbad7bc9b38e8c50acc837ddc causes problems.

It also fails to work (at all) with sccz80+math32/math48 so something is wrong there too.

closed 2020-02-09 by suborb