z88dk/z88dk #1403: BUG: classic zx Spectrum

zx70, 2020-02-15:
I noticed 2 new problems with the default settings of the zx Spectrum target:
Getk() not returning code 13 when ENTER is pressed.
Text does not scroll, causing memory overflows.

suborb, 2020-02-15:
I think it's intended that getk() return 10 for ENTER?

There was an issue with scrolling where a `[` would be printed after the scroll, I've just fixed that though.

Have you got a little test program for scrolling? It may be that I've missed an edge case...

zx70, 2020-02-15:
I noticed both the problems with the existing examples.
globe or spirograph wait for ascii code 13, which worked in the past? 
the scrolling was failing in few cases too

suborb, 2020-02-16:
It looks like getk() used to return 13 until last year in change d732feba42a21b87fd1e6ae1fd1dabc568b7dd9c - to avoid confusion I'll change all the 13 to '\n' in the examples.

I am a bit worried about our handling of the cursor keys though: we've got DOWN and ENTER returning the same code.

closed 2020-02-24 by suborb