z88dk/z88dk #1407: psg/wyz.h file not found trying to use new wyz integration from within sdcc_iy

jsmolina, 2020-02-20:
As shown on this pull request
https://github.com/jsmolina/z88dk-tutorial-sp1/pull/54

Trying to use wyz integration, compiler reports file not found:
#include <psg/wyz.h>
`mssound.h:4:21: fatal error: psg/wyz.h: No such file or directory`

OS: MacOSX
Using Z88dk nightly latest

suborb, 2020-02-20:
I've only added it to classic - it looks like you're using newlib?

jsmolina, 2020-02-21:
yep, as I use the splib. Should I do a symlink?

suborb, 2020-02-23:
I’ll have to update newlib to include the files. Hopefully I’ll get a chance this week.

jsmolina, 2020-03-09:
Is it simple to do it? maybe it's worth to try myself and to a pull request.

suborb, 2020-03-09:
Sorry, I ended up being distracted and forgetting to do this despite figuring out what to do!

My basic plan was to setup base in _DEVELOPMENT/sound/ay:

* Put the wyzplay.asm into ay/z80 and the C function wrappers into c/sccz80 and c/sdcc_ix and c/sdcc_iy
* The code in target/zx/psg can then be copied into _DEVELOPMENT/zx/driver/ay
* Create the header in an appropriate place
* Create the various list files and include as appropriate
* Make classic use copies of newlib code rather than having duplicates

suborb, 2020-03-13:
This was merged as #1419 - the header file is `<sound/aywyz.h>`

closed 2020-03-13 by suborb