z88dk/z88dk #1408: z88dk-lib: add non specific (generic) C libraries

feilipu, 2020-02-21:
Related to #389, and #390, the `z88dk-lib` tool installs libraries associated to specific targets. However, there are a number of general purpose libraries that would be broadly useful and could abstracted for use on any / all targets. These include (for example):

 - time - an implementation of time.h aspires to conform with ISO/IEC 9899 (C90).
 - ff - FatFs is a generic FAT/exFAT filesystem module for small embedded systems. The FatFs module is written in compliance with ANSI C (C89) and completely separated from the disk I/O layer.
 - freeRTOS - real-time operating system for microcontrollers. FreeRTOS is built with an emphasis on reliability and ease of use.

As these library examples are built in C, and are maintained upstream, it wouldn't be sensible to extensively patch them to work with z88dk. But (as I'm finding), building them repeatedly for multiple targets doesn't seem worthwhile either.

The library interfaces (system tick, disk I/O layer, etc) will be target specific, and may ideally be built in assembly as part of the target build, or added later as a target specific library.

Ideally, as noted in #389, the `z88dk-lib` tool should be able to handle multiple header files per library. I've not (yet) seen a need for handling multiple library files per library.

suborb, 2020-02-23:
Why do they need to be installed into the z88dk directory? Don't the -I and -L options remove the need?

feilipu, 2020-02-23:
Yes, that's a fair comment.

Today, the only clear benefit is to have the `m4` tool generate 3 different correct header files (clang, sccz80, sdcc). But that is a clear benefit. If I have to generate z88dk specific decorations in the header files, then best not to do it 3 times over.

This means that the C application is still target dependent (as the included file location changes depending on the the target), as is the command line because the linked library location changes depending on target too #390.

suborb, 2020-02-23:
Following the discussion a bit further, what actually is the point of generating different sets of headers for each compiler? - Could we just use the same header file?

I think the m4 stuff came about when a) the sccz80 preprocessor wasn't great, b) sccz80 used a different function decorator form. It does feel like it complicates 3rd party library usage.

I think the key things that m4 changes are:

1. Adding in `__LIB__` for sccz80
2. Adding in `__smallc` for sccz80
3. Adding `preserve_regs` for sdcc
4. Creating callee calls for sccz80 and sdcc

Some thoughts:

1. Not actually needed (but does force separate entry points for sdcc/sccz80 so useful), could `#define` away for sdcc/clang
2. This could be `#define` away for sdcc/clang
3. This is parsed by sccz80, could be `#define` away for clang
4. Could be wrapped in a `#if !clang`

In classic there's `<sys/compiler.h>` to deal with this (on a somewhat less grand scale to my points above) - we may also need to twiddle some of the annotations based on CPU/compiler combination so static generation isn't always going to work. For example, sdcc/gbz80 doesn't support `__z88dk_fastcall`

feilipu, 2020-02-24:
> In classic there's <sys/compiler.h> to deal with this (on a somewhat less grand scale to my points above) - we may also need to twiddle some of the annotations based on CPU/compiler combination so static generation isn't always going to work.

It looks like generating a single z88dk header with correct decorations, together with `sys/compiler.h`, is the right way to go. Is adding a similar file to newlib a possibility?

I'm working through headers for FreeRTOS #1410 at the moment and it is a pita to manage them all. If or when some of these functions become _callee or _fastcall functions I think the complexity gets much worse.

suborb, 2020-02-25:
> It looks like generating a single z88dk header with correct decorations, together with <sys/compiler.h> to define away the unneeded, is the right way to go. Is adding a similar file to newlib a possibility?

I don't see why not - it would simplify things for 3rd party libraries and for us creating the headers.

It's probably easier if a modified m4 process generates the full annotated headers which are then checked in, m4 stuff deleted.

feilipu, 2020-02-26:
Sounds like part of the code reorg #4 #1391 projects. Not a small job.

closed 2022-05-11 by feilipu