z88dk/z88dk #1414: sccz80 - PHASE expects constant expression

feilipu, 2020-03-02:
I'd like to configure a [`PHASE` statement in sccz80](https://github.com/feilipu/z88dk-libraries/blob/master/freertos/source/include/portmacro.h#L180) like I [can do in sdcc](https://github.com/feilipu/z88dk-libraries/blob/master/freertos/source/include/portmacro.h#L353).
But, this code causes an error...

```sh
Error at file './portable/port.c' line 356: symbol 'configISR_ORG' not defined
Error at file './portable/port.c' line 356: expected constant expression
```

feilipu, 2020-03-02:
Also, by the way, the same situation [here](https://github.com/feilipu/z88dk-libraries/blob/master/freertos/source/include/FreeRTOSBoardDefs.h#L65)... where the definition is not accepted by sccz80.
But here the [same definition is accepted here](https://github.com/feilipu/z88dk-libraries/blob/master/freertos/source/include/FreeRTOSBoardDefs.h#L111) by sdcc.

suborb, 2020-03-02:
I'm guessing configISR_ORG is a #define somewhere?

With sccz80 it's embedded in a string, so I think you might need some  stringification magic, eg, something like:

```
"PHASE " ## #configISR_ORG ##"\n" \
```

Or whatever the magical syntax to get stringification and concatenation to work together.

feilipu, 2020-03-03:
Perfect, thanks.

I used these macros.
``` c
/* General purpose stringify macros. */
#define string(a) __string(a)
#define __string(a) #a
```
And the fact that adjacent strings are automatically concatenated ...

``` c
#ifdef __SCCZ80

#define configTICK_RATE_HZ              256
#define configISR_ORG                   0xFB00
#define configISR_IVT                   0xFF06

#define portEXAMPLE_USAGE()   \
    do{                             \
        asm(                        \
            "PHASE "string(configISR_ORG)"  \n" \
            "ld hl,"string(configISR_IVT)" ; an address\n" \
            "ld hl,__CPU_CLOCK/"string(configTICK_RATE_HZ)"/20-1\n" \
            );                      \
    }while(0)

#endif
```
Gets a perfect (if a little clunky) result. :+1:

closed 2020-03-03 by feilipu