z88dk/z88dk #1416: uthash.h and utstring.h missing

jayacotton, 2020-03-06:
after make clean; make

cc -O2 -MMD -Wall -I../../ext/uthash/src/ -g -pedantic -std=gnu99   -c -o callfunc.o callfunc.c
In file included from ccdefs.h:20,
                 from callfunc.c:9:
define.h:10:11: fatal error: uthash.h: No such file or directory
  #include "uthash.h"

jayacotton, 2020-03-06:
I did an aggressive clean up and forced it to pull a new version of sccz80, this seems to fix the problem.

jayacotton, 2020-03-06:
wrong again.

cc -O2 -MMD -Wall -I../../ext/uthash/src/ -g -pedantic -std=gnu99   -c -o callfunc.o callfunc.c
In file included from ccdefs.h:20,
                 from callfunc.c:9:
define.h:10:11: fatal error: uthash.h: No such file or directory
  #include "uthash.h"
           ^~~~~~~~~~
compilation terminated.

The ~/ext/uthash directory is empty.  Am I missing a package ?

jayacotton, 2020-03-06:
Looks like parts of the ~/ext/* content is not delivered with a git pull of z88dk.

closed 2020-03-06 by jayacotton

suborb, 2020-03-08:
Yes. Ext/ is sub modules so you’ll need to pull those down.