z88dk/z88dk #1417: install: missing destination file operand after '{,//usr/local/share/z88dk/}lib/arch/ez80/ez80_rules.1'

jayacotton, 2020-03-06:
I see this during the install process.  

Tried different shells, no joy.

jayacotton, 2020-03-07:
well, looks like install is not working.  What I did was remove all z88dk packages from anywhere
on my machine....  then did a ./build.sh and that seems to fix it.  still not installed, but the
set_environment command seems to do the trick...

closed 2020-03-07 by jayacotton

feilipu, 2020-03-07:
make install doesn’t and won’t work.

Because you will be building libraries regularly, you’d need to keep doing this as root. The build notes do point out that a system install is not supported.

The environment variables are sufficient.

suborb, 2020-03-08:
Let’s say I’ve fought several battles with install over the years. 

I have a hunch that the install lines only work with BSD install since it does work on a Mac. I suspect I’ll have to revisit and make it work properly. 

However since the common use case is to set the environment variables I’ll defer for a while.