z88dk/z88dk #1424: Can't build ZXNext examples

Dovgalyuk, 2020-03-23:
e.g., z88dk/libsrc/_DEVELOPMENT/EXAMPLES/zxn/sprites/mixed-sw-hw/main.c

It seems that the compiler can't find zxn includes. How to specify path to them?

Xalior, 2020-03-23:
This example works for me using the compile line example given in the main.c - have you set up your ZCCCFG as shown on https://github.com/z88dk/z88dk/wiki/installation ?

aralbrec, 2020-03-23:
Yes that is probably the problem.  But there is another problem with that compile line in that the zx next now treats the sna format as a legacy format so when loading an sna, the computer is set up as a 48k or 128k spectrum with all of the next hardware switched off.

Xalior, 2020-03-23:
I was going to open a pull request for that @aralbrec ;-)

Dovgalyuk, 2020-03-23:
$ echo $PATH
......:/home/pasha/ZXNext/z88dk/bin
$ echo $ZCCCFG
/home/pasha/ZXNext/z88dk/lib/config

test.c:
 #include <arch/zxn.h>
 int main()
 {
    return 0;
 }

$ zcc +zxn test.c
Error at file 'test.c' line 207: syntax error
.... and many more errors

aralbrec, 2020-03-23:
I downloaded the latest build for windows and that also runs fine.  Unfortunately I can't check the linux package until tonight.

Do you mind trying:

`which zcc`

to verify the right zcc is running?  If you installed z88dk previously, that older version may be earlier in your path.  Also, there are other projects that use a decade-old version of zcc in their installs that may also be earlier in the path.

Other than that, the next step would be to see those errors.

Can you:

`zcc +zxn -v test.c > foo.txt`

and report the result either download link or cut & paste here?  You may have to redirect stderr to a file too.

Dovgalyuk, 2020-03-24:
which zcc
this told me that I have zcc in /usr/local/bin
It seems that I installed it for some reason, when started playing with zcc.

After removing all z88dk files from /usr/local I'm still get that error messages.
[error.txt](https://github.com/z88dk/z88dk/files/4373304/error.txt)

Dovgalyuk, 2020-03-24:
Here are also the temporaries:

[tmpXXiStVNo.asm.txt](https://github.com/z88dk/z88dk/files/4373979/tmpXXiStVNo.asm.txt)
[tmpXXiStVNo.err.txt](https://github.com/z88dk/z88dk/files/4373980/tmpXXiStVNo.err.txt)
[tmpXXiStVNo.i.txt](https://github.com/z88dk/z88dk/files/4373981/tmpXXiStVNo.i.txt)
[tmpXXiStVNo.i2.txt](https://github.com/z88dk/z88dk/files/4373982/tmpXXiStVNo.i2.txt)
[tmpXXiStVNo.op1.txt](https://github.com/z88dk/z88dk/files/4373983/tmpXXiStVNo.op1.txt)
[tmpXXiStVNo.opt.txt](https://github.com/z88dk/z88dk/files/4373984/tmpXXiStVNo.opt.txt)

suborb, 2020-03-24:
It's failing at the assembler stage, so comparing your .asm.txt with one that I'm generating shows that instead of `GLOBAL [label]` your file as `!extern` - this is the reason why the compilation is failing.

!extern is internally generated by sdcc for global definitions file - as far as I know it shouldn't actually reach an output file.

I've tried on a Mac and Linux - both x86_64 to reproduce this and I can't. What's your host architecture: I may be able to spin up an environment and check it out.

Can you run `zcc +zxn test.c -O0 -a -v` and share the generated ".s" file? - This is the file generated by sdcc with minimal processing so it would be good to see what it's actually generating.

Dovgalyuk, 2020-03-25:
I'm running it on Ubuntu 18.04/64
I built everything from the source code without downloading the package.

Here is the generated *.s:
[test.c.s.txt](https://github.com/z88dk/z88dk/files/4379528/test.c.s.txt)

feilipu, 2020-03-25:
It almost looks as if something's gone wrong with your build of zsdcc.
Could your patching have been off? Any hunks not take where they should have?

What happens if you do this?
```
phillip@SilverQuad:~$ zsdcc -v

ZSDCC IS A MODIFICATION OF SDCC FOR Z88DK
Build: 4.0.0 #11556 (Linux) Mar 14 2020

sdcc website:
https://sourceforge.net/projects/sdcc/

patch details:
https://github.com/z88dk/z88dk/blob/master/src/zsdcc/sdcc-z88dk.patch

published under GNU General Public License (GPL)
phillip@SilverQuad:~$ 
```

Dovgalyuk, 2020-03-25:
Right, there is no information about patching there.

Dovgalyuk, 2020-03-25:
Trying to patch it:

patch -p0 < sdcc-z88dk.patch
can't find file to patch at input line 5
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|Index: src/SDCCasm.c
|===================================================================
|--- src/SDCCasm.c	(revision 11556)
|+++ src/SDCCasm.c	(working copy)
--------------------------
File to patch:

Dovgalyuk, 2020-03-25:
Ok, I should read "copy “sdcc-z88dk.patch” from z88dk/src/zsdcc into the sdcc-code/sdcc directory"

Dovgalyuk, 2020-03-25:
Thank you!
That was my fault, because I thought that sdcc patching will be performed automatically.

closed 2020-03-25 by Dovgalyuk