z88dk/z88dk #1426: posix - wrappers for FatFs

feilipu, 2020-03-25:
I've found a neat [implementation of posix wrappers](https://github.com/magore/hp85disk/tree/master/posix) for the [ChaN FatFs functions](https://github.com/feilipu/z88dk-libraries/tree/master/ff), and have started on a library for them. The [z88dk-libraries/posix](https://github.com/feilipu/z88dk-libraries/tree/master/posix) library is here (doesn't compile because of conflicts with `<stdio.h>`).

This could be a part of a full posix implementation for z88dk for file I/O. But, I'm not quite sure which direction to take it.

There are already substantial implementations of posix functions is newlib, but some pieces are not present. I don't think it is best to replace what is already there, but it would be good to add in the actual file management as enabled by FatFS into newlib.

I could pick the found posix wrappers code apart, and only add (in C) the pieces that are missing. But, not sure that is the right path, either.

So, open for advice on which direction I should take this. Any input welcome.

aralbrec, 2020-03-30:
I would say namespace it all so there are no conflicts.  One thing I've found out is that it's important to expose the native filesystem in addition to a standard C (read: posix) wrapper as there are sometimes good reasons to use the former.

In the case of FatFS, that should be namespaced as well so that it can be used across targets on top of existing file systems.  I think this posix layer should be considered an extension of that and namespaced as well.  There is a price paid for these things being implemented in C (space and size) and the newlib implementation is trying to avoid that by doing it all in asm.  Most of the pieces are there now for a relatively small implementation -- it's just missing the last mile.  No doubt the completeness of this other posix layer may be higher but on 64k systems, sometimes that's not always desirable.  Namespacing it all means you can have it all and the choice is up to the user.

feilipu, 2020-03-30:
To clarify, by namespace in C you mean just prefacing all the function names with something like
`fopen()` which becomes `_ff_fopen()`. Is that the thought?

aralbrec, 2020-03-30:
Yes that's right.  I wonder if the posix layer is sufficiently independent that it could sit on top of another filesystem.  In that case maybe it should have a different prefix?

feilipu, 2020-03-30:
It is just a very skinny "renaming" of FatFS to comply with posix conventions (that I think FatFS deliberately tried to avoid in the first place).

That said, I'm not sure what the application for this work would be so I'm going to put this on the back of the list. I'll close this issue and come back to it another time, when I know why I'm doing it.

closed 2020-03-30 by feilipu