z88dk/z88dk #1427: "make install" is broken

tinkercnc, 2020-03-28:
After 
`sudo make install`
it spits endless lines with
```
.....
install: missing destination file operand after '{,//usr/local/share/z88dk/}libsrc/_DEVELOPMENT/math/float/math48/z80/am48_derror_pinfnc.asm'
Try 'install --help' for more information.
install: missing destination file operand after '{,//usr/local/share/z88dk/}libsrc/_DEVELOPMENT/math/float/math48/z80/am48_tan.asm'
Try 'install --help' for more information.
.....
```
etc...

And in the source dir there is a subdir with 
```
\{\,/usr/local/share/z88dk/\}
          }include/ }lib/     }libsrc/
...

```

Cheers.

tinkercnc, 2020-03-28:
Oh forgot,!
I'm here
```
commit 59ba2893a6fa8abdf0ce2dbbcc287d8b9ad186b3 (HEAD -> master, origin/master, origin/HEAD)
Merge: d534752a5 8707c4e4d
Author: suborb <dom@suborbital.org.uk>
Date:   Mon Mar 23 20:18:51 2020 +0000

    Merge pull request #1425 from StalePixels/master
    
    Update SpecNext build instructions in examples
```

feilipu, 2020-03-28:
`make install` doesn't work. It is noted in the installation instructions.

You don't need it anyway, as it is assumed you'll need to rebuild target libraries often, as you change configurations.

Please read installation instructions.

suborb, 2020-03-28:
See this comment: https://github.com/z88dk/z88dk/issues/1417#issuecomment-596263034

The install syntax is BSD not GNU so doesn't work on Linux.

tinkercnc, 2020-03-28:
Ok, this did the trick:

`sudo make SHELL=/bin/bash install.`

https://github.com/z88dk/z88dk/wiki/installation/_compare/d84e990257bb216b8faa8e993e449cf8ccaea639...8dc81f80031cb7374b348e6c029bd50682d16c7a

:)

suborb, 2020-03-28:
Ah, I'd forgotten about that gotcha...thanks for the reminder!

closed 2020-03-28 by suborb