z88dk/z88dk #1428: Peephole suggestion

obiwanjacobi, 2020-03-28:
While debugging other stuff I saw a couple of instances of code like this:

```asm
ld hl, 0000
add sp, hl
```

I don't know how often this occurs, but it should be easy to optimize?
[2c]

feilipu, 2020-03-28:
I'm doing other stuff now, but please try the branch in #1429 to see if it helps this issue.
Perhaps there's some side-effects I've missed?

feilipu, 2020-03-28:
Just looking a bit deeper, I've not seen any examples where `ld hl, 0000` comes forward.
Usually I'm seeing `ld hl,0x0000`.

Do you have a piece of code that generates this form?

obiwanjacobi, 2020-03-29:
Uhh, right. I mixed it up - sorry.

```asm
ld      hl, $0000
add     hl, sp
```

Which is fine...
[This sitting at home is getting to me ;-)]

Again, sorry.

closed 2020-03-29 by obiwanjacobi