z88dk/z88dk #1431: [pmd85] Cannot understand some compilation errors

Fabrizio-Caruso, 2020-03-28:
When compiling Cross Chase for the pmd85 with the very same code used for other targets, I get some errors I cannot really understand...

```
Error at file 'chase/item.c' line 142: illegal identifier
Error at file 'chase/item.c' line 147: illegal identifier
Errors in source file chase/item.c:
Error at file 'chase/item.c' line 142: illegal identifier
                   ^ ----
Error at file 'chase/item.c' line 147: illegal identifier
                   ^ ----       }
make: *** [Makefile:7470: pmd85] Error 1

```

[image]

suborb, 2020-03-28:
It looks like a bit of z80 code has crept into an 8080 compilation.

I shall endeavour to fix it.

Fabrizio-Caruso, 2020-03-28:
Where do I do 32 bit arithmetic?
As you can see I am using uint16_t, which is an unsigned 16 bit variable.

UL is a mistake...

suborb, 2020-03-28:
Because the constants are marked as UL sccz80 does the maths in 32 bit mode.

Killing the UL will also remove remove the non-8080 compatible code that's generated here.

Fabrizio-Caruso, 2020-03-28:
@suborb, indeed it is an 8080-only issue.
Removing UL fixes the problem for the pmd85 and vector06c targets.
I still get an error on the m100 on an line that does not exist:

```
Errors in source file cross_lib/text/text_macros.c:
Error at file 'cross_lib/text/text_macros.c' line 41: illegal identifier
                   ^ ---- (null)Error at file 'cross_lib/text/text_macros.c' line 42: illegal identifier
                   ^ ---- (null)make: *** [Makefile:7539: m100] Error 1
```
text:_macros.c has 36 lines.

suborb, 2020-03-28:
Oops, that a shift that slipped through when running in 8085 mode.

Fabrizio-Caruso, 2020-03-28:
Could you please tell how you see these errors?
I see errors on the wrong file and wrong line...

suborb, 2020-03-28:
I'm creating the assembler output with -a and then looking at the .c.asm files that are generated.

Fabrizio-Caruso, 2020-03-29:
@suborb, m100 is still broken as of today (2020-03-29):
```
z80asm -m8085 -b -d  -o"../build/FULL_m100.bin" -m -s -L.   -D__SCCZ80 -L"c:/z88dk//lib/clibs" -I"c:/z88dk//lib" -indos -im100_clib -indos -i"8085_crt0"     "@C:\cygwin\tmp\zcc6B8319.lst"
Errors in source file c:/z88dk/\lib\target\m100\classic\m100_crt0.asm:
Error at file 'c:/z88dk/lib/crt/classic/crt_section.asm' line 34: symbol '__z80asm__ldir' not defined
                   ^ ----         ldir
make: *** [Makefile:7566: m100] Error 1
```

suborb, 2020-03-29:
The m100 issue is effectively the same issue as the gameboy problem in #1433 so let’s continue there.

suborb, 2020-03-30:
Original issue is resolved, missing support library should be dealt with by #1433

closed 2020-03-30 by suborb