z88dk/z88dk #1433: [gb] Cross Chase compilation errors about undefined symbols

Fabrizio-Caruso, 2020-03-28:
If I compile Cross Chase for the gb target I get lots of errors about missing symbols as shown below:
```
/cygdrive/c/z88dk/bin/zcc.exe +gb -Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_data -Icross_lib/display/graphics_data/cc65 -Icross_lib/display/graphics_data/z88dk -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/display/image_settings -Icross_lib/display/redefine_characters -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/psg -Icross_lib/text -Icross_lib/input   \
-D__GB__ \
-DFORCE_XSIZE=20 \
-DFORCE_YSIZE=18 \
-DNO_SET_SCREEN_COLORS \
-DNO_COLOR \
-DFULL_GAME  \
-DCONIO_VT52 \
-DCONIO \
-lndos \
-DNO_INIT_GRAPHICS \
cross_lib/display/display_macros.c cross_lib/display/init_images.c cross_lib/input/input_macros.c cross_lib/text/text_macros.c chase/move_player.c chase/ghost.c  chase/level.c chase/character.c chase/game_text.c chase/main.c chase/strategy.c chase/item.c chase/skull.c chase/bullet.c   chase/horizontal_missile.c chase/rocket.c  \
-create-app -o../build/FULL_gc.bin
Error at file 'c:/z88dk/lib/crt/classic/crt_section.asm' line 71: symbol '__z80asm__ldir' not defined
Error at file '__printf_get_flags_impl.asm' line 68: symbol '__z80asm__ex_sp_hl' not defined
Error at file '__printf_get_flags_impl.asm' line 115: symbol '__z80asm__ex_sp_hl' not defined
Errors in source file c:/z88dk/\lib\target\gb\classic\gb_crt0.asm:
Error at file 'c:/z88dk/lib/crt/classic/crt_section.asm' line 71: symbol '__z80asm__ldir' not defined
                   ^ ----       ldir
Error at file '__printf_get_flags_impl.asm' line 68: symbol '__z80asm__ex_sp_hl' not defined
Error at file '__printf_get_flags_impl.asm' line 115: symbol '__z80asm__ex_sp_hl' not defined
make: *** [Makefile:7550: gb] Error 1
```

suborb, 2020-03-28:
You should have the file {z88dk}/lib/z80asm-gbz80-.lib which contains these emulation functions. I've just checked the nightly build and they are present.

Fabrizio-Caruso, 2020-03-28:
@suborb  I am using the very latest build.
I do see the lib.

[image]

suborb, 2020-03-29:
Cool, if you run z88dk-z80nm then you should the symbols.

It does look like it's all going a bit wrong at the linking stage. So
so if you stick `-Cl-v -v` into the command line it will turn on some extra logging to console for the linker stage. It's this stage that should be reading it, there will be loads of messages, but towards the end you should see something like this:

```
z80asm -mgbz80 -b -d  -o"../build/FULL_gc.bin" -m -s -L.   -v -D__SCCZ80 -L"/Users/dom/z88dk/lib/config//../..//lib/clibs" -I"/Users/dom/z88dk/lib/config//../..//lib" -indos -igb_clib -indos -i"gbz80_crt0"     "@/tmp/tmpYwBysrZf.lst"
Predefined constant: __SCCZ80 = $0001
Reading library '/Users/dom/z88dk/lib/clibs/ndos.lib'
Reading library '/Users/dom/z88dk/lib/clibs/gb_clib.lib'
Reading library '/Users/dom/z88dk/lib/clibs/ndos.lib'
Reading library '/Users/dom/z88dk/lib/clibs/gbz80_crt0.lib'
Library 'z80asm-gbz80-.lib' not found
Library '/usr/local/share/z88dk/lib/z80asm-gbz80-.lib' not found
Reading library '/Users/dom/z88dk/lib/z80asm-gbz80-.lib'
Predefined constant: __CPU_GBZ80__ = $0001
```

Do the search locations that you see correspond to install location of z88dk?

As a quick bodge it looks like if you copy z80asm-gbz80-.lib to the current directory it should read it from there.

Fabrizio-Caruso, 2020-04-19:
@suborb 
Is it just a problem with the lib file not being found?
Is there any progress on this problem?

suborb, 2020-04-19:
Right, running this on a fresh Windows installation we get the following output:

```
-create-app -o../build/FULL_gb.bin
Predefined constant: __SCCZ80 = $0001
Reading library 'c:/z88dk/lib/clibs/ndos.lib'
Reading library 'c:/z88dk/lib/clibs/gb_clib.lib'
Reading library 'c:/z88dk/lib/clibs/ndos.lib'
Reading library 'c:/z88dk/lib/clibs/gbz80_crt0.lib'
Library 'z80asm-gbz80-.lib' not found
Library '/usr/local/share/z88dk/lib/z80asm-gbz80-.lib' not found
Library '/../z80asm-gbz80-.lib' not found
```

So as suggested above, copying the library to the current directory would work.

However, that's not great, so actually configuring the ZCCCFG variable to point to /c/z88dk/lib/config gets us this output:

```
Predefined constant: __SCCZ80 = $0001
Reading library 'C:/z88dk/lib/clibs/ndos.lib'
Reading library 'C:/z88dk/lib/clibs/gb_clib.lib'
Reading library 'C:/z88dk/lib/clibs/ndos.lib'
Reading library 'C:/z88dk/lib/clibs/gbz80_crt0.lib'
Library 'z80asm-gbz80-.lib' not found
Library '/usr/local/share/z88dk/lib/z80asm-gbz80-.lib' not found
Reading library 'C:/z88dk/lib/z80asm-gbz80-.lib'
```

Which means that the library can now be found. There's obviously a prefix issue in the nightly build which I'll need to look at, but usually ZCCCFG is set so it's not a massive issue.

suborb, 2020-04-20:
In addition to the above workaround, the nightly from today now has the correct prefix so we see this:

```
Predefined constant: __SCCZ80 = $0001
Reading library 'c:/z88dk/lib/clibs/ndos.lib'
Reading library 'c:/z88dk/lib/clibs/gb_clib.lib'
Reading library 'c:/z88dk/lib/clibs/ndos.lib'
Reading library 'c:/z88dk/lib/clibs/gbz80_crt0.lib'
Library 'z80asm-gbz80-.lib' not found
Reading library 'c:/z88dk/lib/z80asm-gbz80-.lib'
```

So it's no longer looking in /usr/local/share

closed 2020-04-20 by suborb