z88dk/z88dk #1434: [zxn] How to compile for the ZX Next target? No doc found

Fabrizio-Caruso, 2020-03-29:
I am trying to compile Cross Chase for the ZX Next but I find no doc in:
https://github.com/z88dk/z88dk/wiki/Platform

If I compile with the same options as for targets with support for the generic-console, I get:
```
$ make zxn
/cygdrive/c/z88dk/bin/zcc.exe +zxn -Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_data -Icross_lib/display/graphics_data/cc65 -Icross_lib/display/graphics_data/z88dk -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/display/image_settings -Icross_lib/display/redefine_characters -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/psg -Icross_lib/text -Icross_lib/input   \
-D__ZXN__ \
-DFORCE_XSIZE=32 \
-DFORCE_YSIZE=24 \
-DNO_SET_SCREEN_COLORS \
-DNO_COLOR \
-DFULL_GAME  \
-DCONIO_VT52 \
-DCONIO \
-lndos \
-DNO_INIT_GRAPHICS \
cross_lib/display/display_macros.c cross_lib/display/init_images.c cross_lib/input/input_macros.c cross_lib/text/text_macros.c chase/move_player.c chase/ghost.c  chase/level.c chase/character.c chase/game_text.c chase/main.c chase/strategy.c chase/item.c chase/skull.c chase/bullet.c   chase/horizontal_missile.c chase/rocket.c  \
-create-app -o../build/FULL_zxn.bin
In file included from cross_lib/cross_lib.h:116,
                 from cross_lib/display/display_macros.c:23:
cross_lib/display/display_macros.h:34:20: fatal error: conio.h: No such file or directory
compilation terminated.
```

suborb, 2020-03-29:
The ZX Next is only supported by newlib so generic console/conio isn't available.

Fabrizio-Caruso, 2020-03-29:
Is a conio wrapper available?
How can I do simple conio-like operations such as placing the cursor and print a character?
Is there a manual?

I would like to start with simple conio text "graphics".

aralbrec, 2020-03-30:
Standard zx spectrum programs run fine so anything compiled using the classic library for the zx will continue to work.

If you want to take advantage of the new hardware then you may want to do a zxn-specific compile.  There are hardware sprites, hardware scrolling, 256 colours per pixel, etc.  If you're sticking to a text game there is a 40x32 or 80x32 tilemap mode too.

Fabrizio-Caruso, 2020-03-30:
@aralbrec I would like to compile my game and use some conio-like or simple text "graphics". What can I use? CONIO seems not to work. Can I use VT52 codes to place the cursor? What is the default text resolution?

I would like to use an emulator. Which one can I use with the generated binary? Any special options I should use?

Fabrizio-Caruso, 2020-03-30:
@suborb, where can I find more info on how to do simple text graphics on this target?
I would like to start with something conio-like.

suborb, 2020-03-30:
As aralbrec said, the classic +zx code will work just fine on the Next.

If you want to make use of Next hardware then as far as I can tell you'll have to do it yourself, there's nothing in newlib to make use of the tilemap. The documentation for the tilemap hardware is here: https://www.specnext.com/tilemap-mode/ and I think there's some examples in the forum there: https://www.specnext.com/forum/

To date, I've very deliberately not added any Next support to classic

Xalior, 2020-03-30:
I found https://gitlab.com/thesmog358/tbblue/-/tree/master/src/c/tm to be especially useful when I was starting personally. It will need the NEXTREG macros switching for pairs of OUTs, but otherwise should be a straight forward port/learning example.

-Dx

Fabrizio-Caruso, 2020-03-30:
@suborb I am sorry but I never understood wgat classic lib exactly stands for. Which control codes in printf statements are classic? How do I place the cursor on the screen? Are udg and mono sprites part of  classic lib?

suborb, 2020-04-01:
Pretty much everything you use is classic (UDGs, screen position, sprites, graphics, sound): https://github.com/z88dk/z88dk/wiki/Classic-Overview

Despite being stuck in the house all day I seem to have even less time to do stuff than usual. However I had a go at creating a zxn target for classic with an extra couple of screen modes for the tilemap. 

Unfortunately I don't have any hardware and the cspect emulator crashed on startup under both wine and Windows 10 (on those occasions that it didn't contain a trojan) so I can't go any further with it at the moment.

suborb, 2020-04-02:
You can probably guess what Yodel dumped on my doorstep today. 

I might get some time to play with it over the Easter weekend.

suborb, 2020-04-10:
Hmmm, well it was a massive pain, but I'm getting somewhere:

[image]

Finally, on the 4th machine I tried I managed to get the emulator working (as long as it doesn't attempt to display any dialogues). It's not the mono version, but some library it depends on that just crashes constantly on both Linux and Mac. On Windows I guess I'm missing a dependency but I'm not sure what it.

 Zesaurux is unfortunately like treacle on all my hardware when it's emulating the Next.

suborb, 2020-04-12:
Progress report: I've imported library code from newlib for +zx, +ts2068, +zxn. For gencon I've got tilemap modes for 40 and 80 column. The code to handle the printing reads the registers so after switching the user can repoint.

I'm currently playing around with the palette so that we get functionality roughly on a par with other tilemap systems:

[image]

This isn't quite right - I'm off by one in the colour indices.

Xalior, 2020-04-12:
You may want to upgrade CSpect to at least 2.12.17 (released yesterday) as we found a bug in the way it handles transparency in 1bit Tilemap mode, and it was just fixed.  You’re not likely bumping into that *here* but to stop future head scratching... :)

-Dx

Sent from my iPhone

suborb, 2020-04-13:
I've just merged preliminary support for the ZX Next in classic. Documentation is here:

https://github.com/z88dk/z88dk/wiki/Platform---Sinclair-ZX-Spectrum-Next

This should be enough to get Cross Chase running either using the classic ZX display or using the new tilemap mode. There's a lot more work I'd like to do (full graphics support for example) but it's a start.

Fabrizio-Caruso, 2020-04-13:
@suborb, does this mean I can also use CONIO?

suborb, 2020-04-13:
From the build tonight you should be able to.

Fabrizio-Caruso, 2020-04-14:
@suborb There is no build today

Fabrizio-Caruso, 2020-04-14:
@suborb the build may have failed as I see no build for today.

suborb, 2020-04-14:
Yes, unfortunately the build couldn't compile zsdcc due to a sourceforge issue: https://www.z88dk.org/forum/viewtopic.php?id=11274

Fabrizio-Caruso, 2020-04-15:
@suborb  The problem persists. Still no snapshot today

suborb, 2020-04-15:
I ran the build manually, there's a kit now.

closed 2020-04-15 by suborb

Fabrizio-Caruso, 2020-04-15:
@suborb Still no conio support if I compile it with the same options as other conio enabled targets:

```
$ make zxn
/cygdrive/c/z88dk/bin/zcc.exe +zxn -Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_data -Icross_lib/display/graphics_data/cc65 -Icross_lib/display/graphics_data/z88dk -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/display/image_settings -Icross_lib/display/redefine_characters -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/psg -Icross_lib/text -Icross_lib/input   \
-D__ZXN__ \
-DFORCE_XSIZE=32 \
-DFORCE_YSIZE=24 \
-DNO_SET_SCREEN_COLORS \
-DNO_COLOR \
-DFULL_GAME  \
-DCONIO_VT52 \
-DCONIO \
-lndos \
-DNO_INIT_GRAPHICS \
cross_lib/display/display_macros.c cross_lib/display/init_images.c cross_lib/input/input_macros.c cross_lib/text/text_macros.c chase/move_player.c chase/ghost.c  chase/level.c chase/character.c chase/game_text.c chase/main.c chase/strategy.c chase/item.c chase/skull.c chase/bullet.c   chase/horizontal_missile.c chase/rocket.c  \
-create-app -o../build/FULL_zxn.bin
In file included from cross_lib/cross_lib.h:116,
                 from cross_lib/display/display_macros.c:23:
cross_lib/display/display_macros.h:34:20: fatal error: conio.h: No such file or directory
compilation terminated.
make: *** [Makefile:7622: zxn] Error 1
```