z88dk/z88dk #1437: ZX Spectrum documentation & libraries wrong and incomplete

Baltasarq, 2020-04-02:
I've created an adventure game using ansi console and the dos library. There seems to be enough docs about that. As for creating arcades, there seems to be enough documentation about graphics and sprites and so on.
I've found though that when you get away from that niche projects, then a lot of problems appear that left you with the only possibility of trial & error.
So, to be clear, I want to create an adventure game, but being able to locate text on the screen (not terminal emulation), change colors... to be absolutely clear I want to "emulate" the following BASIC commands: CLS, PRINT AT, INK, & PAPER... at least, for now.
The only documentation I found are the next wiki page (especially the table about the commands for the screen driver), and the spectrum.h header itself:
https://www.z88dk.org/wiki/doku.php?id=platform:zx
https://github.com/z88dk/z88dk/blob/master/include/spectrum.h    [line 317]

CLS
===
The problem with the wiki page is that there is not any example. I guess that many people messing with the z88dk will be already programmers of some kind, but that does not have to be the case.  It would not harm to have a complete abstraction layer instead of sending codes to the screen via fputc_cons() [an API you only learn about by reading spectrum.h].

Suggestion: add the next line to spectrum.h:
#define zx_cls()              fputc_cons(12);

zx_sercursorpos(a, b)
=================
In conjunction with printf(), this allows to emulate PRINT AT. However, there is an error here: the wiki says that:
"""
22,y,x	  - Move to position y,x on the screen (0<=y<=23, 0<=x<=63)
	    NB. y and x are displaced by 32 eg to move the print position
	    to (0,0) use 22,32,32. 32 column mode also uses 64 column
	    coordinates.
"""
Unfortunately, the zx_setcursorpos() lacks that shift on the coordinates, so it is useless right now. Also, by reading the header file the parameters are just called a and b, so you don't really know whether it is x,y or y, x. This is the whole declaration:
#define zx_setcursorpos(a,b)      fputc_cons(22); fputc_cons(a); fputc_cons(b);

The problem is that maybe some projects already use zx_setcursorpos() manually shifting the params...

Suggestion: change zx_setcursorpos to:
#define zx_setcursorpos(row,col)      fputc_cons(22); fputc_cons(row); fputc_cons(col);
And add a warning that you should add 32 to the params yourself.

Suggestion: add zx_movecursorto:
#define zx_movecursorto(row,col)      fputc_cons(22); fputc_cons(row + 32); fputc_cons(col + 32);

INVERSE
=======
inverse is achieved by using fputc_cons(20) and then 48 (off) or 49 (on).
Suggestion: add a new macro:
#define zx_setattrinverse(b)           fputc_cons(20); fputc_cons((b)? 49: 48);

FLASH
=======
flash is achieved by using fputc_cons(18) and then 48 (off) or 49 (on).
Suggestion: add a new macro:
#define zx_setattrflash(b)           fputc_cons(18); fputc_cons((b)? 49: 48);

BRIGHT
=======
bright is achieved by using fputc_cons(19) and then 48 (off) or 49 (on).
Suggestion: add a new macro:
#define zx_setattrbright(b)           fputc_cons(19); fputc_cons((b)? 49: 48);

PAPER & INK
===========
zx_border & zx_colour work great (at least to the extent I'm using them), but they are lacking documentation:
Suggestion: add to the wiki:
zx_border(x); Changes the border on screen. Where x can be INK_BLACK, INK_BLUE, INK_RED, INK_MAGENTA, INK_GREEN, INK_CYAN, INK_YELLOW, and INK_WHITE.

zx_colour(i | p): Changes the whole screen attributes. (Or at least that's how I'm using it) where i could be any of INK_BLACK, INK_BLUE, INK_RED, INK_MAGENTA, INK_GREEN, INK_CYAN, INK_YELLOW, and INK_WHITE, and p could be any of PAPER_BLACK, PAPER_BLUE, PAPER_RED, PAPER_MAGENTA, PAPER_GREEN, PAPER_CYAN, PAPER_YELLOW, PAPER_WHITE.

The last problem is that I'm unable to make codes 16 (ink) and 17 (paper) to work (that's why I am not providing a possible implementation). In my opinion, there should be a function in the ways of zx_colour(), say it could be called zx_setattr():

zx_setattr(i | p): where i could be any of INK_BLACK, INK_BLUE, INK_RED, INK_MAGENTA, INK_GREEN, INK_CYAN, INK_YELLOW, and INK_WHITE, and p could be any of PAPER_BLACK, PAPER_BLUE, PAPER_RED, PAPER_MAGENTA, PAPER_GREEN, PAPER_CYAN, PAPER_YELLOW, PAPER_WHITE.

suborb, 2020-04-11:
Sorry, I completely missed this earlier.

I've not looked at the ZX stuff in a while so docs/code may have drifted apart due to changes elsewhere, apologies for this.

Things changed over the past couple of years ago in the name of cross-platform portability as the gencon was introduced: https://github.com/z88dk/z88dk/wiki/Classic-GenericConsole - you're actually using it on `+zx` since it's the default terminal driver in classic.

This means that the abstraction layer in `<conio.h>` is available to you which does define quite a few of the things you mention. Ideally we should get to the point where `<conio.h>` is the preferred approach with "special stuff" in the appropriate machine file. For the common behaviour the following translations can be used:

    zx_cls() -> clrscr()
    zx_border(n) -> bordercolor(n)
    zx_setcursorpos(row,col) -> gotoxy(col, row)
    zx_ink(c) -> textcolor(c)
    zx_paper(c) -> textbackground(c)

I think some of the problem you have with colours is that the colours used are those defined in `<conio.h>` so the order isn't quite as you expect. You can go back to standard ZX colour indices using the option `-pragma-define:CLIB_CONIO_NATIVE_COLOUR=1`

It looks like there are some convenience functions for inverse, bold, underline which I haven't added - I'll do so when I've finished my current work. Flash/bold/bright are ZX only things so could easily be added to `<spectrum.h>`.

Looking into this a bit more, it does look like `zx_setcursorpos` and `zx_topleft` are meant to work against the ROM driver rather than the default driver. You can link to this with the option `--pragma-redirect:fputc_cons=fputc_cons_rom_rst`

Baltasarq, 2020-04-12:
Thanks for taking the time to address this issue.

> I've not looked at the ZX stuff in a while so docs/code may have drifted apart due to changes elsewhere, apologies for this.

Yep. absolutely. For what I read in your answer, one should rely in conio.h for everything. The problem is that the docs do not address you in that direction, or at least I wasn't able to figure it out.

About the pragmas you suggest for compilation, the "-pragma-define:CLIB_CONIO_NATIVE_COLOUR=1" did the trick. Now I have the spectrum colors without touching my code. Nice. :-)

About the other one, compiling against the ROM driver, it did not work at all. The program just printed a "?" in the second row, first column. Anyway, zx_setcursorpos() and zx_topleft() were already working, there wasn't actually the need to fix anything.

BTW, I guess that by "bold", you mean "inverse".

As I wrote before, I have already written an adventure game using conio, and specially the ansi lib. I was willing to get rid of all the ANSI/conio stuff for this project, thinking that it would add a lot of layers of abstraction and thus harden more memory requirements. Anyway, I wasn't able to get rid of -lndos, though AFAIK I'm not using any of it apart from printf(). Don't know whether there is a zx_print(), but I haven't been able to find it do far.

BTW, I could implement the changes I described above myself, and do a pull request, though I did not know about the policy (i.e., didn't know whether it was to be welcomed or not). Is there a way to somehow automatically include the pragma in spectrum.h? I guess that if you are going to include spectrum.h you are not going to use conio? Maybe just #define CLIB_CONIO_NATIVE_COLOUR 1?

Thanks for your help.

suborb, 2020-04-13:
Pull requests are very welcome! 

Nope, I meant bold! See #1360

Regarding trimming down, there's a few tips here: https://github.com/z88dk/z88dk/wiki/WritingOptimalCode#classic-library - you should be able to get rid of -lndos that way, though it is only a few bytes.

I think the problem with having the pragma in `<spectrum.h>` is that header file defines numerous other functions that you might want to use whilst still using `<conio.h>`, I've added documentation regarding this here: https://github.com/z88dk/z88dk/wiki/Platform---Sinclair-ZX-Spectrum#the-standard-zx-spectrum-console-driver

Baltasarq, 2020-04-14:
Thanks, I will start with a simple Pull Request in the next days.

> Regarding trimming down, there's a few tips here: https://github.com/z88dk/z88dk/wiki/WritingOptimalCode#classic-library - you should be able to get rid of -lndos that way, though it is only a few bytes.

I see, but... one of the modification involves using `printk` instead of `printf`... though `printk` is still declared in `stdio.h`! It still works however, so the linker finds it, thus...

Shouldn't `spectrum.h`declare zx_print as a macro for printk? Something like:

extern void printk(char *fmt,...);
#define zx_print(s) printk(s)

This way it would work, and you won't have to include `stdio.h`...

What do you think?

Baltasarq, 2020-04-20:
Various pull request's merged in master. Thanks!

closed 2020-04-20 by Baltasarq

suborb, 2020-04-20:
@Baltasarq thank you for the contribution!