z88dk/z88dk #1438: [aquarius][appmake?] .caq does not seem to work on the real hardware (it does work on emulator)

Fabrizio-Caruso, 2020-04-04:
No version of Cross Chase can be loaded into a real Aquarius.
Roy Templeman (https://www.facebook.com/roy.templeman) has checked most versions that have been created in the last couple of years. None of them works on the real hardware as you can see on his screen shots.
[image]

They all load on the emulator.
[image]

zx70, 2020-04-05:
To have someone able to check the real hardware is a rare and precious resource for the z88dk project  :)
We first need to discriminate where the problem happens,  is it failing while loading the first loader block? 
Does it fail also with smaller and simpler programs? 
how was the wav created?

Fabrizio-Caruso, 2020-04-05:
@zx70 as you see in the photo by Roy Templeman, it seems to load the first part correctly.

Both the TINY and the FULL versions fail. The TINY version is just 4608 bytes.
The same problem occurs on previous binaries, as well. So it is not a recent problem. 

Roy did not tell me how he generates the wav files.

Fabrizio-Caruso, 2020-04-05:
@zx70 Roy refers that he has tried to load the caq files both directly through his Micro Expander and also by converting them to wav. 
He has never had problems with other caq files.

LoudScotsBloke, 2020-04-05:
Hi guys. Its Roy. I have over 70 retro computers in my collection, so trying stuff on the real hardware is what i am good at. I have a soft spot for the Aquarius, as it was my first machine. I use a modern add-on to the Aquarius, that allows me to have USB support, and also utilise CAQ files, very handy. I have, and create, many CAQs and have never had an issue. But when loading the machine code part of Fabrizio's game, it fails and crashes the Aquarius. The first part ( the basic loader) loads fine, its only the CLOAD*A element. I also tried creating a WAV from the CAQ, but this also failed at exacly the same point, so I think that its the CAQ that is not quite right.

zx70, 2020-04-07:
Many thanks for joining and the details. 
Have you ever tried other z88dk compiled programs ?  If not I'd try a minimalistic test program to understand how big is the issue we have

zx70, 2020-04-09:
Re-checking the post by Fabrizio, I understand you even fail to get the loader correctly .
The logic is similar to the loaders used by "Millypede" or "Alien", I suppose such games can be loaded and run with no problems, right?

LoudScotsBloke, 2020-04-10:
> Re-checking the post by Fabrizio, I understand you even fail to get the loader correctly .
> The logic is similar to the loaders used by "Millypede" or "Alien", I suppose such games can be loaded and run with no problems, right?

Indeed. The majority of loaders for the Aquarius are 2 stage, and I have no issues with them. Just to clarify, the initial load (the first cload basic part) of Fabrizio's loads ok. I then run it, and it initiates the cloadm (cloadm,*A) element. That fails. It crashes the Aquarius.

zx70, 2020-04-10:
thank you for the clarification,  can you list the loader?   is it crashing immediately after run or while loading?

LoudScotsBloke, 2020-04-10:
> thank you for the clarification, can you list the loader? is it crashing immediately after run or while loading?

I can list the basic loader yes. I run it.. it asks the usual 'press play on tape" then continues. At some point through the load, it crashes. Not sure if it's the end or not. It certainly feels like it's the end, and when it runs, that's when it crashes.

zx70, 2020-04-10:
one more question, please:   is it a standard memory model or an expanded one?

Fabrizio-Caruso, 2020-04-10:
@zx70 Roy told me that he uses a 16k ram pack. The Aquarius with no memory expansion is not really usable with Z88DK as it has less than 2k of available free memory left.

LoudScotsBloke, 2020-04-11:
> one more question, please: is it a standard memory model or an expanded one?

As Fabrizio mentioned, yes, I'm using a standard 16k ram pack.

Fabrizio-Caruso, 2020-04-29:
@zx70 , we have bad news about the tests by @LoudScotsBloke on a real Aquarius with 16k expansion

I have compiled with:
```
aquarius_hello_stack:
	$(Z88DK_PATH)$(MYZ88DK) +aquarius -pragma-redirect:fputc_cons=fputc_cons_generic -pragma-output:STACKPTR=20000 $(SCCZ80_OPTS) -vn \
	$(SOURCE_PATH)/../test/hello_fputc.c \
	-create-app -o minimal_hello
```
the following minimal example:
```
#include<stdio.h>

int main()
{

    fputc_cons('h');
    fputc_cons('e');
    fputc_cons('l');
    fputc_cons('l');
    fputc_cons('o');
    while(1){};
    return 0;
}
```
which produces the tiny 

[minimal.zip](https://github.com/z88dk/z88dk/files/4553837/minimal.zip)

The result is similar to the one already reported for other tests with the first file (the loader) loading and running and the second file freezing as shown on the attached screenshot from the real hardware:
[image]

Fabrizio-Caruso, 2020-04-30:
@zx70  This may help:
https://github.com/MackW/AquariusIntelLoader

Fabrizio-Caruso, 2020-05-02:
This may also help

[image]

Fabrizio-Caruso, 2020-05-03:
The function CreateCAQ seems to be doing exactly what we need:
https://github.com/MackW/AquariusIntelLoader/blob/master/AquarisBasicMaker/CAQ.cs#L5

zx70, 2020-05-07:
The tool you are suggesting is very similar to the loader we provided time ago.   An optimization was inserted then, to save about 60 bytes and it could have been the culprit.   In case it will not be enough a little extra variant can be tried.

zx70, 2020-05-08:
+fixed the graphics (-lgfxaq48), it was gone in the latest rationalization of the GFX library.

Fabrizio-Caruso, 2020-10-22:
@LoudScotsBloke confirms that the new caq and wav work on the real hardware.
Has tried both the .caq and a converted .wav files.
He has used the Micro Expander to load the .caq files.

closed 2020-10-22 by Fabrizio-Caruso

zx70, 2020-10-22:
that's great news,  thankyou!