z88dk/z88dk #1439: Unoptimal code generation for calling functions with more than 3 parameters

seedy1812, 2020-04-12:
The attached code produces when calling a function  with 4 or more parameters  uses the stack frame to store values temporary. This adds 4*19 T states per parameter and that does not include code to create space for the frame.

The command line I am using is is
 zcc +zxn -vn -m test.c -o test.asm" -S

`
typedef unsigned int           uint16_t;

void		MAP_SubRegion3(uint16_t x1, uint16_t y1,uint16_t x2);
void		MAP_SubRegion4(uint16_t x1, uint16_t y1,uint16_t x2,uint16_t y2);
void		MAP_SubRegion5(uint16_t x1, uint16_t y1,uint16_t x2,uint16_t y2,uint16_t x3);
void		MAP_SubRegion6(uint16_t x1, uint16_t y1,uint16_t x2,uint16_t y2,uint16_t x3,uint16_t y3);
void		MAP_SubRegion7(uint16_t x1, uint16_t y1,uint16_t x2,uint16_t y2,uint16_t x3,uint16_t y3,uint16_t x4);

uint16_t  			ipos[10];
void test0_3()
{
	MAP_SubRegion3(ipos[0],ipos[1],ipos[2]);
}

void test0_4()
{
	MAP_SubRegion4(ipos[0],ipos[1],ipos[2],ipos[3]);
}

void test0_5()
{
	MAP_SubRegion5(ipos[0],ipos[1],ipos[2],ipos[3],ipos[4]);
}

void test0_6()
{
	MAP_SubRegion6(ipos[0],ipos[1],ipos[2],ipos[3],ipos[4],ipos[5]);
}

void test0_7()
{
	MAP_SubRegion7(ipos[0],ipos[1],ipos[2],ipos[3],ipos[4],ipos[5],ipos[6]);
}`

Example output for test()_5
`test.c:22: void test0_5()
;	---------------------------------
; Function test0_5
; ---------------------------------
_test0_5:
	push	ix
	ld	ix,0
	add	ix,sp
	push	af
	push	af
;test.c:24: MAP_SubRegion5(ipos[0],ipos[1],ipos[2],ipos[3],ipos[4]);
	ld	de, (_ipos + 8)
	ld	hl,(_ipos + 6)
	ld	(ix-4),l
	ld	(ix-3),h
	ld	hl,(_ipos + 4)
	ld	(ix-2),l
	ld	(ix-1),h
	ld	bc,(_ipos + 2)
	ld	hl,(_ipos)
	push	de
	ld	e,(ix-4)
	ld	d,(ix-3)
	push	de
	ld	e,(ix-2)
	ld	d,(ix-1)
	push	de
	push	bc
	push	hl
	call	_MAP_SubRegion5
	ld	hl,10
	add	hl, sp
	ld	sp, hl
;test.c:25: }
	ld	sp, ix
	pop	ix
	ret`

aralbrec, 2020-04-14:
This has been a bit of a long-standing annoyance with sdcc for a while.  Philip doesn't seem to be taggable so maybe the only way to address it is on sdcc's bugtracker.

sdcc wants to collect all parameters into registers before it generates the stack frame for the call.  Since it only has three 16-bit register pairs available, it resorts to temporary space on the stack to keep more parameters.  Instead what it should be doing is going straight into building the stack frame and push the parameters one at a time as it fetches them.  If there are repeat expressions, either push multiple times or retrieve the result from the stack and push again.

With the -SO3 and --opt-code-size (and --max-allocs-per-node200000) options, z88dk is trying to fix up most of the common cases of poor code generation.  But with this case it is difficult to fix up because you have to look deep into the code to see an optimization can happen and the deeper you go the more specific the case is.

seedy1812, 2020-04-14:
The problem is the variables are elements in a global array and pushing the addresses with a fixed address. As long as the variables are not being modified it is quicker just to read the memory rather than the 2 reads  or 2 writes using the ix register.
ld (ix+dd),r8 is 3 bytes and takes 19 cycles - so 39 for the pair. Whilst ld hl,(xxxx) is 3 bytes but only 16 cycles - almost twice as fast.

aralbrec, 2020-04-15:
Yes but it doesn't do that because it wants all function parameters gathered before it generates the stack frame.  So with test05() above, it wants to gather five parameters before it begins to push onto the stack and since the z80 only has three register pairs available, it gets into trouble and starts putting parameters into temp space on the stack which is the indexed locations.  After it has all five parameters, it will generate the stack frame.

This is not a good approach for the code generation but it is baked into sdcc at the moment.  It would be better if it just created the stack frame one parameter at a time instead of waiting for all the parameters to be gathered beforehand.

This problem is very difficult to fix in post-processing which is what we are doing in z88dk -- the SO3 and opt-code-size rules are about correcting sdcc's output where it stumbles.  But it's not really possible to correct this sort of output post- code generation.  It's a problem that needs to be solved in the code generator and that is maintained by Philip at sdcc.  I'll set up a ticket there and link to it here later.

In the meantime, you can wait for a fix or try to alleviate it by changing how parameters are passed and avoid passing too many arguments in the first place.  Many of the ideas below you probably know about and most are probably unsightly but they will improve things.

In the specific example you gave, which may be a toy example,

```
void test0_5()
{
	MAP_SubRegion5(ipos[0],ipos[1],ipos[2],ipos[3],ipos[4]);
}
```

I noticed that those are five integers in a row so you may want to pass the address of the first integer instead.

```
void test0_5()
{
	MAP_SubRegion5(&ipos[0]);
}
```

Or change to passing parameters via a static memory parameter block.  You have to be careful as this requires the function called not to nest or recurse.  Having the parameters in memory may improve code generation of the function itself (the one receiving the parameters) but it depends on the form of the function.  zsdcc will keep locals in registers if possible but the z80 has few registers so it can't do that if the working set of values becomes more than modest.  So usually the best mix is a combination of a few locals and the rest static, keeping an eye on the number of working set values in each stage of a function and perhaps delimiting each stage with blocks {}.

```
struct params_s
{
   unsigned int p0;
   unsigned int p1;
   unsigned int p2;
   unsigned int p3;
   unsigned int p4;
   unsigned int p5;
};

struct params_s params;

void test0_5()
{
   // if the params are contiguous as the function call suggests
   // memcpy(&params, ipos, sizeof(unsigned int)*5);
   // MAP_SubRegion5();

   params.p0 = ipos[0];
   params.p1 = ipos[1];
   params.p2 = ipos[2];
   params.p3 = ipos[3];
   params.p4 = ipos[4];

   MAP_SubRegion5();
}
```

Or you could write the call yourself in asm:

```
// c file

extern void test0_5(void);

;; asm file

SECTION code_user
PUBLIC _test0_5
EXTERN _MAP_SubRegion5, _ipos

_test0_5:

   ld hl,(_ipos + 4*2)
   push hl
   ld hl,(_ipos + 3*2)
   push hl
   ld hl,(_ipos + 2*2)
   push hl
   ld hl,(_ipos + 1*2)
   push hl
   ld hl,(_ipos + 0*2)
   push hl

   call _MAP_SubRegion5   ;; assuming no return values

   ld hl,10
   add hl,sp
   ld sp,hl

   ret
```

seedy1812, 2020-04-15:
The code I am writing moves a rectangle ( map ) about and I need to add new active area(s) and sub old unused area so I have switch statement ( produces better code this way )
```
	switch(dir)
	{
		case 0:
			MACRO_MAP_AddRegion(p[MAP_x].x,p[MAP_y].y,p[MAP_xw].x,p[MAP_Y].y);
			MACRO_MAP_SubRegion(p[MAP_X].x,p[MAP_yh].y,p[MAP_Xw].x,p[MAP_Yh].y);
				MACRO_MAP_AddRegion(p[MAP_x].x,p[MAP_Y].y,p[MAP_X].x,p[MAP_yh].y);
				MACRO_MAP_SubRegion(p[MAP_xw].x,p[MAP_Y].y,p[MAP_Xw].x,p[MAP_yh].y);
			break;
		case 1:
			MACRO_MAP_AddRegion(p[MAP_x].x,p[MAP_y].y,p[MAP_xw].x,p[MAP_Y].y);
			MACRO_MAP_SubRegion(p[MAP_x].x,p[MAP_yh].y,p[MAP_xw].x,p[MAP_Yh].y);
			break;
		case 2:

			MACRO_MAP_AddRegion(p[MAP_x].x,p[MAP_y].y,p[MAP_xw].x,p[MAP_Y].y);
			MACRO_MAP_SubRegion(p[MAP_X].x,p[MAP_yh].y,p[MAP_Xw].x,p[MAP_Yh].y);
				MACRO_MAP_AddRegion(p[MAP_Xw].x,p[MAP_Y].y,p[MAP_xw].x,p[MAP_yh].y);
				MACRO_MAP_SubRegion(p[MAP_Xw].x,p[MAP_Y].y,p[MAP_x].x,p[MAP_yh].y);
			break;
```
I can change the macro to try different things - pass by parameters or  to a fixed int16_t[4] array and not pass any parameter to he functions. If I use the later then I get better code generation with the array p being volatile
```
l_MAP_Update_00119:
;map1.c:316: MACRO_MAP_AddRegion(p[MAP_x].x,p[MAP_y].y,p[MAP_xw].x,p[MAP_Y].y);
	ld	bc,(_p)
	ld	(_rectangle), bc
	ld	bc,(_p + 0x0004)
	ld	(_rectangle + 0x0002),bc
	ld	bc,(_p + 0x0002)
	ld	(_rectangle + 0x0004),bc
	ld	hl,(_p + 0x000a)
	ld	(_rectangle + 0x0006),hl
	call	_MAP_AddRegion
;map1.c:317: MACRO_MAP_SubRegion(p[MAP_X].x,p[MAP_yh].y,p[MAP_Xw].x,p[MAP_Yh].y);
	ld	bc,(_p + 0x0008)
	ld	(_rectangle), bc
	ld	bc,(_p + 0x000c)
	ld	(_rectangle + 0x0002),bc
	ld	bc,(_p + 0x0006)
	ld	(_rectangle + 0x0004),bc
	ld	hl,(_p + 0x000e)
	ld	(_rectangle + 0x0006),hl
	call	_MAP_SubRegion

```
With it not being volatile I get a few edge cases

```
;map1.c:316: MACRO_MAP_AddRegion(p[MAP_x].x,p[MAP_y].y,p[MAP_xw].x,p[MAP_Y].y);
	ld	hl,_rectangle
	ld	a,(ix-3)
	ld	(hl), a
	inc	hl
	ld	a,(ix-2)
	ld	(hl), a
	ld	bc,(_p + 0x0004)
	ld	(_rectangle + 0x0002),bc
	ld	bc,(_p + 0x0002)
	ld	(_rectangle + 0x0004),bc
	ld	hl,(_p + 0x000a)
	ld	(_rectangle + 0x0006),hl
	call	_MAP_AddRegion
```

aralbrec, 2020-04-15:
That's much better :)  I'm a little sad that it's using BC for the 16-bit loads and saves as that takes an extra 4 cycles over using HL but that is something that can be fixed.

spth, 2020-06-08:
> This has been a bit of a long-standing annoyance with sdcc for a while. Philip doesn't seem to be taggable so maybe the only way to address it is on sdcc's bugtracker.

Shouldn't @spth work (but I am not that familiar with GitHub, as most of my work is done in svn)?