z88dk/z88dk #1442: (classic) ZXN support

suborb, 2020-04-13:
I've started so I might as well finish....

- [x] Classic ZX screen graphics support
- [x] Timex lores screens graphics support
- [x] Timex hires screen graphics support
- [ ] Tilemap graphics
- [ ] Level2 graphics
- [ ] Level2 gencon
- [ ] .dot file creation

Alongside this sort the following out:

- [ ] TS2068 portable graphics support (#1274)
- [ ] Non-ROM scrolling for +zx/+ts2068 gencon

Fabrizio-Caruso, 2020-04-15:
The very latest snapshot as of todat 15-05-2020, has no CONIO support if compiled with the same options that work with CONIO-enabled targets:
```
$ make zxn
/cygdrive/c/z88dk/bin/zcc.exe +zxn -Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_data -Icross_lib/display/graphics_data/cc65 -Icross_lib/display/graphics_data/z88dk -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/display/image_settings -Icross_lib/display/redefine_characters -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/psg -Icross_lib/text -Icross_lib/input   \
-D__ZXN__ \
-DFORCE_XSIZE=32 \
-DFORCE_YSIZE=24 \
-DNO_SET_SCREEN_COLORS \
-DNO_COLOR \
-DFULL_GAME  \
-DCONIO_VT52 \
-DCONIO \
-lndos \
-DNO_INIT_GRAPHICS \
cross_lib/display/display_macros.c cross_lib/display/init_images.c cross_lib/input/input_macros.c cross_lib/text/text_macros.c chase/move_player.c chase/ghost.c  chase/level.c chase/character.c chase/game_text.c chase/main.c chase/strategy.c chase/item.c chase/skull.c chase/bullet.c   chase/horizontal_missile.c chase/rocket.c  \
-create-app -o../build/FULL_zxn.bin
In file included from cross_lib/cross_lib.h:116,
                 from cross_lib/display/display_macros.c:23:
cross_lib/display/display_macros.h:34:20: fatal error: conio.h: No such file or directory
compilation terminated.
make: *** [Makefile:7622: zxn] Error 1
```

suborb, 2020-04-15:
Because this target was supported by newlib initially, newlib is the default target. Thus, ou have to specify the option `-clib=classic`: this is mentioned on the wiki page for the example compilation: https://github.com/z88dk/z88dk/wiki/Platform---Sinclair-ZX-Spectrum-Next

For ease of testing in emulators (and it also works on the real hardware), you should also add the option `-subtype=nex` to create a .nex output file. This is mentioned on the wiki page for the example compilation: https://github.com/z88dk/z88dk/wiki/Platform---Sinclair-ZX-Spectrum-Next

closed 2021-05-20 by suborb