z88dk/z88dk #1451: (z80asm) Linking very slow on windows

suborb, 2020-04-25:
There's a forum thread here: http://specnext.com/forum/viewtopic.php?f=17&t=1667

Which I think shows that linking is extremely slow when using newlib and windows. If my supposition is correct (is it @pauloscustodio?) then fiixing it for windows should make *nix linking even faster.

>Running on a virtualised windows 10 (i5-6260U), the command supplied runs in about 14 seconds.
>
>Compiling with -clib=new (so using sccz80) takes about 9 seconds.
>
>Compiling with zcc +zxn -clib=classic hexdump.c -o hexdump -subtype=dot -create-app  -lesxdos -v - so using the classic library to create a dot file compiles in 3.8 seconds. 
>
>Running on my ancient MacBook (i7-3557U), the times are: 3.8, 1.8 and 0.75 seconds respectively which is altogether much more acceptable.
>
> Looking at the log as it's printing to screen, it appears that most of the time is being spent at the linking stage. What I suspect is happening is that z80asm is searching for library routines by opening the library file, scanning through for the routine, closing the file and doing it again for the next library routine. On *nix this behaviour goes un-noticed, but because (in my limited experience) Windows is horrendously slow at file IO and newlib is decomposed into lots of small routines it becomes really noticeable.

pauloscustodio, 2020-04-25:
> What I suspect is happening is that z80asm is searching for library routines by opening the library file, scanning through for the routine, closing the file and doing it again for the next library [...]

This is exactly what is happening.

The linking stage is the most arcane part of z80asm, and is written for an era of small memory sizes and all linking being done on disk - I have worked mostly on the parser and code generator. I guess this issue might nudge me in trying to make it use more memory and less file IO.

aralbrec, 2020-05-07:
It has in fact gotten a lot worse.  I've just updated to the current version and am building the newlib under windows and it must be an order of magnitude slower than 1.99C.

suborb, 2020-05-07:
I think we might be able to a temporary patch using fmemopen() equivalence. That is we read in the library file initially and then fmemopen() to the memory associated with it. Then when we're scanning through the library file we're just looking in memory rather than hitting disc.

It's a bit kludgey, but should speed things up a bit on windows.

Of course, on windows it's even kludgier since fmemopen() doesn't exist, but https://github.com/Snaipe/fmem/blob/master/src/fmem-winapi-tmpfile.c should help us.

pauloscustodio, 2020-05-07:
Would it help to do it in a more portable way of allocating memory for the whole library file and reading it in one read system call?

suborb, 2020-05-07:
That’s what I was thinking - read it all into memory. The fmemopen stuff allows us to use the regular object file parsing utilities minimising the code changes.

pauloscustodio, 2020-05-07:
The difference being that fmemopen uses the virtual memory paging system, and malloc still needs to copy from file to memory.
I'll have a try.

pauloscustodio, 2020-05-11:
Unrelated to the slowness of linking, but caught me doubting my configuration and re-checking with the current nightly build: z88dk-win32-20200511-caf4993-16292.zip

The appmake stage fails.
```
$ time zcc +zxn -v -startup=4 -clib=sdcc_iy hexdump.c -o hexdump -subtype=dotn -create-app
...
appmake +zxn --dotn  -b "hexdump" -c "C:\msys64\tmp\zcc23F03"
Error: Section MAIN overlaps section code_main by 874 bytes
zxn: Aborting... one or more binaries overlap
Building application code failed

real    1m9.674s
user    0m0.015s
sys     0m0.031s
```

suborb, 2020-05-11:
I think it might be your environment? I'm using git-bash for reference:

```
dom@DESKTOP-TCFK71R MINGW64 /c/z88dk
$ zcc +zxn -startup=4 -clib=sdcc_iy hexdump.c -o hexdump -subtype=dotn -create-app -m
Notice: Main binary occupies [32768,36842]
Notice: Space to end of dot is 5158 bytes
Notice: Main bank allocation mask is 0xf0
Notice: Main bank load mask is 0x10
Page 0, main bank allocate
Page 1, main bank allocate
Page 4, main bank allocate and load
Page 5, main bank allocate
dotn_last_page = 223
dotn_num_extra = 0

dom@DESKTOP-TCFK71R MINGW64 /c/z88dk
$ ls -al bin/sccz80.exe
-rwxr-xr-x 1 dom 197121 802096 May 11 04:18 bin/sccz80.exe*

dom@DESKTOP-TCFK71R MINGW64 /c/z88dk
$ date
Mon, May 11, 2020  7:50:25 PM
```

pauloscustodio, 2020-05-11:
You're right, fixed after a make clean and re-build.

closed 2020-05-14 by pauloscustodio

aralbrec, 2020-05-15:
The linking speed on windows has sped up quite a bit, cheers!

commit 8208c93a (2020-05-13): Fix #1451 - link very slow on Windows

Slowness caused by z80asm opening, searching and closing the library
file for each unresolved symbol, repeating the process until all
references are solved. This is a leftover from a past without virtual
memory and so that no object file is loaded to main memory unless
it is actually linked in (and to resolve symbols depth-first).

The problem is exacerbated by the slow file I/O of Windows and by
the 760 sections and 141 library modules linked in, and 581 binary
files created by the hexdump.c build (add -z80-verb option to zcc to
see this).

Solved by loa [... cut]
commit a70942ad (2020-05-14): Fix #1451 - link very slow on Windows

Slowness caused by z80asm opening, searching and closing the library
file for each unresolved symbol, repeating the process until all
references are solved. This is a leftover from a past without virtual
memory and so that no object file is loaded to main memory unless
it is actually linked in (and to resolve symbols depth-first).

The problem is exacerbated by the slow file I/O of Windows and by
the 760 sections and 141 library modules linked in, and 581 binary
files created by the hexdump.c build (add -z80-verb option to zcc to
see this).

Solved by loa [... cut]
commit 716cb808 (2020-05-14): Merge pull request #1472 from z88dk/z80asm_1451

Fix #1451 - link very slow in Windows