z88dk/z88dk #1454: (classic) Add support for Specialist

suborb, 2020-04-27:
Another USSR monitor machine:

- 8080
- 34k ish RAM
- 384x256 pixel mapped B&W display\

Possible:

- [ ] gencon with fonts
- [ ] firmware output
- [ ] firmware input
- [ ] hires graphics
- [ ] inkey keybard
- [ ] interrupt?
- [ ] sound?
- [ ] Other models

Currently struggling to drive the monitor program. "I" then enter a filename but the cursor doesn't stop blinking.

Tech info: http://emu80.org/dev/dev_s.html

Fabrizio-Caruso, 2020-06-02:
How can I read the keyboard with this target?

suborb, 2020-06-02:
I have to admit I got rather fed up with this one so I put it on ice and didn't announce it.

There's no way of polling the keyboard - the only input is `fgetc_cons()` so it will need to be a turn based variant.

Fabrizio-Caruso, 2020-06-02:
@suborb Thanks! That is good for me. I can test this target in turn-based mode.