z88dk/z88dk #1456: [m100] Does this target support the Olivetti M10?

Fabrizio-Caruso, 2020-04-28:
I supposed the TRS-80 M100 target only covers the TRS-80 M100 and binary compatible systems.
So the Olivetti M10 is not cover even though it has the very same hardware with a different mapping in memory for some of the chips: 
https://www.old-computers.com/museum/computer.asp?st=1&c=475
Unless your crt normalizes the mapping of all the TRS-80 M100-based hardware models so that the binary is compatible with all of them. Is this the case?

Fabrizio-Caruso, 2020-05-03:
@zx70 do you know if this taget coves the Olivetti M10 as a sub-target? I know it is not binary compatible with the TRS-80 M100 target but it uses the very same hardware which is mapped in memory at different addresses.

zx70, 2020-05-05:
The idea for this port is to include the whole family, probably keeping aside as a special case the NEC pc8201 & 8300.
At the moment I'm keeping this port on hold,  and it is not working.
The only interesting result at the moment is the ROM disassembly, which can be used to produce binary-exact ROMS for the M10, M100 or KC85 with the same source file.   It is a valuable cross-reference to get exactly what you are asking for. 
I'm sorry but at the moment I'm keeping this port on hold.

closed 2020-06-28 by suborb