z88dk/z88dk #1457: [laser500][appmake] 44khz rate seems too much for real hw but downsampling to 22khz fixes it

Fabrizio-Caruso, 2020-04-30:
@LoudScotsBloke has tested the wav file that appmake produces for my game:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/releases/download/I8080/FULL_vz500.wav
which I do not compile as a normal uncompressed wav:
```
	$(Z88DK_PATH)$(MYZ88DK) +laser500 -vn -create-app -Cz--audio \
...
	-lndos \
	-create-app  \
...
```

He has only managed to load the game on real hardware by downsampling the wav to 22khz with Audicity.This works for him consistently (3 times in a row).

On the other hand, using the original wav just works on the emulator but always produces errors on real hardware
[image]

suborb, 2020-04-30:
This one was tested on real hardware, see: https://github.com/z88dk/z88dk/issues/854#issuecomment-418540565

So I don't know what's happened or why downsampling makes it work.

Fabrizio-Caruso, 2020-05-01:
@suborb I can close this issue if you want. It is more a feature request than an issue.
Indeed the current wav is good and can be loaded on the real hardware (depending on the method and tools used with some luck and by tinkering with the volume).
I suppose that lower sampling rate means more reliability because you are compressing fewer sounds in the same time unit.
Is it possible to tell the compiler to select a lower sampling rate? This would be a useful feature for all targets that produce wav files.

@LoudScotsBloke could give us more details about his "fix" for the wav file.

zx70, 2020-05-09:
If you are referring to 22050 samples, the only expected feature is an accuracy loss.  I had negative feedbacks on the format I chose, though.  It is a peculiar one, permitting to declare CD quality but encoded with 8 bits rather than 16 which makes it highly compressable (e.g. in zip format).  When converted with a tool like Audacity in some case this helps its portability (in example it looks like the Spectaculator emulator is in trouble with it, while EightyOne works fine).
A new audio format option is possible, but a lower rate reduces the frequency accuracy in many cases (all those frequencies requiring an odd number of samples).   Coul you first try to test an Audacity converted file where the output rate is unchanged ?

LoudScotsBloke, 2020-05-09:
Hi, just to confirm what I did, I loaded the original WAV file in to Audacity, changed the sample rate from 44 to 22, then just exported the WAV file.

zx70, 2020-05-09:
it was clear to me
I was suggesting to try to export a wav file leaving it at 44
I had claims regarding the way the wav file was encoded and I need to be sure that the sample rate is the real problem

LoudScotsBloke, 2020-05-10:
Hi. I tried exporting at 44, and I got the same didn't load issue. I also tried amplifying the native wav at 44, still no joy. Non amplified, exported @ 22, and it loaded fine.

zx70, 2020-05-10:
Thank you for this extra test, 
this is something I do not fully understand, especially considering that the exact frequencies we're talking about are 44.100 and 22.050, right?  It is a matter of just halving the periods and declaring a slower rate.   It is worth to try to add a valid alternative,  let's see what will come out.

zx70, 2020-05-10:
Fabrizio, while I'm working on it you could try also a wav format variant which is already avalable:  -Cz--fast

zx70, 2020-05-11:
New option:  -Cz--22
Downscaling requiem a bit of approximation band canbe Itunes in the ways.   Coul you please try if we're lucky with my first guess?

zx70, 2020-05-15:
For your information, I just discovered something strange is happening also on the ZX Spectrum port (#1473).  It shouldn't be related, especially because editing the sound file fixes your issue, but I'm not 100% sure.

zx70, 2020-05-18:
J.ACE and the whole VZ and NEC target families are now support the 22khz downscaling option.
The wave timing periods are slightly cropped, which probably has no relevant impacts on the results but could be useful for fine tuning in some real or emulated case.

closed 2020-06-07 by zx70