z88dk/z88dk #1458: classic cp/m - bdos read & write sequential

feilipu, 2020-05-04:
In another place, [there was a discussion](https://groups.google.com/d/msg/retro-comp/rNpTvk5v9Lg/_9z5yY4oAgAJ) comparing 256 SPT with 32 SPT in CP/M. I did some bench marking and found there is quite a bit of overhead (but that is all beside the point).

In summary, using constant hardware (RC2014 + 82C55 IDE + SDD DOM) for a 1MB file copy I got:
 - Raw I/O time for sector copy: 14 seconds (theoretical).
 - z88dk + fatfs (on FAT32) for reference: 21.5 seconds.
 - Raw I/O + CP/M sector deblocking: 24 seconds (theoretical).
 - CP/M using PIP: 42 seconds.

So 18 seconds of internal housekeeping for PIP. Obviously I can do better with C and the z88dk classic library...

 - CP/M using crafted C program: 53 seconds. 

...What? How can my very slim simple C copy program be beaten by a general purpose tool from 1979?

So I've read the PL/M code tonight, and I think the reason is that in certain circumstances PIP uses the BDOS Function 20 READ SEQUENTIAL and Function 21 WRITE SEQUENTIAL.

Question. Does the classic library implement a class of functions that give direct access to these faster BDOS calls? (I don't want to be beaten).

suborb, 2020-05-04:
Looking at this https://github.com/z88dk/z88dk/blob/master/libsrc/target/cpm/fcntl/read.c

If you read/write a block (i.e. not a byte) then it should be using those BDOS calls.

feilipu, 2020-05-04:
Wouldn't I be looking for [`CPM_READ`](https://github.com/z88dk/z88dk/blob/master/include/cpm.h#L98), or `CPM_WRITE` in the `bdos()` call, to use the sequential R/W functions? I can't see these two being used.

suborb, 2020-05-04:
Sorry you’re right. They use RRAN - I didn’t check the mappings. 

In that case. No.

feilipu, 2020-05-05:
Possibly, the cpm/fcntl functions could be extended to differentiate based on the [`fb->mode`](https://github.com/z88dk/z88dk/blob/master/include/cpm.h#L58) parameter to use either sequential or random read/write?

I can have a go at adding this sequential read/write capability. I think it would be useful.

Is there a sensible / preferred method to extend this mode flag?

feilipu, 2020-05-05:
Could read just be fixed by substitution of...
```c
            if ( size % SECSIZE == 0 ) {
                bdos(CPM_SDMA,buf);
                if ( bdos(CPM_READ,fc) ) {
                    return cnt-len;
		}
            } else {
```

suborb, 2020-05-06:
That's what I was suggesting with the "loop" comment - I've not looked at the code in great detail though - did you give it a go?

feilipu, 2020-05-06:
Yes, I am having a go at it, but slow progress because I’m hampered by a lack of practical knowledge about the BDOS api.

So I have written a small assembly copy program to get a feel for how BDOS works, and I’ll continue to extend it until I’m getting better than PIP outcomes.

I don’t know whether the idea of having a flag for sequential vs random write is a good or possible thing. Better if the logic in the function can work out the fastest way.

From what I’ve read (but haven’t proven with code), it should be possible to assume sequential mode for the first call, and then convert to random mode on subsequent calls (if they’re not sequential), using the BDOS function to convert to random mode.

But making that conversion assumes preserving state, which would have to go into the FCB. I’m not 100% sure whether the BDOS FCB has a place for this, but I think that it does, based on the availability of a function for that.

What my initial attempts have shown is that reading / writing 1 CP/M sector at a time is desperately slow, even in sequential mode. So multiple consecutive reads and writes are required for throughput.

The PIP program goes out of its way to merge its input and output buffers into a single larger buffer for fast copies. Otherwise it reserves two buffers for its translation activities. So it is clear that a large buffer for continuous reading and writing (even for modern SSDs) is important.

feilipu, 2020-05-07:
Notes on PIP PL/M source code - For reference.

- Declares the `0006` location as the address field to find maximum memory available.
- Overwrites CCP, as it is allowed to do.
- Makes sure it uses all free RAM for its sector buffers, up to a maximum of 32kB buffer pool.
- Relative location of last 128 Byte sector buffer is `7F80h`, which splits into 2 16kB buffer pools.
- There is a maximum of 255 sector buffers, which is length/128-1, when used as a single pool.

```
IF FASTCOPY THEN /* COPY DIRECTLY TO DBUFF */
        DO CALL SET$DBLEN; /* EXTEND DBUFF */
            DO WHILE NOT REAL$EOF;
            CALL FILLSOURCE;
            IF REAL$EOF THEN
                NDEST = HARDEOF; ELSE NDEST = DBLEN;
            CALL WRITEDEST;
            END;
```

feilipu, 2020-05-09:
Resolved with PR #1465.

Testing these small changes shows an improvement for the 1MB file copy from 53 seconds to 51 seconds, in direct comparison. I've not found any simple way to do better.

I think there's too much code between the `fread()` call and the eventual `read()` (and also for write case) to approach assembly calls to BDOS (39 seconds), or PIP (42 seconds). It is what it is.

feilipu, 2020-05-15:
Added [one further commit](https://github.com/z88dk/z88dk/commit/d7ebf1318108165351546980307b97b58add4938) to reduce the number of unnecessary BDOS calls.
This shaves another 2 seconds off the 1MB file copy, getting it down under 49 seconds.

But there is an issue remaining, and the issue was there before PR #1465. Because the BDOS READ and RRAN calls don't return the actual number of Bytes read, the `fread()` function guesses by [returning the number of Bytes it requested BDOS to read](https://github.com/z88dk/z88dk/blob/master/libsrc/stdio/fread.c#L132), rather than the number of Bytes actually read by BDOS. I think this is a bug.

The result is that the read size can't be less than a sector, and some empty bytes get added to the end of an irregular file copy.

suborb, 2020-05-15:
As far as I remember, CP/M doesn't have a file length indicator, just an extent count so binary files always have extra stuff on the end.

For text files we have to read byte/byte to check for the ^Z EOF marker.

feilipu, 2020-05-15:
Yes. Quite true. And the PR and commits haven't changed this behaviour.

But, that doesn't clarify for me how both `PIP` and my completely naive assembly [`cp`](https://github.com/z88dk/z88dk-ext/blob/master/os-related/cp.asm) program can get the Byte count from an irregular length file copy perfect, yet using the library an irregular length file has nulls appended at the end. Quite curious.

suborb, 2020-05-15:
How are you getting the file length?

feilipu, 2020-05-15:
Using a CP/M tool called ddir. Was trying to find it just now. I thought it was in z88dk-ext.

feilipu, 2020-05-15:
Yes, I knew I'd seen it around somewhere. #1251 refers to it.
It is in the [osca](https://github.com/z88dk/z88dk/blob/master/examples/osca/ddir.c) directory.
It provides (supposedly) a byte accurate measure, although for CP/M the comments refer to the end of the extent.

feilipu, 2020-05-16:
Further investigation by dumping the file and comparing before and after copy shows that the actual file information is identical.

It looks like what is happening is that the next file extent is being enabled before being written by classic library, but this is not being done by `PIP` or the naive `cp` program. So the resulting file looks to 128 Bytes longer than it really is.

But, the `hexdump` program doesn't output an extra 128 Bytes of null, so it is not being confused.
It looks like just the `ddir` program is being confused, somehow.

feilipu, 2020-05-16:
I'm taking off the bug label, because this is not a critical issue, but it is odd.

zx70, 2020-05-30:
Ddir is a conversionof mine of a tool originally written for the bds c.
It was surprisingly close to the msdos one.
I then expanded it to the OSCA target,  which suggested me to develop a compatible api between the two targets.    In com you can't determine the exact file length,  just the approx. Blocks number.   The programs able to do it use termination codes for specific file formats ,  like chr 26 on txt, or similar.

feilipu, 2020-06-02:
There remains an issue that a file being written has the number extents increased before they are written, and if the last extent is not needed, then the extent count is not reduced.

But, since no one reported this as a bug, and the behaviour has not been changed by this conversation, I will close this issue.

It remains as a record.

closed 2020-06-02 by feilipu

zx70, 2020-06-03:
So It is probably the classic fcntl engine or its drivers allocating a block before really using it.

suborb, 2021-04-25:
And this switching from random to sequential seems to be causing unusual results with this sequence:

```
  file=fopen("file.bin", "wb");
  memset(s1, 0x31, 1024);
  fwrite(s1, 21, 1, file);

  memset(s1, 0x32, 1024);
  fwrite(s1, 1001, 1, file);

  memset(s1, 0x33, 1024);
  fwrite(s1, 2, 1, file);
  fclose(file)

```