z88dk/z88dk #1466: math16 - half precision floating point - eoi

feilipu, 2020-05-12:
In the field of [analytics and deep learning](https://docs.nvidia.com/deeplearning/performance/mixed-precision-training/index.html) there's quite a bit of interest in using half precision floating point numbers. This [CUDA article describes the half precision background](https://devblogs.nvidia.com/mixed-precision-programming-cuda-8/).

The [cuda_fp_16.h](https://docs.nvidia.com/cuda/cuda-math-api/group__CUDA__MATH____HALF__ARITHMETIC.html#group__CUDA__MATH____HALF__ARITHMETIC) reference.

Obviously, a z80 is not the ideal platform for deep learning, but it is interesting that half floating point math would work well with 8 bit machines.

I'm willing to start on a library, but if it can't be integrated into the compilers, then there's no point.
And, I'm willing to be discouraged from this thought bubble is not really interesting.

Is there any interest or value in adding a half precision maths library to z88dk?
Other thoughts?

suborb, 2020-05-12:
I’ve had some thoughts about this a few months ago. I remember finding the core of a 16 bit library. 

 Happy to play with sccz80 and add a float16 type so it can be used natively and we can compare/contrast performance vs allocating space for a float32

feilipu, 2020-05-12:
[Here's one starting point](https://github.com/artyom-beilis/float16).

suborb, 2020-05-12:
I've pushed some changes to sccz80 to support -fp-mode=ieee16 to the feature/maths_binary16 branch.

The format we're generating is the IEEE binary16 format: sign bit is MSB, 5 bits of exponent, 10 of mantissa.

I'm not happy with the ABI at the moment, but it keeps it semantically the same as the other maths libraries. Compared to the math32 work this pushes 16 bits onto the stack instead of a 32bit number. A fastball value for the compiler support routines is passed in hl.

feilipu, 2020-05-12:
I'll get started then... No excuses now.
Probably will go with `math16`, in line with `math32` and `math48`.

feilipu, 2020-05-19:
It may look like I’m not making progress with this but I have been thinking about the best way to do it. I’m still not 100% fixed, but here’s some thoughts.

The Z80 is pretty good at handling 8 bit operations, and float16 having the odd 5 bit exponent and 11 bit mantissa means that natural zero, overflow, and sign register handling has to be replicated with messy bit checking. Also, thinking about z180 and z80n there is also a tendency for me to do things in `16_8x8` to maximise their multiply hardware.

Also, remembering the hell of trying to create enough accuracy to do polynomial evaluations (for the higher functions) with math32, I’m inclined to do something a bit different to what has been done before, so that there’s at least some chance these derived functions will work.

Currently, my thinking is to do all calculations on an internal float25 format, using `DEHL` with `D[7]` sign bit. `E` for the unbiased 8 bit exponent, and `HL` for 15+1 bit mantissa. The ABI would use the same main and alt register set as per float48 and float32.

Therefore (except simple exponent scaling or exception handling), all calculations would have the range of `float32` and have a mantissa that maximised the “free” accuracy of the z180, z80n hardware.

Having the extended range for internal calculations would minimise the need to handle sub-normals. Handling sub-normal numbers is a pain I can do without, and I’ll probably drop that feature (at least in the interim). It can be added back in the output conversion routine, later

Holding the extended mantissa range until a common conversion / denormalisation / output routine would centralise rounding and denormalisation code into one routine, (that someone who knows math better could mathematically fix if it were too broken). Noting that the paranoia floating point test program flags issues with both float32 (critical) and float48 (major).

Anyway, still thinking / plotting (and dealing with irl distractions).

suborb, 2020-05-19:
>  Noting that the paranoia floating point test program flags issues with both float32 (critical) and float48 (major).

I don't think I'd run paranoia before. And it's not really cheered me up either.

genmath doesn't complete: sqrt() seems to be pretty broken
bbcmath: completes with 3 serious, 4 defects, 2 flaws
math48: completes with 3 serious, 1 defects, 2 flaws
mbf32: doesn't complete, radix is recalculated as 0, sqrt is  then slightly broken (possibly comparison routines?)
math32: completes with 2 failures, 6 serious, 8 defects, 3 flaws, number printing is broken.

I'll try and nurdle genmath through the suite and see where we stand. Given that mbf32 is our only 8080/gbz80 option I'll see what I can do there.

I'm guessing the math32 printing issue will be related to the some of the reported issues around multiplication - I do remember having issues with printing whilst you were writing math32

----
I think internally using 16+8 (if I understand you correctly) makes sense to leverage the limited CPU features we've got at our disposal whilst being quicker than math32.

feilipu, 2020-05-20:
> I don't think I'd run paranoia before. And it's not really cheered me up either.
It made me 😢 too. I don't understand the results well enough to dive into which "bit" of the `math32` code is wrong. But I suspect that rounding is a key failure.

Probably a separate issue (or issues) on getting them (all) through paranoia.

suborb, 2020-05-22:
A z80 24 bit float implementation: https://github.com/Zeda/z80float/tree/master/f24 - there's a few other different sizes in that repo as well.

feilipu, 2020-05-29:
This is a bit frustrating. The [add / subtract](https://github.com/z88dk/z88dk/pull/1471/files#diff-5ebb3b02f3e7e7620af8282217b49b67) and [multiply](https://github.com/z88dk/z88dk/pull/1471/files#diff-b153f067315350d952ed2d4b451b1d3a) functions are working "perfectly". But using the NR process for [division](https://github.com/z88dk/z88dk/pull/1471/files#diff-c8a153f26b01a57bd03a4b0e30dae23a) doesn't converge. Staring harder at it doesn't seem to help.

I think I'll go around and fix other things, and come back to division / inversion.

feilipu, 2020-06-05:
So it is done. Just need to test the linkages, and then do a few more functions (at some later date).

feilipu, 2020-06-07:
@suborb I think that the `math16` library has to be an adjunct maths library, rather than "the" maths library. There's no point using it to do trig / log functions, etc with 3.5 significant digits. So, it won't ever be a full replacement for `math48` or `math32`.

But the 4 standard arithmetic functions, and comparison functions are useful, and can be used to do the high speed things for which a half float it is suited.

This means that two maths libraries will need to be active at one time. Not least to print the "half" type numbers after converting them to 32-bits. (or __TO DO__ 48-bits).

So, somehow the compiler needs to know when to use the "real math" library for an addition or a comparison, and when to use the "half" float library. The only way I can think of for this to happen automatically is if the "half" type were recognised by the compiler, so the appropriate intrinsic function could be selected.

So, I'm questioning how to make this new `math16` library useful?

suborb, 2020-06-07:
I suspect for sccz80 adding a distinction between float and double would help here.

feilipu, 2020-06-07:
I've merged the PR #1471, for further testing etc.

__TO DO__
- f48 to f16 fo f48 conversion
- sqrt function

Work out what to do with this now  :shrug: :man_shrugging: :woman_shrugging:

suborb, 2020-06-07:
I'm starting to form a few ideas as to what to do...

I can see a point in the trig functions - I feel they would be quite useful as part of a display driver: I'd be interested to see how cube.c performs for example. For the size of screens we deal with, the coordinates can be exactly represented by a half.

I'm turning against the idea of supporting a distinction between float and double in sccz80 - it won't work with zsdcc. So int terms of printing, I suspect the easiest way is to introduce a new format specifier so that two float libraries can be used together.

feilipu, 2020-06-07:
To to the trig / log functions just the `poly()` function is missing.
Will only take an evening to add that in too, then trig comes for "free".

suborb, 2020-06-07:
I think you may have the sccz80 ABI wrong. The right hand parameter is passed in registers, with the left hand on the stack - I think you've implemented the sdcc ABI where all parameters are passed on the stack.

feilipu, 2020-06-07:
Yes. I did.

I "know" the sccz80 ABI is left-stack, right-registers. But, when I was testing, that solution wasn't working and I thought somehow I had phantom memory. When I changed it to be left, right, return on the stack it started working.

Probably a fortuitous case of two mistakes cancelling each other out.
I'll fix it.

For the `fma()` function with three parameters though, all three are on the stack, right?

feilipu, 2020-06-08:
I'm definitely getting two sccz80 parameters being provided on the stack.
Testing using this code.
```sh
zcc +hbios -clib=new -v --list -m -SO3 --c-code-in-asm --math16 --math32  @ftest180.lst -o ftest -create-app
``
@suborb Do you get the same effect?

```C
union floatlong
{
    float f;
    int32_t l;
};

void main(void)
{
    float m = 0;
    float n = 0;
    
    half_t hm, hn;
    union floatlong resulta_m32, resultd_m32;

    resulta_m32.l = 0;
    resultd_m32.l = 0;
    
    for (m =  0.05; m < 1000.0; m *= 1.5)
    {
        for (n = -10.0; n < 100.0; n += 3.3)
        {
            hm = m16_f16_f32 (m);
            hn = m16_f16_f32 (n);

            resulta_m32.f = m16_f32_f16 ( m16_sub ( hm, hn ) );

            resultd_m32.f = (m-n);

            printf("  m %.4g", m);
            printf(" n %.4g", n);
            printf(" a_m32: %.4g", resulta_m32.f);
            printf(" d_m32: %.4g\n\n", resultd_m32.f);
        }
        printf("\n");
    }
}
```
With sccz80 I get this code, which works with the abi the way it is (and doesn't with the "correct" abi).
```asm
114   00CD              ;            resulta_m32.f = m16_f32_f16 (  m16_sub_callee( hm , hn )  );
114   00CD              	C_LINE	115,"./ftest.c"
115   00CD              	C_LINE	115,"./ftest.c"
115   00CD  21 04 00    	ld	hl,4	;const
115   00D0  39          	add	hl,sp
115   00D1  E5          	push	hl
115   00D2              ;m16_sub_callee( hm , hn )  ;
115   00D2              	C_LINE	116,"./ftest.c"
116   00D2              ;hm ;
116   00D2              	C_LINE	116,"./ftest.c"
116   00D2  21 0C 00    	ld	hl,12	;const
116   00D5  39          	add	hl,sp
116   00D6  CD 00 00    	call	l_gint	;
116   00D9  E5          	push	hl
116   00DA              ;hn ;
116   00DA              	C_LINE	116,"./ftest.c"
116   00DA  21 0C 00    	ld	hl,12	;const
116   00DD  39          	add	hl,sp
116   00DE  CD 00 00    	call	l_gint	;
116   00E1  E5          	push	hl
116   00E2  CD 00 00    	call	m16_sub_callee
116   00E5  CD 00 00    	call	m16_f32_f16
116   00E8  C1          	pop	bc
```

feilipu, 2020-06-08:
Just in terms of printing half-floats, I don't think that it is an issue worth building into the compiler via a special format conversion. It is no hardship to do a cast or conversion conversion from `half_t` to `float_t` before printing. This doesn't need to be "fast". Perhaps the casting could be made automatic by the compiler by integrating the conversion routines into the abi (like `f2sint`, etc), though?

What I feel is more important is to be able to write natural C expressions using `half_t` variables and have the constants be presented in `half_t` too. And then to build `for` and `while` loops, and `if` statements (generally comparisons) and have the numbers stay in their natural half float format.

That way a complicated "something" can be calculated quickly, and then the final result is printed either to screen or console if desired.

suborb, 2020-06-08:
> ANOTHER EDIT noted that the call preamble for m16_sub_callee and l_f16_sub are different. I was (am) watching the m16_sub_callee code, and that seems to be the problem...?

Yes, that's it. I was referring to the l_f16_ functions.

suborb, 2020-06-08:
> What I feel is more important is to be able to write natural C expressions using half_t variables and have the constants be presented in half_t too. And then to build for and while loops, and if statements (generally comparisons) and have the numbers stay in their natural half float format.

I think this means that we're going to end up supporting `_Float16` - let me look into it and see if I can find a free download of the specification.

feilipu, 2020-06-08:
> Yes, that's it. I was referring to the l_f16_ functions.

Ok. I think I [fixed it](https://github.com/z88dk/z88dk/commit/df5d0deaaf52c4969eadb31ad84a547355abdddf) (or returned it to working).

That makes 3 special cases for my personal gotcha list.  :safety_vest: :warning:
- fastcall calling for functions that are not named as such.
- function name mangling, that doesn't mangle.
- abi varies depends on whether intrinsic or not.

feilipu, 2020-06-09:
>  Let me look into it and see if I can find a free download of the specification.

The [Intel BFLOAT16 Hardware Numerics Paper](https://software.intel.com/sites/default/files/managed/40/8b/bf16-hardware-numerics-definition-white-paper.pdf).
A free copy of [IEEE754-2008](http://irem.univ-reunion.fr/IMG/pdf/ieee-754-2008.pdf).
Some [words on fp16 here, by Cleve Molar](https://blogs.mathworks.com/cleve/2019/01/16/variable-format-half-precision-floating-point-arithmetic/).

suborb, 2020-06-09:
I found a draft of the spec I was looking for: http://www.open-std.org/JTC1/sc22/wg14/www/docs/n1778.pdf

I've pushed some support for sccz80/_Float16 into feature/sccz80_float16_type - very much a WIP.

feilipu, 2020-06-10:
I've [converted the abi](https://github.com/z88dk/z88dk/commit/040798a933b7b230cf319c793e2631c4d7ced2a1) to `f16_function` from `m16_function`, which works better.

Added `l_f16_fabs` and `l_f16_neg` to the sccz80 abi to go with the n1945 document.

feilipu, 2020-06-10:
>  `sccz80_float16_type` very much a WIP.

Ok to try?

suborb, 2020-06-10:
> That makes 3 special cases for my personal gotcha list. 🦺 ⚠️
> 
> * fastcall calling for functions that are not named as such.

Oh, which ones?

> * function name mangling, that doesn't mangle.

I'm not sure what you mean here?

> * abi varies depends on whether intrinsic or not.

Yeah, sorry about that. Basically there's a concept of "floating point accumulator" which is either the alternate register set or the variable `FA` for > 32 bit maths libraries and a subset of `dehl` for smaller types. The math32 ABI is the same as for `longs.

Ideally for math16 I'd want to use `hl` as LHS and `de` as RHS to match the `int` ABI but I think that can be left for a later optimisation. Arguably, `bc`,`de` would be better register choices but I think that ship sailed a few decades ago.

feilipu, 2020-06-11:
> _Float16 - there's just no need, and as you suggested, a big and a small library are complementary.

Ok that's perfect. The half library is just an adjunct math library, and shouldn't conflict.

I was hoping to use a more "regular" `stdint.h` like `typedef` for this type, something like `float16_t` or `f16_t` or even (paralleling the double) `half_t`. I couldn't see a standard mentioned in the documents, though.

What am I casting float 16 numbers with in the current implementation?

Not to concerned about the ABI at the moment. I think it is good the way it is. Having the large number of registers free allowed me to use the "expanded" `f24` format for all the calculations, and saves time packing and unpacking.

suborb, 2020-06-11:
It’s an actual type. See https://github.com/z88dk/z88dk/blob/feature/sccz80_float16_type/testsuite/Issue_1466_float16.c for some test cases. So in sccz80 use is very natural. 

I don’t think sdcc supports it at the moment so that would have to use a half_t based interface

feilipu, 2020-06-11:
Ok. I'll rework the math.h to suit `_Float16`, and typedef it in [`math.h`](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/math.h#L57) too.

feilipu, 2020-06-11:
So some successes.
I'm still having trouble with my implementation of `l_f16_xxx`.  The call (below) looks right with LHS on stack and RHS in `HL`. But, it doesn't work. However the `f16_xxx` call works properly.

Casting works properly. Haven't tested the comparisons yet.

```
124   00BD              ;            hm = (half_t)m;
124   00BD              	C_LINE	126,"./ftest.c"
126   00BD              	C_LINE	126,"./ftest.c"
126   00BD  21 1A 00    	ld	hl,26	;const
126   00C0  39          	add	hl,sp
126   00C1  E5          	push	hl
126   00C2  21 24 00    	ld	hl,36	;const
126   00C5  39          	add	hl,sp
126   00C6  CD 00 00    	call	l_glong
126   00C9  CD 00 00    	call	l_f32_ftof16
126   00CC  D1          	pop	de
126   00CD  CD 00 00    	call	l_pint
126   00D0              ;            hn = (half_t)n;
126   00D0              	C_LINE	127,"./ftest.c"
127   00D0              	C_LINE	127,"./ftest.c"
127   00D0  21 18 00    	ld	hl,24	;const
127   00D3  39          	add	hl,sp
127   00D4  E5          	push	hl
127   00D5  21 20 00    	ld	hl,32	;const
127   00D8  39          	add	hl,sp
127   00D9  CD 00 00    	call	l_glong
127   00DC  CD 00 00    	call	l_f32_ftof16
127   00DF  D1          	pop	de
127   00E0  CD 00 00    	call	l_pint
127   00E3              ;            r = hm * hn;
127   00E3              	C_LINE	129,"./ftest.c"
129   00E3              	C_LINE	129,"./ftest.c"
129   00E3  21 1C 00    	ld	hl,28	;const
129   00E6  39          	add	hl,sp
129   00E7  E5          	push	hl
129   00E8  2B          	dec	hl
129   00E9  2B          	dec	hl
129   00EA  CD 00 00    	call	l_gint	;
129   00ED  D5          	push	de
129   00EE  E5          	push	hl
129   00EF  21 1E 00    	ld	hl,30	;const
129   00F2  39          	add	hl,sp
129   00F3  CD 00 00    	call	l_gint	;
129   00F6  CD 00 00    	call	l_f16_mul
129   00F9  D1          	pop	de
129   00FA  CD 00 00    	call	l_pint
129   00FD              ;            resulta_m32.f = (float) r;
```

suborb, 2020-06-11:
> ```
> 129   00EA  CD 00 00    	call	l_gint	;
> 129   00ED  D5          	push	de
> 129   00EE  E5          	push	hl
> 129   00EF  21 1E 00    	ld	hl,30	;const
> 129   00F2  39          	add	hl,sp
> 129   00F3 [...]

That `push de, push hl` is wrong. The push de is erroneous, it looks like it thinks it's a float32. Can you post the original code so I can add it as a test case.

feilipu, 2020-06-12:
Confirmed working.

I'm still having trouble with the `l_f16_div` function not working (`f16_div` works). Somehow there's something being overwritten, but I can't see it.

Otherwise all good :+1:

feilipu, 2020-06-12:
I've checked pre and post (dec-) increment. Working at first pass :+1:

Checked conditionals. Not working.
Where `r` is a `_Float16`, and using above codebase, it is still picking it as a 4-byte float.

```
149   015D              ;            if ( r > 0 )
149   015D              	C_LINE	150,"./ftest.c"
150   015D              	C_LINE	150,"./ftest.c"
150   015D  21 1C 00    	ld	hl,28	;const
150   0160  CD 00 00    	call	l_gintspsp	;
150   0163  21 00 00    	ld	hl,0	;const
150   0166  CD 00 00    	call	l_f32_gt
150   0169  7C          	ld	a,h
150   016A  B5          	or	l
150   016B  CA 78 01    	jp	z,i_8
150   016E              ;                printf("  yes!");
150   016E              	C_LINE	151,"./ftest.c"
151   016E              	C_LINE	151,"./ftest.c"
151   016E              ;"  yes!";
151   016E              	C_LINE	152,"./ftest.c"
152   016E  21 2A 00    	ld	hl,i_1+42
152   0171  E5          	push	hl
152   0172  3E 01       	ld	a,1
152   0174  CD 00 00    	call	printf
152   0177  C1          	pop	bc
152   0178              ;        }
152   0178              	C_LINE	152,"./ftest.c"
152   0178              .i_8
152   0178  C3 93 00    	jp	i_5
```

suborb, 2020-06-12:
Thanks for the heads up, pretty simple fix for both divide and comparison. The comparison is dumb, I set the return type to INT before calling the operation so it just defaults to calling a DOUBLE routine. The div problem just modifies the stack incorrectly.

I'm going to add math16 to the standard classic z80 library and get the maths test suites running which will help winkle out these issues a lot quicker than eyeballing.

I've also got to sort out static initialisation - I've not touched it so it's completely wrong at the moment.

suborb, 2020-06-12:
Getting there - the comparison functions seem to work - they pass the tests in `test/suites/math`. I'll push later this evening with various fixes. However I've run into an accuracy problem I wasn't expecting. 

```
void func()
{
        _Float16 a = 8;
        _Float16 b = 4;
        printf("%04x\n", *(int *)&a);
        a /= b;
        printf("%04x\n", *(int *)&a);
}
```
Yields:

```
4800
3fff
```
This is really noticeable since using `a /= 4` uses ldexp, so the answer `4000`

feilipu, 2020-06-12:
Truncates without rounding. It comes from the format conversion process. 

I had rounding in initial commits, but it was ill considered. So removed until properly thought through.

Issue was simple rounding through two conversions was promoting a bit up to 16 positions. To much.

suborb, 2020-06-12:
In this case the constants are being generated by the compiler so there's no format conversion going on.

feilipu, 2020-06-13:
Checked through with `math48` and `math32`.

Added in nibble rounding of least significant bit. That captures truncated bits, but doesn't allow the least significant bit to propagate in significance. (Caveat: not a mathematician).

Will complete the last functions over the next days. But if you'r happy, don't hold up merging because of that.
- [ ] `sqrt()`
- [ ] `poly()`

feilipu, 2020-06-13:
I ran it `math16` through the mandelbrot benchmark, and works perfectly.
The only thing I had to touch was getting the constants to be _Float16 [on these lines](https://github.com/z88dk/z88dk/commit/f2c9dde800892b9372f3918226ad7deec70debe5#diff-08949eac6168065b9b81e59bd3df96c7L92).
The multiplicative constants were converted to `l_f16_ldexp` correctly. :+1: 

Other benchmarks will have to wait until higher functions are available.

```C
  Cr = (2.0*x/w - (half_t)1.5);
  Ci = (2.0*y/h - (half_t)1.0);
```

```
sccz80 / new c library
1826 bytes less page zero
cycle count  = 3265477446
time @ 4MHz  = 3265477446 / 4*10^6 = 13 min 36 sec

sccz80 / new c library / math32
3663 bytes less page zero
cycle count  = 1517530881
time @ 4MHz  = 1517530881 / 4*10^6 =  6 min 20 sec

sccz80 / new c library / math16
1891 bytes less page zero
cycle count  = 1103113465
time @ 4MHz  = 1103113465 / 4*10^6 =  4 min 36 sec
```

suborb, 2020-06-13:
I've got a PR up to add sccz80, just reviewing the user interface and I think we might need a few changes to match the ISO spec linked above:

- We should always include `<math/math_math16.h>` when math.h is included
- The constants need to have F16 in them, eg `HUGE_VAL_F16`
- Functions should be `ceilf16` rather than `f16_ceil` for example
 
I'd also like to embed math16 into the classic libraries - I don't think we're going to get another implementation so there's a bit of build juggling for me to figure out.

The conversion routines for bbcmath/mbf/genmath also need to written - I think I should be able to steal the ones you did for math48.

feilipu, 2020-06-14:
It is a bit lazy to not calculate [the magic number](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math16/z80/asm_f16_sqrt.asm#L108) for the `invsqrt()` initialisation in `_f24`, but got to leave something for a rainy afternoon.

feilipu, 2020-06-15:
Done the reorganisation to suit ISO standard.
Last thing on my list is to get `powf16()` working, and therefore everything underneath that.

suborb, 2020-06-15:
Since math16 is now bound to math32 I'm going to have to do a little bit of reorganisation of the classic libraries to extract the asm_f32_XXX from -lmath32.

suborb, 2020-06-16:
That must be me misreading there sqrt code - math32 must be needed for printing rather than sqrt itself.

feilipu, 2020-06-16:
Ah... you got me. The `_f32` format is just needed to calculate a seed using the "Quake magic number". Other than the `_f32` format, there's no code needed. We're clean of `_f32` code.

Any (either) of the print libraries are working perfect.

feilipu, 2020-06-21:
The [`poly()` function is done](https://github.com/z88dk/z88dk/commit/fe7821d3ab160f2cacfe69854cef343c339271ab).

I decided to use `_f32` float format for the coefficients (rather than `_f16` half format), for a couple of reasons.

- Accuracy. The `poly()` function uses the `_f24` format (16-bit mantissa) internally for multiply-add. Using IEEE float coefficients (with 24-bit mantissa) provides the most accuracy that the function can consume.
- Performance. Converting the coefficients from `_f32` to `_f24` for calculations is faster than converting from `_f16` to `_f24`, even though more bytes are stored.
- Convenience. I already have tables of `_f32` coefficients proven for math32, so it is much easier to just reuse them.

At some stage, I'll reduce the order of the coefficients to reduce the number of Horner iterations required, in line with the accuracy of the `_f16` format.

 - [ ] optimise the number of Horner iterations for each of the higher functions, to provide just enough accuracy for the `_f16` format.

On to the higher functions...

feilipu, 2020-06-21:
@suborb I've [committed a selection of higher functions](https://github.com/z88dk/z88dk/commit/3fc4bff06fb113edbc14b2a77be5dca6fcc2c9ff), and compiled them with sccz80. But, I'm not sure how to sensibly link them for general use with sccz80, or for zsdcc compiles too. I guess they could be compiled with sdcc_ix like math32, but that would be much more difficult to maintain. Could you have a look please, when time permits?

feilipu, 2020-06-26:
I think I've isolated an issue with an incorrect return from a callee function.
In the [`powf16()`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math16/c/powf16.c) function there are some comparisons, which when enabled (not commented out), somehow cause the number of pops from the function to reflect 4-byte variables (floats), rather than the 2-byte variables they should be. If the comparisons are disabled, then the return provides the right number of pops. In the repository, it is not declared as callee, so the return is different.

The standard [sccz80 math.h](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sccz80/math.h#L578) include is active.

```c
half_t powf16 (half_t x, half_t y)
{
//  if(x <= 0.0) return (half_t)0.0;
//  if(y == 0.0) return (half_t)1.0;
//  if(y == 1.0) return x;

    return expf16 (logf16 (x) * y);
}
```
```asm
38    0000              ;    return expf16 (logf16 (x) * y);
38    0000              	C_LINE	40,"./functions/powf16.c::powf16_callee"
40    0000              	C_LINE	40,"./functions/powf16.c::powf16_callee"
40    0000              ;logf16 (x) *y;
40    0000              	C_LINE	41,"./functions/powf16.c::powf16_callee"
41    0000              ;x;
41    0000              	C_LINE	41,"./functions/powf16.c::powf16_callee"
41    0000  21 04 00    	ld	hl,4	;const
41    0003  CD 00 00    	call	l_gintsp	;
41    0006  CD 00 00    	call	logf16
41    0009  E5          	push	hl
41    000A  21 04 00    	ld	hl,4	;const
41    000D  CD 00 00    	call	l_gintsp	;
41    0010  CD 00 00    	call	l_f16_mul
41    0013  CD 00 00    	call	expf16
41    0016  C1          	pop	bc
41    0017  F1          	pop	af
41    0018  F1          	pop	af
41    0019  C5          	push	bc
41    001A  C9          	ret
```

But this C code produces too many pops.
It also seems to be looking for the variables at the wrong stack offset.
```c
half_t powf16 (half_t x, half_t y)
{
//  if(x <= 0.0) return (half_t)0.0;
//  if(y == 0.0) return (half_t)1.0;
    if(y == 1.0) return x;

    return expf16 (logf16 (x) * y);
}

```
```asm
38    0000  21 02 00    	ld	hl,2	;const
38    0003  CD 00 00    	call	l_gintspsp	;
38    0006  21 00 3C    	ld	hl,15360	;const
38    0009  CD 00 00    	call	l_f16_eq
38    000C  7C          	ld	a,h
38    000D  B5          	or	l
38    000E  CA 1C 00    	jp	z,i_2
38    0011  21 04 00    	ld	hl,4	;const
38    0014  CD 00 00    	call	l_gintsp	;
38    0017  C1          	pop	bc
38    0018  F1          	pop	af
38    0019  F1          	pop	af
38    001A  C5          	push	bc
38    001B  C9          	ret
38    001C              
38    001C              
38    001C              ;    return expf16 (logf16 (x) * y);
38    001C              	C_LINE	40,"./functions/powf16.c::powf16_callee"
40    001C              .i_2
40    001C              	C_LINE	40,"./functions/powf16.c::powf16_callee"
40    001C              ;logf16 (x) *y;
40    001C              	C_LINE	41,"./functions/powf16.c::powf16_callee"
41    001C              ;x;
41    001C              	C_LINE	41,"./functions/powf16.c::powf16_callee"
41    001C  21 0A 00    	ld	hl,10	;const
41    001F  CD 00 00    	call	l_gintsp	;
41    0022  CD 00 00    	call	logf16
41    0025  E5          	push	hl
41    0026  21 0A 00    	ld	hl,10	;const
41    0029  CD 00 00    	call	l_gintsp	;
41    002C  CD 00 00    	call	l_f16_mul
41    002F  CD 00 00    	call	expf16
41    0032  C1          	pop	bc
41    0033  C1          	pop	bc
41    0034  C1          	pop	bc
41    0035  C1          	pop	bc
41    0036  F1          	pop	af
41    0037  F1          	pop	af
41    0038  C5          	push	bc
41    0039  C9          	ret
```

suborb, 2020-06-26:
I don't know how long it's been like that, but the code went to the effort of saving the "auto base pointer" before generating the cleanup but didn't restore it after it was done so it thought the "auto base pointer" was moving down down in memory.

feilipu, 2020-06-27:
Is there a way to use `z80rules.8` with `sccz80`? I’ve looked in `zcc` but it they are only called out for clang compiles.

Just thinking the few rules there would save quite a few cycles in `_Float16` function code, for not much overhead.

feilipu, 2020-06-27:
And a new issue. I'm having trouble getting comparison with hex constants to work.
There is some internal conversion process going on, and I'm not sure where it comes from.

```asm
167   0121              ;            if ( hm <  (0xC8DA)  )
167   0121              	C_LINE	169,"./ftest.c::main"
169   0121              	C_LINE	169,"./ftest.c::main"
169   0121  21 1A 00    	ld	hl,26	;const
169   0124  CD 00 00    	call	l_gintspsp	;
169   0127  21 46 7A    	ld	hl,31302	;const
169   012A  CD 00 00    	call	l_f16_lt
169   012D  7C          	ld	a,h
169   012E  B5          	or	l
169   012F  CA 3F 01    	jp	z,i_8
```

```asm
167   0121              ;            if ( hm < (half_t) (0xC8DA)  )
167   0121              	C_LINE	169,"./ftest.c::main"
169   0121              	C_LINE	169,"./ftest.c::main"
169   0121  21 1A 00    	ld	hl,26	;const
169   0124  CD 00 00    	call	l_gintspsp	;
169   0127  21 46 7A    	ld	hl,31302	;const
169   012A  CD 00 00    	call	l_f16_lt
169   012D  7C          	ld	a,h
169   012E  B5          	or	l
169   012F  CA 3F 01    	jp	z,i_8
```

```asm
167   0121              ;            if ( hm < 0x78DA )
167   0121              	C_LINE	169,"./ftest.c::main"
169   0121              	C_LINE	169,"./ftest.c::main"
169   0121  21 1A 00    	ld	hl,26	;const
169   0124  CD 00 00    	call	l_gintspsp	;
169   0127  21 8D 77    	ld	hl,30605	;const
169   012A  CD 00 00    	call	l_f16_lt
169   012D  7C          	ld	a,h
169   012E  B5          	or	l
169   012F  CA 3F 01    	jp	z,i_8
```

suborb, 2020-06-27:
I don't think it's ever supported hex floating point constants. So what's happening is the integer value of the hex constant is being converted to _Float16 internally.

I'll need to do a bit of work to support it I'm afraid - I haven't found a quick workaround.

suborb, 2020-06-27:
> Is there a way to use z80rules.8 with sccz80? I’ve looked in zcc but it is are only called out for clang compiles.

It should be `-opt-code-speed=inlineints`

feilipu, 2020-06-27:
> I don't think it's ever supported hex floating point representations. So what's happening is the integer value of the hex constant is being converted to _Float16 internally.

Ok. It is not a big deal, either way. I was only using hex to compare because SDCC couldn't support the conversion. The only reason that it is needed is to pass a `_Float16` NaN back as a return, and that works fine. I'll just convert the limits to decimal, which is easy because [I already have them](https://github.com/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sccz80/math.h#L130).

suborb, 2020-06-27:
Try these values:

```
6.10352e-5;
-6.10352e-5;
```

They will get you the values you expect (from https://blogs.mathworks.com/cleve/2017/05/08/half-precision-16-bit-floating-point-arithmetic/)

feilipu, 2020-06-28:
I've worked on the [n-body benchmark](https://github.com/z88dk/z88dk/wiki/Benchmarks#n-body). And the results are pretty good. Quite happy with the outcome so far.

Compiler_Library | Maths Library | Bytes | Ticks | Seconds @ 4MHz
-- | -- | -- | -- | --
SCCZ80_NEW | `math48` | 3,608 | 2,372,283,755 | 9 min 53 sec
SCCZ80_NEW | `math32` | 5,264 | 1,025,105,884 | 4 min 16 sec
SCCZ80_NEW | `math16` | 3,227 | 0,384,230,543 | 1 min 36 sec

I had to [increase the DT (delta time)](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/n-body/z88dk-new/n-body.c#L189) to ensure the resulting values (arising from a 1/DT calculation) stayed within `half_t` range, so the 1,000th iteration is 10x later, and therefore it is a bit different than expected. But it is still within 2.5 decimal points. So we're about 6x faster than `math48`.

suborb, 2020-06-28:
That's an impressive improvement in performance - why is math32 so large btw?

feilipu, 2020-06-28:
`math32` was compiled with fast integer maths settings, so all loops are unrolled, and the z80 has a special `16_8x8` unrolled multiply. Also, the functions for add and normalise have a lot of linear branch code. Square root and inverse (for divide) have in-lined N-R iterations. It is quite bloaty in general.

I've actually thought about going back to rewrite the N-R iterations as subroutines, but it would be a difficult rework as there are no free registers. So pretty low on the to-do list at the moment.

feilipu, 2020-06-29:
Just poking about in the classic examples, and there are examples for the spectrum that use sine and cosine.
Would be good to have a play with that at some stage, to see what effect 16-bit floating point has "on-screen".

[sinf16()](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math16/c/sinf16.c) is working, but not linked into the build yet. The trig, log, exp C functions have been tested working already.

[33 more replies]

closed 2020-07-12 by feilipu