z88dk/z88dk #1468: [info] Which targets are known to the devs to be "non-working" or "non-tested"

Fabrizio-Caruso, 2020-05-12:
It would be very useful if it were clear which targets are only there as stubs but are not meant to be used, yet, because they are known not to be working at all (not even a simple "hello world" can has been seen working by the devs).

Which targets are known to you (devs) not to be working?
only xo7, c7420, m100?
maybe ab800?
I have seen many new targets. I have no idea whether I should try to test them because I have no idea if they are just there for internal testing by the devs.

Which targets have not been tested on either emulator nor real hardware?
only rex?

suborb, 2020-05-12:
As I'm working on new/existing targets, I'm adding information at the top of the wiki page along the lines of:

[image]

This shows what features are available - if it's there then I will have tested it in an emulator. I've only recently started doing this so it doesn't cover everything, for example ZX Spectrum, VG5k, Gameboy, Jupiter ACE don't appear there and we know they are working.

So, an alternate way to determine this is from the Platform page: https://github.com/z88dk/z88dk/wiki/Platform - if gencon is supported then I will have run in an emulator.

Fabrizio-Caruso, 2020-05-12:
@zx70 Yes, indeed but how do I know if a target is known to be not working?
Let us put side the untested cases. Which targets are just stubs as far as you know?
I know xo7, c7420, m100 are not meant to be used. abc800 is very hard to test. rex is even hard to test.

As you say, not having generic console is not not enough as lots of targets with no generic console do work fine such as NC100, NC200, Knights 200, Z88, etc.

zx70, 2020-05-13:
The targets you listed are the ones I am aware of.   In most of the cases I was declaring they were experimental,  but I'm noticing that it is not enough to stop someone's enthusiasm :)
This is painful but stimulating.  Without a similar situation I'd have never fixed the zx81 port!

Fabrizio-Caruso, 2020-10-25:
I have gone though testing ALL Z88DK targets except pacman and the ones that are not testable without the real hardware.
I confirm that ALL of them work including abc800, vgl (not listed as a target).
I could compile and run Cross Chase (at least in turn-based, stdio mode).
So I confirm that we only have the 3 non-working targets mentioned above.

closed 2020-10-25 by Fabrizio-Caruso

feilipu, 2020-10-25:
AFK for  a few weeks, but I can help with [testing HBIOS, RC2014, SCZ180, and YAZ180](https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/docs/STATUS.md#8-bit-new-retro-targets-with-no-known-emulators) targets. It would be easiest for me if you there was a standard recipe for preparing a STDIO (turn-based for RC2014 with no clock) version.

All of these targets only have STDIO within z88dk, although HBIOS, and RC2014 can be extended with currently unsupported video interfaces. The Z180 platform targets (SCZ180, YAZ180) have a clock implementation, and HBIOS has a clock interface too.

Fabrizio-Caruso, 2020-10-25:
@feilipu  I can provide it with the binaries or instructions on how to compile them. For the very purpose of testing Z88DK and not just testing Z88DK to run my games, you have probably already tested the targets and could tell us if they are working and usable.
Newvrtheless my test is useful as it is a non-trivial relatively large game (bigger than 12k).

feilipu, 2020-10-25:
Those newlib targets should work as advertised. But it is always good to test more deeply.
Perhaps open a new issue specifically for testing newlib targets using sccz80 and zsdcc?
I can't get to it till mid November though, but soon enough I guess.

zx70, 2020-10-25:
I recently discovered this guy's work:
https://www.mh-aerotools.de/hp/

he mastered the old cpm-80 programming tricks and was able to write code for the GSX-80 extension,  both by providing new drivers and by writing the API for the old FORTRAN language. 
I remember that the drivers need to be in relocatable format (PRL), but since the standard programming languages were used, this shouldn't be mandatory on the programs side.

it's a bit out of topic, but I didn't want to open a new issue and be in  the way for whoever is willing to begin playing on that immediately  ;)