z88dk/z88dk #1474: (zsdcc) Compile times larger?

aralbrec, 2020-05-15:
I'll be investigating this later but I am wondering if anyone has noticed significantly longer compile times using zsdcc in the current build compared to 1.99C?

On one project, using windows, the compile time of the most dastardly file is 2 hours and 20 mins on my laptop.  This compared to about 45 minutes using 1.99C.  I will be rolling back to make sure I am remembering correctly and to make sure something hasn't changed in that file to cause this but 2 hours is really excessive.

suborb, 2020-05-15:
My hunch is MSVC vs mingw32 too.

```
x86_64-w64-mingw32-gcc -v
Using built-in specs.
COLLECT_GCC=x86_64-w64-mingw32-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-w64-mingw32/6.2-win32/lto-wrapper
Target: x86_64-w64-mingw32
Configured with: ../../src/configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --prefix=/usr --enable-shared --enable-static --disable-multilib --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --libdir=/usr/lib --enable-libstdcxx-time=yes --with-tune=generic --enable-version-specific-runtime-libs --enable-fully-dynamic-string --enable-libgomp --enable-languages=c,c++,fortran,objc,obj-c++ --enable-lto --with-plugin-ld --enable-threads=win32 --program-suffix=-win32 --program-prefix=x86_64-w64-mingw32- --target=x86_64-w64-mingw32 --with-as=/usr/bin/x86_64-w64-mingw32-as --with-ld=/usr/bin/x86_64-w64-mingw32-ld --disable-libatomic
Thread model: win32
gcc version 6.2.0 20160822 (GCC)
```

Which may be lacking some newer optimisations. I can't post the following on the specnext forum (error 403), in the following Linux is using gcc 8.3.0:

> To stop us comparing apples and oranges I've re-run the tests on a Linux lxc container running on the same machine that the Windows kvm instance is running on.

>-clib=sdcc_iy 2.2s
>-clib=new  0.75s
>-clib=classic 0.5s

>All of the sdcc builds are dominated by the time zsdcc takes to compile: I've not built an MSVC version but it would be interesting to see whether that performs better. I suspect it would, but I've no idea by how much. Another interesting experiment would be to use the docker image to do the compilation on windows and see what the difference is. Just out of interest how does your self-built sdcc perform in comparison with zsdcc?

Fabrizio-Caruso, 2020-05-15:
Compilation now takes more than 10 times more.
I can confirm it for sure.
If I try to build my game (for about Z88DK 80 targets with no optimization), it takes about 6 hours.

aralbrec, 2020-05-16:
Are you running on windows?

I haven't noticed much change on sccz80 but sccz80 has always been fast.  The recent change to z80asm's linking has made a big difference in linking speed on windows.

I do think there's a possibility that visual studio compiled binaries will perform better on windows than cross-compiled ones which maybe we can test with zsdcc.  I did update the vs sln file ( `win32/z88dk.sln` ) so that most of the z88dk binaries can be compiled with visual studio again, including sccz80, but zsdcc is a separate problem.  zsdcc cannot be compiled with visual studio without numerous changes which I did make way back in 1.99C but I am not sure if they carry forward to the current version.  We'll find out I guess.

ped7g, 2020-05-16:
The latest clang on windows should (even in generic case) produce binaries on par with MSC - gcc/clang were already for years quite ahead in vectorization and other special cases, especially if you spend some time hand-holding the compiler and adjusted the C++ source to get desired machine code, but in regular mundane code the MSC was somewhat more friendly toward windows and usually it has least friction when you are full MS stack based and already locked-in to the MS. But lately (except some setup friction) the clang can produce production-grade binaries. Of course it still makes sense from development perspective to build+test with all three major compilers, as that usually uncovers most bugs in the source, but for final binaries the clang can now handle all major targets for most of the use cases/projects.

suborb, 2020-05-16:
I've just tried out sdcc compiled with mingw clang-9.0, unfortunately it segfaults - as does a compilation using a much older version of clang. This makes me think that my cross mingw clang setup isn't quite right (though I've set it up two different machines).

Compiling with gcc 9.2-win32 20191008 doesn't yield any performance improvement.

aralbrec, 2020-05-16:
Just to update:  1.99C compile of the problem file took 35 minutes on my windows laptop versus the 2 hours 20 mins of the current version in one configuration and 3 hours 40 minutes in another.

spth, 2020-06-08:
Could you try to compile that file with just plain sdcc, and see if the slowdown is there too?
If yes, maybe also track down in between which sdcc versions it happened to file a bug in upstream sdcc?

aralbrec, 2020-06-24:
@spth  Yes I will give it a try if an msvc build does not improve things.  It may take a while yet to get around to it as I am pretty short on time.  If you want to give it a try maybe I can ask the author to forward to file to you.

suborb, 2020-07-05:
Scribbling down some notes.

Just tested with a few binaries hanging around the place (zsdcc) trying to get the range where this happened.

Using the hex_newlib.c file attached and then invoking zsdcc -S [file] we get:

zsdcc 3.8.5 (10892) -> ~0.7 seconds
zsdcc 11283 -> 3.5 seconds

Somewhat curiously, when compiling with classic (hex_classic.c), the time is roughly double the above (removing the `__smallc` annotations makes no significant difference).

[hex_classic.c.txt](https://github.com/z88dk/z88dk/files/4875356/hex_classic.c.txt)
[hex_newlib.c.txt](https://github.com/z88dk/z88dk/files/4875357/hex_newlib.c.txt)

suborb, 2020-07-06:
Oh dear. I managed to find a backup (from just after I compiled 3.8.5) of the build container from last year and redeployed it.

I ran the configure command and generated the binary. Same results as above (3-4seconds).

Thought I'd glance at the command history. Noticed something very odd.

For the past year I've been copy/pasting the following:

```
CXXFLAGS=-I/usr/x86_64-w64-mingw32/include/ ./configure --disable-mcs51-port --disable-avr-port --disable-ds390-port  --disable-ds400-port --disable-hc08-port --disable-pic-port --disable-pic14-port  --disable-pic16-port --disable-stm8-port --disable-tlcs90-port --disable-s08-port --disable-pdk13-port --disable-pdk14-port --disable-pdk15-port --disable-ucsim --disable-device-lib --disable-packihx --host=x86_64-w64-mingw32
```

On the old machine the command was:

```
./configure --disable-mcs51-port --disable-gbz80-port --disable-avr-port --disable-ds390-port --disable-ds400-port --disable-hc08-port --disable-pic-port --disable-pic14-port --disable-pic16-port --disable-stm8-port --disable-tlcs90-port --disable-s08-port --disable-ucsim --disable-device-lib --disable-packihx --host=x86_64-w64-mingw32
```

Spot the difference. Yup, I was overriding the C++ flags. Check the profiling dump above. What takes the time? Yup, the C++ code. What's the main C++ code in sdcc? The register allocator. Why is sdcc slow at compiling? The register allocator. What flags did I end up removing? The optimisation flags.

I'll just go hang my head in shame and figure how to automate/idiot proof this stage.

Binaries updated, the build tonight should be fast now.

closed 2020-07-06 by suborb

feilipu, 2020-07-06:
Just a thought... is disabling this right ` --disable-gbz80-port`?
The [readme](https://github.com/z88dk/z88dk/blob/master/src/zsdcc/readme.md) line leaves it in.

Also, the whole pdk series could be disabled too, as it would make the build faster/smaller.
`--disable-pdk13-port --disable-pdk14-port --disable-pdk15-port --disable-pdk16-port`