z88dk/z88dk #1475: (sccz80) Compiler time longer by a significant factor

Fabrizio-Caruso, 2020-05-15:
I am experiencing much longer compilation times on actually both SCCZ80 and ZSDCC (#1474) with maybe a tenfold increase.

I can confirm this with CROSS CHASE. 
Compilation with no optimization options of my game (for about 80 targets) now takes about 6 hours.

suborb, 2020-05-15:
When did this change?

Fabrizio-Caruso, 2020-05-15:
@suborb I am not sure as I don't build all of the targets very often but
I think it was OK last time I had to do it for my binary release (April 3rd).

Fabrizio-Caruso, 2020-05-15:
@suborb I need to run some real measurements. I massively build both sccz80 and zsdcc targets but I have seen it very slow on any target.

suborb, 2020-05-15:
Thankfully it's a quiet time so there's not many changes.

I'm focusing on sccz80 (the last time zsdcc was updated was for 15/3/2020) since it's really unusual for that to be slow.

* Nightly build changes: Correcting the prefix (so #1433 fix)
* There's been no optimisation rule changes since January.
* z80_crt0s.lib - Only really +zxn changes
* lib/ - Only new targets added
* include/ - Only +zx/+zxn changes

Looking at the binaries (dates correspond to nightly builds) - I'm ignoring appmake changes since they shouldn't be relevant.

15/5/2020 - Paulo's z80asm change
26/4/2020 - Commands are flushed to screen (windows fix)

The only one that could have a significant effect would be the z80asm change, I've only seen build time decrease as a result of this particular change. 

Have you made any changes on your side regarding your build?

Fabrizio-Caruso, 2020-05-15:
@suborb First of all thanks for your analysis. 
Yes, I have changed both the code and my Makefiles. The code include some optimizations and a some reshuffling of some code in different files.

suborb, 2020-05-15:
I'm missing makewzd.exe so my build stops early (the +oz build fails), but some times from make z88dk_targets.

This is on a kvm based Win10pro machine (i5-6260)

z88dk from 15/5, Cross chase 223d8fea  -> 14m8
z88dk from 15/5, Cross chase 4ca6e869 -> 14m25
z88dk from 1/4, Cross chase 4ca6e869 -> 13m2 (stopped at Mikro80 since it didn't exist then)

Given that z88dk 1/4 stopped 13 machines earlier than the others I think it's fair to say that the compilation speed has improved over the timeframe being tested.

So either the timeframe is wrong, or you've got an environmental problem?

Fabrizio-Caruso, 2020-05-15:
@suborb you are probably right (as usual).
I am wondering what is causing the compilation to take so long now.
6 hours with no optimizations means that it will take me half a day on a fast quad core...

closed 2020-05-15 by Fabrizio-Caruso

suborb, 2020-05-16:
Another data point:

z88dk from 1/3, Cross chase 4ca6e869 -> 12m35 (stopped at Mikro80 again)

Between 1/3 and 1/4 there was an zsdcc upgrade. That's obviously contributed to some slowdown, but it's nothing like the massive increase that you're seeing.

All of my hardware is practically vintage: my main dev machine is a 2012 laptop and the newest "modern" hardware I own is from (I think) 2016 so if it works well for me then it should be much better for people who've upgraded in the past 5 years.

It might be worth you trying the docker config here: https://github.com/z88dk/z88dk/wiki/Docker-Usage - on the same hardware I do see significant difference in performance between Windows and Linux, I don't think I can easily test a VM within a VM to see what the "real world" difference is.

suborb, 2020-06-01:
This issue was resolved by the investigation on #1482