z88dk/z88dk #1477: "The system cannot find the file" but no visible error message about the file and error details

Fabrizio-Caruso, 2020-05-19:
I am compiling Cross Chase for the Laser500 with some new settings and I get
```
PROCESSING mv.asm.m4
m4 -I "c:/z88dk/\lib\target\laser500\classic" -I "c:/z88dk//src/m4"  --define=__STARTUP=1  < "mv.asm.m4" > "C:\c
ygwin\tmp\zccEFFB19"
The system cannot find the file specified.
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_l-m:29: laser500_bitmap] Error 1
```
I compile in vernose -v mode but I do not see any error.

I have attached the output produced by the compilation.
[no_debug_info.txt](https://github.com/z88dk/z88dk/files/4651215/no_debug_info.txt)

suborb, 2020-05-19:
```
$ make laser500_bitmap
/cygdrive/c/z88dk/bin/zcc.exe +laser500  -vn -create-app -Cz--audio -v \
-Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_data -Icross_lib/display/graphics_data/cc65 -Icross_lib/display/graphics_data/z88dk -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/display/image_settings -Icross_lib/display/redefine_characters -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/psg -Icross_lib/sound/lcc1802/comx -Icross_lib/text -Icross_lib/input -Icross_lib/rand \
-D__LASER500__ \
-DFORCE_BIT_BANG -DSOUNDS \
-DFULL_GAME \
-DNO_GRAPHICS -DNO_PRINT -DNO_INPUT -DNO_SET_SCREEN_COLORS \
-lndos \
cross_lib/display/display_macros.c cross_lib/display/init_images.c cross_lib/input/input_macros.c cross_lib/text/text_macros.c chase/move_player.c chase/ghost.c  chase/level.c chase/character.c chase/game_text.c chase/main.c chase/strategy.c chase/item.c chase/skull.c chase/bullet.c   chase/horizontal_missile.c chase/rocket.c  \
cross_lib/display/init_graphics/z88dk/laser500/laser500_bitmap_init_graphics.c \
mv a.bin ../build/laser500_bitmap.bin
```

I've snipped the file. You've added mv, a.bin and ../build/laser500_bitmap.bin into the list of files to be compiled. That is, you've got an errant backslash in your makefile after `laser500_bitmap_init_graphics.c`

Fabrizio-Caruso, 2020-05-19:
Sorry for this stupid issue.

closed 2020-05-19 by Fabrizio-Caruso